arendst/Tasmota #5353: Sonoff iFan02 and SetOption59 and retain last speed

WayneManion, 2019-02-26:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
I like to use SetOption59 set to 1 on my devices so Home Assistant can provide near-instant status updates. I have a Sonoff iFan02 with SetOption59 set to 1, and while the light sends ~/tele/STATE messages immediately when the light turns on and off, turning the fan on/off and changing the fan speed does not provide a message on the ~/tele/STATE topic. I have a workaround in Home Assistant for this, but it would be more consistent if fan speed changes sent a a state message.

It would also be nice if the iFan02 could retain the last previous fan speed when turned off. That way turning it back on would go back to the previous speed. 

**Describe the solution you'd like**
Have the iFan02 send ~/tele/STATE messages when the fan it toggled or changes speeds. 

As for fan speed retention, maybe sending the message "4" on the ~/cmnd/FanSpeed topic would re-activate the retained fan speed. 

**Describe alternatives you've considered**
I have a wildcard MQTT topic setup in Home Assistant to catch the messages in the ~/stat/RESULT and ~/tele/STATE topics regarding fan speed state. The ~/tele/STATE messages are sent only when the device gives its timed status reports. The ~/stat/RESULT messages are sent when the fan speed changes. 

Thanks.

andrethomas, 2019-02-26:
I think its a good idea in principle - I'm not sure if all conventional fans would like this though - most of them prefer to start on the highest speed and then slow down to the desired speed. This is also the main reason why ceiling fans traditionally start on the fastest speed setting and then click down to the lower speeds... the motor is not guaranteed to start straight off the bat on the lowest speed setting.

Yes, it should in theory, but it can be problematic if it does not.

digiblur, 2019-02-26:
I do believe the iFan02 is smart enough and it clicks in the other speed at startup on a low speed for a few seconds.  At least that's why I believe the clicking is.  Not sure if Theo did that or if the other MCU does that.

andrethomas, 2019-02-26:
@digiblur cool noted - mine still waits in a DHL bag to be installed :)

arendst, 2019-02-27:
Guys, on the iFa02 the mcu is king! Tasmota can only follow and try to stay in sync with whatever the mcu thinks it should do.

So yes, going from off to slow the mcu starts full power and drops back within some seconds to the desired fanspeed settings. Tasmota has no control over this.

WayneManion, 2019-02-27:
The speed memory was just a secondary concern. The main issue is the lack of consistency with SetOption59 on the iFan02. Is that something that can be addressed without a lot of trouble?

ascillato2, 2019-03-06:
Closing this issue as it has been answered by Theo:

> on the iFa02 the mcu is king! Tasmota can only follow and try to stay in sync with whatever the mcu thinks it should do.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2