arendst/Tasmota #5355: Sonoff dual R1 button sopped working after upgrading to 6.4.1

BugsonPL, 2019-02-26:
Sonoff dual R1 button sopped working for Relay2 with double press after upgrading to 6.4.1.

Be for update it was working fine.

Unfortunately I dno't remember the previous build.

 I can make some additional tests if needed.

ascillato, 2019-02-26:
Please, complete the troubleshooting template in order to have more information so as to properly help you. Thanks

ascillato, 2019-02-26:
And add also the output of the console when doing that. Remember to put weblog to 4

ascillato2, 2019-02-27:
Please, also try the latest version from the development branch at thehackbox.org/tasmota

ascillato2, 2019-02-27:
Closing this issue as there is no feedback. Please, when you have time, complete the troubleshooting template. Thanks

closed 2019-02-27 by ascillato2

gerardovf, 2019-12-13:
I can confirm that version 6.4.1 is working in several (more than 4) Sonoff dual R1 I am using. Any other newer version doesn't allow to use the external button to switch both relays.

gerardovf, 2020-08-16:
It's finally solved. Checked on version 8.4.0.2. Thanks Theo!