arendst/Tasmota #5357: How to full flash Sonoff?

shayshtern, 2019-02-27:
When I flash the Sonoff it's still remember the WiFi settings. So how do I fully flash it? 🙏🏼

localhost61, 2019-02-27:
I  suggest you first make a backup of your current setting
`http://MyDeviceIP/dl` where you have to replace MyDeviceIP with the one you assigned to your device.
Then `http://MyDeviceIP/cm?cmnd=reset%201` will do the trick.

A more classic way is by using the console to enter the [command](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) :
`Reset 1` and it should restart with the default settings of your firmware.

shayshtern, 2019-02-27:
@localhost61 I mean flash with PC.
with FTDI to USB.

localhost61, 2019-02-27:
`esptool.py --port COM3 erase_flash`
If you never used it before, look for [esptool](https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool) in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki)

shayshtern, 2019-02-27:
[image]
Can I do it with ESPEasy?

localhost61, 2019-02-27:
I don't know, I never used it. On your screen I see no option for that.

shayshtern, 2019-02-27:
In the Firmware (.bin) I choose blank_1MB.bin and click Flash ESP Easy FW.
Then reconnect to flash mode with PC and flash with sonoff.bin.
CLOSE.

localhost61, 2019-02-27:
Close this issue if you're done. Thks

closed 2019-02-27 by shayshtern

localhost61, 2019-02-27:
I don't know were you get blank_1MB.bin, it seems to me you're mixing Tasmota and EspEasy projects.
EspEasy flasher is a loader for EspEasy firmware, it's not intended to be used with Tasmota firmware.
Of course, you can try but then you should understand what you do... and learn from your own experiments.