arendst/Tasmota #5358: IrReceive & Gosund SP1 not working

Meinsda, 2019-02-27:
Hello

I want to use a Gosound SP1 to switch a TV and also to control them via IR. First step i use a IR receiver chip TSOP4838. I use rx or tx to communicate with the esp but get no communction and see no receiving Information. I have test the pins as outout port from the esp and they are working.
Should the irreceiver work on rx/tx or not=

Peter

ascillato2, 2019-02-27:
Hi,

Please, could you be so kind on completing the troubleshooting template in order to have more information so as to properly help you? Thanks

Meinsda, 2019-02-27:
x  Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
x  Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
x  Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
x  Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
x  Development/Compiler/Upload tools used : Arduino 1.8.8
x  Hardware used : Gosund SP1 & TSOP4838
x You have tried latest release or development binaries? : yes
 Provide the output of command status 0 :
22:08:06 MQT: stat/sonoff/STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:08:06 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T06:38:25","StartupUTC":"2019-02-27T14:29:41","Sleep":50,"BootCount":40,"SaveCount":235,"SaveAddress":"F9000"}}
22:08:06 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-02-25T20:04:10","Boot":4,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
22:08:06 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Worki",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000001"]}}
22:08:06 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":548,"Free":452,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
22:08:06 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3320","IPAddress":"192.168.88.121","Gateway":"192.168.88.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.88.1","Mac":"EC:FA:BC:93:AC:F8","Webserver":2,"WifiConfig":4}}
22:08:06 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.88.120","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_93ACF8F8","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:08:06 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 27 21:08:06 2019","Local":"Wed Feb 27 22:08:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:36","Sunset":"18:29"}}
22:08:06 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:08:06 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-27T22:08:06","ENERGY":{"TotalStartTime":"2019-02-25T20:21:47","Total":0.399,"Yesterday":0.047,"Today":0.352,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
22:08:06 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-27T22:08:06","Uptime":"0T06:38:25","Vcc":3.475,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Worki","BSSId":"D4:CA:6D:BE:44:87","Channel":1,"RSSI":100}}}

ascillato2, 2019-02-28:
Hi, sorry. Sensors are not supported in devices with power measurements due to safety concerns. It is very dangerous due GND is tied to mains ac.

Closing this issue as it is not supported. Sorry

ascillato2, 2019-02-28:
As a workaround, please use a wemos mini or a nodemcu. Thanks

closed 2019-02-28 by ascillato2