arendst/Tasmota #5359: Fallback topic is different between 6.4.1.18 Core_2.4.2 and Core_2.5.0

wongnam, 2019-02-27:
**Describe the bug**
I found that my Sonoff "Fallback_topic" is changed if i flashed it to 6.1.4.18 Core_2.5.0, It's normal with 6.4.1.18 Core_2.4.2

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_ Yes
- [x] _Hardware used_ :  Sonoff S22
- [x] _Development/Compiler/Upload tools used_ : VSC + PlatformIO 
- [x] _If a pre-compiled release or development binary was used, which one?_ : Self_compile
- [x] _You have tried latest release or development binaries?_ : Dev version 6.4.1.18 Core_2.5.0
- [x] _Provide the output of command_``status 0`` :
```
00:00:00 Project sonoff Sonoff Version 6.4.1.18(sonoff)-2_5_0
00:00:00 WIF: Connecting to AP1 iot-2 in mode 11N as sonoff-1737...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-1737 with IP address 192.168.12.111
17:33:19 MQT: Attempting connection...
17:33:19 MQT: Connected
17:33:19 MQT: tele/sonoff/LWT = Online (retained)
17:33:19 MQT: cmnd/sonoff/POWER = 
17:33:19 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff TH","Version":"6.4.1.18(sonoff)","FallbackTopic":"cmnd/DVES_0F86C9C9_fb/","GroupTopic":"sonoffs"}
17:33:19 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1737","IPAddress":"192.168.12.111"}
17:33:19 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
17:33:19 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
17:33:19 MQT: stat/sonoff/POWER = OFF
17:33:27 MQT: tele/sonoff/STATE = {"Time":"2019-02-27T17:33:27","Uptime":"0T00:00:14","Vcc":3.447,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":56,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:34:37 CMD: status 0
17:34:37 MQT: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:34:37 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:24","StartupUTC":"2019-02-27T10:33:13","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":7,"SaveAddress":"F5000"}}
17:34:38 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.18(sonoff)","BuildDateTime":"2019-02-25T17:39:28","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
17:34:38 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-2","Wong"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
17:34:38 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":496,"Free":504,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0F082790","04000000","000000CE","000013C0"]}}
17:34:38 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1737","IPAddress":"192.168.12.111","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"2C:3A:E8:0F:86:C9","Webserver":2,"WifiConfig":4}}
17:34:38 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F86C9C9","MqttUser":"admin","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:34:38 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 27 10:34:38 2019","Local":"Wed Feb 27 17:34:38 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00","Sunrise":"06:08","Sunset":"18:02"}}
17:34:38 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-02-27T17:34:38"}}
17:34:38 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-02-27T17:34:38","Uptime":"0T00:01:25","Vcc":3.448,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

**To Reproduce**
Fallback_topic should not be changed even update to latest FW or Core version.

**Screenshots**
This screen shot of Core_2.5.0
[image]

this is a screenshot of Core_2.4.2
[image]

ascillato2, 2019-02-27:
Hi, as you are selfcompiling and as the tool chain of the core 2.5.0 have changed. Could you please try the precompiled bins? Thanks

ascillato, 2019-02-27:
I'm not totally sure, but I think this issue of the core was already addressed in the Arduino repository, so updating your core 2.5.0 to stage should solve this issue.

wongnam, 2019-02-28:
Yes, the issue is gone after update the to latest Espressif 8266 2.0.1 (Core_2.5.0). Thanks.

The strange is VSC+PlatformIO did not announce that the core had a new update, so I had to manually update it.

closed because it was resolved.

closed 2019-02-28 by wongnam

meingraham, 2019-03-01:
I upgraded one of my devices from precompiled 
```
6.3.0.16(a516b25-sonoff)
2018-12-09T18:24:40
2_4_2/2.2.1(cfd48f3)
```

To precompiled
```
6.4.1.19(9e8ed1a-sensors)
2019-03-01T17:05:04
2_5_0/3.0.0-dev(c0f7b44)
```
MQTT Client | DVES_231F92**92**
MQTT Fallback Topic | cmnd/DVES_231F92**92**_fb/

These fields are now 8 characters rather than 6. I just wanted to confirm that this is the way 2.5.0 will configure this ID. Is it the last MAC octet which is duplicated or are the last two digits derived from something else?

[image]

```
STATUS = {"Status":{"Module":18,"FriendlyName":["<myName>"],"Topic":"<myTopic>","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T00:24:35","StartupUTC":"2019-03-01T16:53:23","Sleep":50,"CfgHolder":4617,"BootCount":90,"SaveCount":449,"SaveAddress":"FB000"}}
STATUS2 = {"StatusFWR":{"Version":"6.4.1.19(9e8ed1a-sensors)","BuildDateTime":"2019-03-01T17:05:04","Boot":6,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":2,"LogHost":"<logHost>","LogPort":514,"SSId":["<mySSID>",""],"TelePeriod":300,"Resolution":"559580C0","SetOption":["00208108","280500000000000000000000000000000000","000000C0"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":598,"Free":404,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDEE794","0007A3A4","B7FFBFCC","001ABBC0"]}}
STATUS5 = {"StatusNET":{"Hostname":"<hostName>-8082","IPAddress":"192.168.1.xxx","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"<myMAC>:23:1F:92","Webserver":2,"WifiConfig":4}}
STATUS6 = {"StatusMQT":{"MqttHost":"<mqttHost>","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_231F9292","MqttUser":"<mqttUser>","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 01 17:17:58 2019","Local":"Fri Mar 01 12:17:58 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"06:56","Sunset":"18:22"}}
STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
STATUS10 = {"StatusSNS":{"Time":"2019-03-01T12:17:58","ANALOG":{"A0":67},"ENERGY":{"TotalStartTime":"2018-12-10T17:32:10","Total":66.505,"Yesterday":0.856,"Today":0.000,"Power":0.00,"ApparentPower":0.00,"ReactivePower":0.00,"Factor":0.00,"Voltage":119.00,"Current":0.000}}}
STATUS11 = {"StatusSTS":{"Time":"2019-03-01T12:17:58","Uptime":"0T00:24:35","SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"Wifi":{"AP":1,"SSId":"<mySSID","BSSId":"<apMAC>","Channel":1,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:09"}}}
```

Mike

wongnam, 2019-03-01:
it had changed since #4706.

meingraham, 2019-03-01:
It's not an IDE issue. I am using pre-compiled binaries. My question is less about the change in the fallback topic naming convention. It's more about the device ID changing from 6 characters to 8 characters and where the last two characters are derived from.