arendst/Tasmota #5360: Sonoff POW R2 unstable and very slow with 6.4.1 sonoff.bin

sfromis, 2019-02-27:
Unstable and very slow operation of Sonoff POW R2 after flashing with Tasmota 6.4.1 sonoff.bin

Searching reveals inconclusive talk about a similar issue #3482, closed. I've found the probable cause.

Hardware Sonoff Pow R2 V1.0 2017-11-30, flashed sonoff.bin with https://github.com/letscontrolit/ESPEasy/

After configuring to Sonoff POW R2 (43), web interface and button got horrible response times, often over 10 seconds. Rarely worked with hass.io and TasmoAdmin, which balks at response times over 5000 ms. Occasional "software watchdog" reboots, and relay clicking. Problem persisted after configuring back to Sonoff Basic. With cleared flash, and reflashing sonoff.bin, I've also seen rare examples of the bad response times while keeping it configured as Sonoff Basic.

After some head-scratching and circling, and getting an extra POW R2 to exclude hardware failure, I found that it worked as expected after reflashing with esptool.py, flash size 4M, instead of the probable ESPEasy default of 1M, which matches many other ESP8266 devices. I'd not be surprised if similar issues could occur with other mismatched flash sizes.

I'd like to see Sonoff-Tasmota being able to more gracefully handle this, or at least have a strong warning about this hard to debug issue.

Jason2866, 2019-02-27:
Strange problems can occur if flash isnt erased completly BEFORE flashing Tasmota.
There is no problem to flash a 1M sonoff firmware to devices with greater flash

sfromis, 2019-02-27:
I had specifically erased flash between attempts, several times, to avoid previous data (like configurations) to be have an effect, as reflashing without erasing made no difference to the brokenness.

After yet another flash erase, and uploading the standard Tasmota image with the correct flash size specification, it worked. And "strange problems" are a good reason for having a more graceful handling, or a strong warning.

ascillato, 2019-02-27:
Good to know that now it is working for you.
You have access to the wiki if you want to add that information. Thanks 

If your issue is solved, please close this issue. Thanks

sfromis, 2019-02-27:
To be clear, in my last sentence I carefully phrased the issue as a suggestion for enhancement, as I was already past the troubleshooting stage. If I just wanted it to work for myself, I'd not have posted here.

As for wiki updates - with my involvement being at the newbie level, I don't see myself as having enough of an overview to amend the documentation.

Jason2866, 2019-02-27:
I have a Pow and a Pow R2 and dont have any issue with flashing a 1 M firmware.
All devices can use 1M firmware. It is mandantory to erase the flash before.
@ascillato2  The info is wrong! POW is working with 1M image!
@sfromis Dont add this to wiki. I will delete it

ascillato2, 2019-02-27:
Sfromis said:

>uploading the standard Tasmota image with the correct flash size specification, it worked. And "strange problems" are a good reason for having a more graceful handling, or a strong warning.

That is ok, but I found that it is already explained in the wiki that all sonoff devices uses 1M firmware.

So, closing issue as it has been answered. 

Thanks.

closed 2019-02-27 by ascillato2

sfromis, 2019-02-27:
Well, of course the standard firmware is indeed 1M, but flashing it without telling esptool that the device has 4M was the difference making it break for me, leading to very weird behavior.

ascillato, 2019-02-27:
Sorry, I flashed all my pows with esptool.py as explained in the (telling 1M memory flash map) and worked fine. Are you using sonoff.bin? Weird things will happen if the memory flash map in esptool.py do not match the firmware size map. And also it has to fit in the flash.

You can flash 1M firmware to a 4M memory device but esptool.py has to match the firmware.

sfromis, 2019-02-27:
Yes, as explained I'm using the completely standard current version of sonoff.bin, which was flashed using the version of esptool.exe shipped with ESPEasy, activated from gui FlashESP8266.exe. As this setup worked perfectly for getting a Sonoff TH16 to run Tasmota, I assumed that I had a good method. The specific esptool.exe self-reports as https://github.com/igrr/esptool-ck v0.4.12

With the POW R2, I got bad results every time in several attempts (trying different stuff with configuration method, and power source), including when clearing flash first. Worked mostly in the initial Sonoff Basic config, but very badly after configuring as the correct Sonoff POW R2.

Resuming later on another computer with Linux, I switched to esptool.py, as I could not see the exact command line from FlashESP8266.exe, again using sonoff.bin 6.4.1, matching the command line flash size to the actual hardware (4M). Well, I did change esptool version, and flash size at the same time, but had no reason to suspect one esptool possibly being worse than the other. My experience was that weirdness persisted until following the hardware 4M, but can't swear on anything now.

It does concern me that the otherwise pretty good Tasmota software can go haywire in such a weird and unclear way, whatever happened under the hood. Was considering switching to ESPHome, but gave Tasmota yet another try first.

ascillato, 2019-02-27:
In the wiki is explained esptool.py because you can set also the DOUT type of flashing. If the flashing software you used was set to QIO or DIO, Tasmota will not work.

sfromis, 2019-02-28:
Well, I wanted to see the command line parameters to esptool.exe exactly because I wanted to check such details. However, the flashing procedure I found specifically targeted flashing Tasmota on ESP8266 based Sonoff devices, and worked excellently for Sonoff TH16, which is based on the same controller as POW R2 (ESP8266EX). Thus, Tasmota was able to work just fine with that way of flashing, and on POW R2 it also worked initially, the breakdown only happened after reconfiguring to module type 43. Other configuration changes before that had no issues.

ESPEasy has a fixed issue about changing to use mode DOUT, but I could not easily see if this also applies when only using the two mentioned .exe files.  https://github.com/letscontrolit/ESPEasy/issues/831

Jason2866, 2019-02-28:
Thank you for your finding that EspEasyFlasher is NOT a recommended tool for flashing Tasmota
We will edit wiki.

zomfg, 2019-12-01:
Currently struggling with this exact issue with an R2. I tried:

- `FlashESP8266.exe` (which is what I used for the initial flash of my T1/TX/Basic/RF Bridge, they work perfectly fine)
- manually using `esptool.exe` with 
  - `esptool.exe -cp COM3 -cz 0x400000 -ca 0x0000 -v -ce`
  - `esptool.exe -cp COM3 -bz 4M -bm dout -ca 0x0000 -cb 115200 -cf tasmota_7111.bin` (this method does not set `Program Flash Size` to 4M)
- `esptool.py` in a linux vm
  - `sudo esptool.py --port /dev/ttyUSB0 erase_flash`
  - `sudo esptool.py  --port /dev/ttyUSB0 write_flash -fs 4MB -fm dout 0x0 tasmota_7111.bin`
  - `sudo esptool.py  --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 tasmota_7111.bin`

but it's still either slow or doing watchdog reboots every few seconds (and in turn, is slow) when configured as Pow R2 and is snappy when switched to Basic

What steps am I missing? Something else can be done?

sfromis, 2019-12-01:
I'm not sure exactly what caused the issue, but currently my Sonoff POW R2 units are working as expected after flashing with:
`esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff-650.bin`
Of course, I erased flash first. (I've not upgraded these to v7 yet).

zomfg, 2019-12-01:
Well... I was just reading [ESPHome wiki](https://esphome.io/components/sensor/cse7766.html), and the part about `UART / RX / 4800bauds` made me remember that I saw a lot of "setting to 4800" in Termite.

And since ESP has only one RX... I unplugged RX/TX of my usb-to-serial, not much changed, then went to config / logging and disabled Serial logging.

Now it seems to be running just as good as my other sonoffs.
Could this be it?

sfromis, 2019-12-01:
I'd expect Tasmota to be setting up serial exclusively for the power monitoring sensor, when selecting the Pow R2 device type, but as my config scripts have "SerialLog 0" for all Sonoff units, I'm not sure if this has made a difference to power monitoring units. Anyway, serial logging really only makes sense for box-open debugging, but I don't remember if I specifically was disabling this when having the issues a long time ago now.

ascillato, 2019-12-02:
In Tasmota, when you select Sonoff Pow R2 as a module, seriallog is set automatically to zero.

zomfg, 2019-12-02:
Mine stayed at 2 for whatever reason (just like my other sonoffs).