arendst/Tasmota #5362: Add Flasher voltage

eladiogalvez, 2019-02-27:
Hi,
In the next post:

https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller

Can you confirm voltage configuration for flash programmer. Is 3.3?

I have this flasher, is compatible?

€ 1,41 | MH-ET en FT232BL USB FTDI USB 3,3 V 5,5 V a TTL serie Módulo adaptador USB Mini puerto USB a TTL módulo FT232RL mejora
https://s.click.aliexpress.com/e/clmexPMH

Possible is interesting add this information to post.

lalo-uy, 2019-02-27:
Yes it is.
Set the jumper to 3.3V

ascillato2, 2019-02-27:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-27 by ascillato2

eladiogalvez, 2019-02-27:
And can you confirm if binary downloads are compatible and which version can be installed?

Minimal?
Basic?

ascillato, 2019-02-27:
The sonoff.bin version will work.

Minimal is an interim firmware when updating by OTA for 1M devices. For more information on that one, please check the wiki.

Basic is a small version that does not have sensors enabled. Aimed for devices with low quality power source. Not your case. 

For more information on different firmware files, please check the releases page. 

If you need further assistant please find us in the Tasmota Support Chat. Thanks