arendst/Tasmota #5363: What ESP board is this LED controller based on?

maxtheking, 2019-02-27:
So a while back i bought a few generic Wi-Fi LED controllers on eBay, i posted images below. They currently use the Tuya cloud, but i want to convert them using tuya-convert so i can control them locally with Home Assistant instead of them reaching out to some chinese Tuya server everytime i turn them on for example. Does anyone know what ESP board they're based on? Can't find any info about them on the internet. 
[image]
[image]

Jason2866, 2019-02-27:
No, and no one can. Because even with the same case there can be different.
Only way is opening the device

ascillato2, 2019-02-27:
Closing this issue as it has been answered.

Please, if you need further assistant please, do not hesitate on asking in the Tasmota Support Chat. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-27 by ascillato2