arendst/Tasmota #5364: Maxico template

davepal, 2019-02-27:
Instead of   {"NAME":"W-US003","GPIO":[255,17,255,255,255,255,255,255,255,22,21,255,255],"FLAG":1,"BASE":18}

I feel it should be -   {"NAME":"W-US003","GPIO":[255,17,255,255,255,255,255,255,255,52,21,255,255],"FLAG":1,"BASE":18}

GPIO 13 set to LED1 instead of Relay2

ascillato2, 2019-02-27:
Hi,

Thanks. 

Please add it to https://blakadder.github.io/templates/

Github issues are for bug reporting. Thanks

closed 2019-02-27 by ascillato2

DavinKD, 2019-02-27:
I did it that way out of preference.  It's actually something that should be discussed, I guess.  I prefer to have my SmartThings device handler control whether the LED comes on our not when the power is turned on.  I could see how others would want it to be automatic.  How should we handle this?

DavinKD, 2019-02-27:
Wanted to tag @blakadder on this as well.

DavinKD, 2019-02-27:
As another example, look at the XS-S005 Template I just uploaded.  It has a blue and a red led.  Which should be LED1, some people may want red, some blue?  Maybe the template documentation could have "alternate" templates with a short description?