arendst/Tasmota #5365: upgrade SONOFF TH16 to get sensor reading

muscaiu, 2019-02-27:
I'm trying to upgrade my SONOFF TH16 to have sensor reading.
This is my current version
[image]

Managed to upgrade it to 6.4.1, and using the file downloaded from https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.4.1/sonoff-sensors.bin to update, but it's always failed.

[image]

What am i doing wrong?

ascillato, 2019-02-27:
You have 2 things. First, you need to go to configuration and choose sonoff TH. Now you have selected sonoff basic.

Then, if you want to update (you don't need that for sonoff TH) you need to use minimal firmware first to free flash memory to then flash the full version.

If this solves your issue, please close it. 

Remember that for support, it is available the Tasmota Support Chat. Github issues are for bug reporting. Thanks

ascillato, 2019-02-27:
The commands issue is because your console is not sending the proper newline characters. Please, use the Tasmota console instead (the one that is inside Tasmota, in the web menu). Thanks

ascillato, 2019-02-27:
Have you tried to choose sonoff TH in the configuration menu?

muscaiu, 2019-02-27:
@ascillato thanks for the quick reply!

I configured another TH10 few weeks ago and that worked pretty easy to get sensor reading.

Tried your advice  to use minimal firmare and then used the sonoff-sensors.bin to upgrade and upgrade worked, but **still the Temp readings are not showing up on the main menu** for this TH16, there is no info on the top.
Any more ideas why that is?

PS: I didn't know that the Tasmota console works same as Termite, used it and setting static ip worked, thanks!

muscaiu, 2019-02-27:
> 
> 
> Have you tried to choose sonoff TH in the configuration menu?

Yes, and now it's showing Sonoff TH Module instead of Basic, but still no temp readings.
[image]

muscaiu, 2019-02-27:
On the previos TH10 i used a SI7021(03) sensor and that showed up even if the sensor was not plugged into the SONOFF.

But this time i'm using a DS18B20  sensor and it looks like the temp is only showing if the sensor it's actually plugged into the SONOFF >< Didn't know that, but i thought hey maybe i try it :)

All good! Thanks again for the tips! Closing

[image]

closed 2019-02-27 by muscaiu