arendst/Tasmota #5366: SYF05 Troubleshooting by 2 equipment

elzershark, 2019-02-27:
absolutely funny. I have 2 lamps. if I set both "correct", "SYF05 module", then one starts after a certain time, always every 40 seconds. Permanently. I put one of them on z.b. Generic and without setting (in generic), then one works again. If I switch back to "SYF05 module", it takes about 10 minutes and one of the two always starts again. every 40 seconds approx

ascillato, 2019-02-27:
Please, could you be so kind on completing the troubleshooting template in order to have more debug information so as to properly help you? Thanks

elzershark, 2019-02-27:
Version 6.4.1.19(sonoff)-2_3_0
https://www.amazon.de/Luminea-Home-Control-WLAN-Birne/dp/B074SZ6VQC/
https://www.elzershark.com/index.php?attachment/22-clipboard-1-jpg/
Right is = SM16716 PWR (142)

I am currently testing without SM16716 PWR (142) setting. It looks like, as soon as I pick that, it starts again.

ascillato2, 2019-02-27:
Please, could you be so kind on completing the troubleshooting template in order to have more debug information so as to properly help you? Thanks

elzershark, 2019-02-27:
That does not work right away. I have to wait for the right time ...

elzershark, 2019-02-27:
without SM16716 PWR (142)

21:07:53 CMD: status 0
21:07:53 MQT: stat/LED2/STATUS = {"Status":{"Module":69,"FriendlyName":["LED2"],"Topic":"LED2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:07:53 MQT: stat/LED2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:51","StartupUTC":"2019-02-27T19:54:02","Sleep":0,"CfgHolder":4617,"BootCount":122,"SaveCount":230,"SaveAddress":"F5000"}}
21:07:53 MQT: stat/LED2/STATUS2 = {"StatusFWR":{"Version":"6.4.1.19(sonoff)","BuildDateTime":"2019.02.25 19:21:46","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:07:53 MQT: stat/LED2/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["elzershark",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
21:07:53 MQT: stat/LED2/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000407","0FDAE794","000783A0","23B617CE","00003BC0"]}}
21:07:53 MQT: stat/LED2/STATUS5 = {"StatusNET":{"Hostname":"Led2","IPAddress":"192.168.178.23","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"CC:50:E3:0F:FA:5B","Webserver":2,"WifiConfig":7}}
21:07:53 MQT: stat/LED2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.24","MqttPort":1884,"MqttClientMask":"LED2","MqttClient":"LED2","MqttUser":"Sonoff","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:07:53 MQT: stat/LED2/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 27 20:07:53 2019","Local":"Wed Feb 27 21:07:53 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:36","Sunset":"18:29"}}
21:07:53 MQT: stat/LED2/STATUS10 = {"StatusSNS":{"Time":"2019-02-27T21:07:53"}}
21:07:53 MQT: stat/LED2/STATUS11 = {"StatusSTS":{"Time":"2019-02-27T21:07:53","Uptime":"0T00:13:51","Vcc":1.871,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":838,"POWER":"ON","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","Channel":[0,0,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"elzershark","BSSId":"9C:C7:A6:8E:11:3E","Channel":6,"RSSI":76,"LinkCount":1,"Downtime":"0T00:00:04"}}}
21:09:16 MQT: tele/LED2/STATE = {"Time":"2019-02-27T21:09:16","Uptime":"0T00:15:14","Vcc":1.904,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","Channel":[0,0,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"elzershark","BSSId":"9C:C7:A6:8E:11:3E","Channel":6,"RSSI":76,"LinkCount":1,"Downtime":"0T00:00:04"}}

ascillato, 2019-02-27:
Without the requested information, we can't help you. Sorry

ascillato, 2019-02-27:
Please, at least share the status 0 of both devices and also the output of the console

elzershark, 2019-02-27:
00:00:00 CFG: aus Flash geladen am F8, zählen 251
00:00:00 SRC: Restart
00:00:00 Projekt sonoff LED2 Version 6.4.1.19(sonoff)-2_3_0
00:00:00 WIF: Prüfe Verbindung...
00:00:00 WIF: Verbindungsversuch...
00:00:00 WIF: Repariere Problem #2186
00:00:00 WIF: verbinden mit AP1 elzershark in Modus 11N wie Led2...
00:00:01 WIF: Prüfe Verbindung...
00:00:01 WIF: Verbindungsversuch...
00:00:02 WIF: Prüfe Verbindung...
00:00:02 WIF: Verbindungsversuch...
00:00:03 WIF: Prüfe Verbindung...
00:00:03 WIF: Verbindungsversuch...
00:00:04 WIF: Prüfe Verbindung...
00:00:04 WIF: verbunden
00:00:04 HTP: Web-Server aktiv bei Led2 mit IP-Adresse 192.168.178.23
00:00:05 MQT: Verbindungsversuch...
00:00:05 MQT: verbunden
00:00:05 MQT: tele/LED2/LWT = Online (beibehalten)
00:00:05 MQT: cmnd/LED2/POWER = 
00:00:05 MQT: abonniere cmnd/LED2/#
00:00:05 MQT: abonniere cmnd/sonoffs/#
00:00:05 MQT: abonniere cmnd/LED2_fb/#
00:00:05 MQT: tele/LED2/INFO1 = {"Module":"SYF05","Version":"6.4.1.19(sonoff)","FallbackTopic":"cmnd/LED2_fb/","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/LED2/INFO2 = {"WebServerMode":"Admin","Hostname":"Led2","IPAddress":"192.168.178.23"}
00:00:05 MQT: tele/LED2/INFO3 = {"RestartReason":"Power on"}
00:00:05 MQT: stat/LED2/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/LED2/POWER = ON
00:00:06 APP: (UTC) Wed Feb 27 20:48:05 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
21:48:08 APP: Anzahl Startvorgänge 128
21:48:09 CFG: in Flash gespeichert am F7, zählen 252, Bytes 3584
21:48:12 CMD: status 0
21:48:12 SRC: WebConsole from 192.168.178.22
21:48:12 RSL: empfangenes topic /status, Datengröße 1, Daten 0
21:48:12 RSL: Gruppe 0, Index 1, Befehl STATUS, Daten 0
21:48:12 MQT: stat/LED2/STATUS = {"Status":{"Module":69,"FriendlyName":["LED2"],"Topic":"LED2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:48:12 MQT: stat/LED2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T00:00:14","StartupUTC":"2019-02-27T20:47:58","Sleep":0,"CfgHolder":4617,"BootCount":128,"SaveCount":252,"SaveAddress":"F7000"}}
21:48:12 MQT: stat/LED2/STATUS2 = {"StatusFWR":{"Version":"6.4.1.19(sonoff)","BuildDateTime":"2019.02.25 19:21:46","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:48:12 MQT: stat/LED2/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["elzershark",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
21:48:12 MQT: stat/LED2/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000407","0FDAE794","000783A0","23B617CE","00003BC0"]}}
21:48:12 MQT: stat/LED2/STATUS5 = {"StatusNET":{"Hostname":"Led2","IPAddress":"192.168.178.23","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"CC:50:E3:0F:FA:5B","Webserver":2,"WifiConfig":7}}
21:48:12 MQT: stat/LED2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.24","MqttPort":1884,"MqttClientMask":"LED2","MqttClient":"LED2","MqttUser":"Sonoff","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:48:12 MQT: stat/LED2/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 27 20:48:12 2019","Local":"Wed Feb 27 21:48:12 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:36","Sunset":"18:29"}}
21:48:12 MQT: stat/LED2/STATUS10 = {"StatusSNS":{"Time":"2019-02-27T21:48:12"}}
21:48:12 MQT: stat/LED2/STATUS11 = {"StatusSTS":{"Time":"2019-02-27T21:48:12","Uptime":"0T00:00:14","Vcc":2.782,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":74,"POWER":"ON","Dimmer":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"elzershark","BSSId":"9C:C7:A6:8E:11:3E","Channel":6,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
21:48:13 MQT: tele/LED2/STATE = {"Time":"2019-02-27T21:48:13","Uptime":"0T00:00:15","Vcc":2.782,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":109,"POWER":"ON","Dimmer":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"elzershark","BSSId":"9C:C7:A6:8E:11:3E","Channel":6,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:48:23 WIF: Prüfe Verbindung...
21:48:23 WIF: verbunden

*********************************

00:00:00 CFG: aus Flash geladen am F7, zählen 252
00:00:00 SRC: Restart
00:00:00 Projekt sonoff LED2 Version 6.4.1.19(sonoff)-2_3_0
00:00:00 WIF: Prüfe Verbindung...
00:00:00 WIF: Verbindungsversuch...
00:00:00 WIF: Repariere Problem #2186
00:00:00 WIF: verbinden mit AP1 elzershark in Modus 11N wie Led2...
00:00:01 WIF: Prüfe Verbindung...
00:00:01 WIF: Verbindungsversuch...
00:00:02 WIF: Prüfe Verbindung...
00:00:02 WIF: Verbindungsversuch...
00:00:03 WIF: Prüfe Verbindung...
00:00:03 WIF: Verbindungsversuch...
00:00:04 WIF: Prüfe Verbindung...
00:00:04 WIF: verbunden
00:00:04 HTP: Web-Server aktiv bei Led2 mit IP-Adresse 192.168.178.23
00:00:05 MQT: Verbindungsversuch...
00:00:05 MQT: verbunden
00:00:05 MQT: tele/LED2/LWT = Online (beibehalten)
[... 104 lines cut ...]
[... cut ...]
21:50:06 MQT: stat/LED2/STATUS10 = {"StatusSNS":{"Time":"2019-02-27T21:50:06"}}
21:50:06 MQT: stat/LED2/STATUS11 = {"StatusSTS":{"Time":"2019-02-27T21:50:06","Uptime":"0T00:00:26","Vcc":2.788,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":106,"POWER":"ON","Dimmer":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"elzershark","BSSId":"9C:C7:A6:8E:11:3E","Channel":6,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:04"}}}

netpok, 2019-02-28:
Okay first guess:
If I see correctly this is an RGBW bulb, based on your logs it restarts with the reason `power on`.
On the logs I see all channels are turned to max, most of these bulbs are designed to work only in RGB color or White mode, so they have a power supply capable of providing ~10W. Turning on all channels are leading to overload and thermal reset of these supplies, because RGB leds are around 3*2-3W and White leds are around 7-10W. 

Please test the stability of the product with only turning on the white channel to 100% and test it with only turning the RGB channels to 100 percent.

Commands for these:
white: `Color 000000ff00`
RGB: `Color ffffff0000`

Also next time please use triple bacticks ` ``` ` for providing logs like this:
[image]

For results like this
```
As they lead to a nicely formatted readable log
like this.

And can display very long lines without word wrapping, like thiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiis.
```

Jason2866, 2019-02-28:
@netpok We will get now and than issue entry for this https://github.com/arendst/Sonoff-Tasmota/pull/4991#issuecomment-459831215
Do you have a idea how we can avoid?

blakadder, 2019-02-28:
Answered with comment by [netpok](https://github.com/arendst/Sonoff-Tasmota/issues/5366#issuecomment-468211063). Please be aware of the capablities of your device before blaming Tasmota!

closed 2019-02-28 by blakadder

elzershark, 2019-02-28:
> Commands for these:
> white: `Color 000000ff00`
> RGB: `Color ffffff0000`

Absolutely perfect. many thanks. That's exactly what it was. Thank you very much 😃