arendst/Tasmota #5367: HX711 store sensor value in eeprom for power down and restore value on boot

WilldabeastHA, 2019-02-28:
I would like tasmota to store a value from the HX711 senor, power down the D1 Mini, then recall that value as it boots back up.

Currently when the power shuts off the sensor/tasmota forgets how much is on the scale, when it reboots it starts at zero, with the original weight on it. 

I'm using this to weigh my propane tank for my gas grill. It's battery powered so when I push a button, I want it to turn on and send the weight value, then turn off. Because this is a remote location there is no option to have this plugged into the wall. 

Thank you!

ascillato, 2019-02-28:
Hi,

This behaviour can be easily achieved using rules. 

After you power your device, it connects to wifi, to mqtt and then it sends the weight in kg in the tele message, right? 

In that case (using latest Tasmota version v6.4.1.19) you can go to the console and type:

```
Rule1 1 
Rule1 on tele-HX711#weight do mem1 %value%*1000 endon on system#boot do sensor34 6 %mem1% endon
```

Please, tell us if this solves your issue. In that case, remember to close it. Thanks.

WilldabeastHA, 2019-02-28:
I'm not sure if it's working correctly. 

`10:37:35 RUL: SYSTEM#BOOT performs "sensor34 6 4*1000"
10:37:35 MQT: stat/test1/RESULT = {"Sensor34":{"WeightRef":3628,"WeightCal":24,"WeightMax":20000,"WeightItem":4.0}}
10:37:43 MQT: tele/test1/STATE = {"Time":"2019-02-28T10:37:43","Uptime":"0T00:00:15","Vcc":3.091,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"xxxxxxxx","BSSId":"FC:EC:DA:B1:75:43","Channel":1,"RSSI":50}}
10:37:43 MQT: tele/test1/SENSOR = {"Time":"2019-02-28T10:37:43","Switch1":"ON","HX711":{"Weight":0}}
10:37:43 RUL: TELE-HX711#WEIGHT performs "mem1 0*1000"
10:37:43 MQT: stat/test1/RESULT = {"Mem1":"0*1000"}`

ascillato, 2019-02-28:
Are you using latest Tasmota version v6.4.1.19?

WilldabeastHA, 2019-02-28:
I just update but now it gets a fatal error and boot loop 
`{"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x4023ef3c epc2:0x00000000 epc3:0x00000000 excvaddr:0x40274011 depc:0x00000000"}`

ascillato, 2019-02-28:
So, something went wrong in your update.
Please, erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins.

WilldabeastHA, 2019-02-28:
flash erased and uploaded with precompiled bin. When I enable the rule I get a fatal error and boot loop `{"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x40204fe1 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}`

WilldabeastHA, 2019-02-28:
it has something to do with the *1000 - If I remove that from the rule it works. I think the problem is that the actual scale resets to 0 on a reset. you can see that right after the item weight sets to 4 the next tele shows it at 0. 

`Rule1 on tele-HX711#weight do mem1 %value% endon on system#boot do sensor34 6 %mem1% endon

12:03:49 MQT: stat/test1/RESULT = {"Sensor34":{"WeightRef":3628,"WeightCal":25,"WeightMax":20000,"WeightItem":4.0}}

12:04:57 MQT: tele/test1/SENSOR = {"Time":"2019-02-28T12:04:57","HX711":{"Weight":0}}`

ascillato, 2019-02-28:
Ok, try the following please:

```
Rule1 1 
Rule1 on tele-HX711#weight>0 do mem1 %value% endon on system#boot do sensor34 6 %mem1% endon
```

WilldabeastHA, 2019-02-28:
The rule works and the weight gets assigned but the HX711 sensor reads zero when it powers back on, with the weight still on the scale. 

`12:49:26 RUL: SYSTEM#BOOT performs "sensor34 6 4.000"
12:49:26 MQT: stat/test1/RESULT = {"Sensor34":{"WeightRef":3628,"WeightCal":25,"WeightMax":20000,"WeightItem":4.0}}
12:49:34 MQT: tele/test1/SENSOR = {"Time":"2019-02-28T12:49:34","HX711":{"Weight":0}}`

[image]

ascillato, 2019-02-28:
Ok, that is because `sensor34 6` command expects decigrams and not kg. 

Please try the following:

```
Rule1 1 
Rule1 on tele-HX711#weight>0 do backlog var1 %value%; mult1 10000; event savedata endon on event#savedata do mem1 %var1% endon on mqtt#connected do sensor34 6 %mem1% endon
```

WilldabeastHA, 2019-02-28:
on reboot the weight from the HX711 shows 0. `13:15:29 MQT: tele/test1/SENSOR = {"Time":"2019-02-28T13:15:29","HX711":{"Weight":0}}`

Now that I'm thinking about it. Saving the known value of weight won't matter. It's going to turn on send a weight and turn off. While it's off the gas will be used and the tank will be lighter that its previously known weight. So if its turning on after the gas was used its setting its weight back to the previous know weight, which means its always going to think its 4kg. 

Does this need to be set up in a different way? is there a way for the HX711 to turn on and know there is 4kg on it based on the calibrated numbers? or is the only way this works is when the sensor turns on there has to be no weight on it, and then weight placed on it, to correctly identify the weight.

ascillato, 2019-02-28:
> Now that I'm thinking about it. Saving the known value of weight won't matter. It's going to turn on send a weight and turn off. While it's off the gas will be used and the tank will be lighter that [...]

Ups. You are totally right.

So, that means that as the HX711 requires to measure a differential weight, seems that is not possible to be used as you want.

If you see, any weight meter need to start without anything on the measuring plate and then you can put things to measure.

So, you are right and can't be used that way. Sorry.

WilldabeastHA, 2019-02-28:
Thank you for your help!

closed 2019-02-28 by WilldabeastHA

ascillato2, 2019-02-28:
If you need further assistant please do not hesitate on asking here or find us in the Tasmota Support Chat. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

WilldabeastHA, 2019-02-28:
ascillato what if i were to keep the HX711 powered on all the time off the vcc and ground terminals of that chip and have the D1 mini power down and up?

ascillato2, 2019-02-28:
It is the same. Sorry.

joba-1, 2019-02-28:
just for the record: that it does not work is a software issue, not
hardware, means it is solvable: dont calibrate at startup, but on request
(button press, mqtt message, whatever) and permanently store the
calibration value for later use (in flash or retained mqtt).
-- 
iphone so i typo

ascillato, 2019-02-28:
If you have a workaround to make it doable, please explain

WilldabeastHA, 2019-02-28:
@joba-1 If I don't calibrate it then the weight is incorrect?

joba-1, 2019-03-01:
I don't have a readymade workaround and I did not look at the code (changes
could be needed in tasmota or the driver), but the sensor is just a bunch
of cleverly arranged variable resistors (wheatstone bridge) and an ad
converter measuring the varying voltage differences over these resistors.
There is nothing that varys between power cycles. Calibration is usually
done to accommodate for varying weight of your setup (scale + container).
But this does not change in your case. So calibrate once with an empty
container should be enough.
Ok, it will drift a bit depending on ambient temperature and maybe a bit
long term but that should not be too much to make it impossible to detect
an empty container. Just repeat the calibration at times and see how much
the values change with it.

WilldabeastHA, 2019-03-09:
@joba-1 how would I go about changing the tasmota code to accommodate this?

joba-1, 2019-03-09:
Have you tried this (from the sensor code)? To me that sounds like it would already do what you want.
```
/*
  xsns_34_hx711.ino - HX711 load cell support for Sonoff-Tasmota
  ...
*/

#ifdef USE_HX711
/*********************************************************************************************\
 * HX711 - Load cell as used in a scale
 *
 * Source: Sparkfun and https://github.com/bogde/HX711
 *
 * To reset the scale:
 * - Execute command Sensor34 1
 *
 * To calibrate the scale perform the following tasks:
 * - Set reference weight once using command Sensor34 3 <reference weight in gram>
 * - Remove any weight from the scale
 * - Execute command Sensor34 2 and follow messages shown
\*********************************************************************************************/
...
```

WilldabeastHA, 2019-03-15:
@joba-1 yes I have used the settings and calibration you have shared above. I can get the scale to function properly while its powered on. The issue is when the power is cut or the device restarts with the weight on the scale, it starts back at 0. It doesn't remember that there was a weight on the scale when it boots up.

thucar, 2019-05-12:
It seems to me that the Sensor34 6 command is not functioning at all, or is not functioning as expected. If I issue a sensor34 6 xxxx command, then the value red by the sensor never changes.
For instance:
I have my scale calibrated and it has a known weight of 5kg sitting on it. It shows 5kg. I issue a Sensor34 6 xxxx command to change WeightItem value. The sensor still displays 5kg

```
07:39:48 MQT: tele/sonoffk/SENSOR = {"Time":"2019-05-12T07:39:48","HX711":{"Weight":5,"Count":56}}
07:40:00 CMD: sensor34 6
07:40:00 RSL: Received Topic /sensor34, Data Size 1, Data 6
07:40:00 RSL: Group 0, Index 34, Command SENSOR, Data 6
07:40:00 MQT: stat/sonoffk/RESULT = {"Sensor34":{"WeightRef":5000,"WeightCal":21,"WeightMax":100000,"WeightItem":90.0}}
07:40:42 CMD: sensor34 6 0
07:40:42 RSL: Received Topic /sensor34, Data Size 3, Data 6 0
07:40:42 RSL: Group 0, Index 34, Command SENSOR, Data 6 0
07:40:42 MQT: stat/sonoffk/RESULT = {"Sensor34":{"WeightRef":5000,"WeightCal":21,"WeightMax":100000,"WeightItem":0.0}}
07:40:43 CFG: Saved to flash at F4, Count 90, Bytes 3584
07:45:08 MQT: tele/sonoffk/SENSOR = {"Time":"2019-05-12T07:45:08","HX711":{"Weight":5}}
```

darkdevil373, 2021-05-02:
The command Sensor34 7 won‘t work. Everytime i Power off an on my nodeMCU, the weight stays on 0 Even the weight is on it

lalo-uy, 2021-05-03:
Soring the sensor value is not the correct solution, since the weight may have changed while device is off. On calibration the offset shuld be saved and restored on power on.