arendst/Tasmota #5368: Help with custom code (timeperiod) needed

passuff, 2019-02-28:
I would like to read temperatures from TH16 more often than 10s. Therefor I would like to change the minimal timeperiod from 10 to 2s. I thought I can do it with tht following line indecode-config.py:

`    'tele_period':                  ('<H',  0x2F8,       (None, '0 <= $ <= 1 or 2 <= $ <= 3600',('MQTT',       '"TelePeriod {}".format($)')) ),
`

I did not work, 10 is still the minimum accepted value. Maybe someone can support me.

ascillato, 2019-02-28:
Yes, sorry. It is limited to 10 seconds. It can't be done without a major redesign of Tasmota.

ascillato2, 2019-02-28:
Closing this issue as it has been answered.

If you need further assistant please find us in the Tasmota Support Chat. Github issues are for bug reporting. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-28 by ascillato2