arendst/Tasmota #5370: Smart Wifi RGB led controller Spider Z with nuvoton chip

arcidodo, 2019-02-28:
I order an sonoff led strip flashed with Tasmota. 

link:
https://www.aliexpress.com/item/SONOFF-L1-Smart-LED-Light-Strip-Dimmable-Waterproof-WiFi-Flexible-RGB-Strip-Lights-Work-with-Alexa/32960596437.html?channel=twinner

if i select the module PS-16-DZ and the the on/off works with only the red color also dimming is working correctly. 

how can i get the RGB function to work? :-)

[image]
[image]
[image]

ascillato, 2019-02-28:
Hi,

You need to find out where the gpios are connected and then select generic module and set what you have found.

Then, when you have it working you can make a new template for that device. (information on this can be found in the wiki) 

If you want to share your template, you can send it as explained in the wiki.

If you need further assistant please find us in the Tasmota Support Chat. Thanks

ascillato2, 2019-03-06:
Please, when you have time, please tell us if you could make it work. Thanks

closed 2019-03-06 by ascillato2

jakass202, 2019-03-22:
Can’t find the GPIO0 to put it in flash mode

eford321, 2019-04-02:
Flashed by holing GPIO0 low at the chip.  Still not able to find working GPIOs for the controller.

arcidodo, 2019-04-02:
@eford321 same  here!

eford321, 2019-04-02:
Really strange.  I am unable to get the controller to respond to any GPIO settings.  I decided to test for the IR GPIO and found the remote still functions as if the original software is still running.  All remote functions work and the Sonoff-Tasmota console does not show any activity other than booting.

eford321, 2019-04-02:
Found out why.  There is a Nuvoton N76E003AT20 processor on the board.  The ESP8285 must be communicating via i2c or serial to the Nuvoton, which controls the PWM for LEDs.

reef-actor, 2019-04-09:
This seems to be a rebranded Sonoff L1, discussion in [issue 4935](https://github.com/arendst/Sonoff-Tasmota/issues/4935)

Wriman, 2020-04-25:
Hi! Can you send me the homeasisstant config?

reef-actor, 2020-04-27:
@Wriman see https://github.com/arendst/Tasmota/issues/4935#issuecomment-539700171