arendst/Tasmota #5371: Copy existing GPIO selections to the configure template menu - Feature Request

digiblur, 2019-02-28:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Not really.

**Describe the solution you'd like**
In the configure template menu create a button that will copy the existing "Configure Module" selections over to the template screen.

**Describe alternatives you've considered**
Pulling up both screens and going back and forth introducing human error.

**Additional context**
This will allow a user to easily figure out the GPIO pins for his/her device, then easily create a template to send to @blakadder 's Wiki & Templates.

netpok, 2019-03-01:
Nice idea, until it's implemented you can just use `template 0` command to copy current selected module to the template.

digiblur, 2019-03-01:
Oh cool I didn't know the template 0 would do that, I'll check that out.  

Define what you mean by web?  I'm just talking about a simple button on the Tasmota Web GUI in the Configure Template screen that pulls over what is currently configured and running on the device in the configure module screen.  

Basically like this, a user gets a new plug, figures out all the GPIO pin settings and tests their plug.  The user then heads over to the configure template screen hits copy and it pulls over the info.   And thinking this through, if there was a text box with the template JSON that would be great as well.  They could then just copy that info into their clipboard and then they would send it themselves and other info to have it submitted for user templates for others to use.

netpok, 2019-03-01:
I meant web interface, the copying of the currently selected configuration is doable.

About the JSON if I remember correctly there was a discussion earlier, in the end the decision was that it should be placed on the `Configure Other` tab.

digiblur, 2019-03-05:
I tried the template 0 command but it didn't copy things over.  It just spits out what is in the template which was not configured of course.

netpok, 2019-03-06:
Is there a chance that your module is already configured to use template? Because `template 0` copies the currently selected module, on the other hand you can copy any existing module via `template <module-id>`.

digiblur, 2019-03-06:
No, template defined. This is the configure module menu.  

[image]

Go to console and typed template 0 and only this happens.

....RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}

netpok, 2019-03-06:
Oh, I get it now, `template 0` command actually copies the selected template, not the custom GPIO configuration. This is designed this way so there is currently no way to do what you want as far as I know (except manually copying them).

arendst, 2019-03-08:
Let's conclude:

You want the module changes made by a user to merge with the current active template (which doesn't have to be a user template) and store it as the user template?

For now I'll add an option 255 to command template which will do just that. See what confusion comes out of it...

digiblur, 2019-03-08:
@arendst basically yes.   Scenario time to make it easy.  

I have a brand new RGBW LED controller.  I sit and figure out where the 4 PWM channels are at to map it correctly plus the button on the controller.  I want to submit it as a user template to put in the Wiki for others to enjoy.  

For now as it stands, the only way I can do this is to pull up the current Module Config screen on one monitor, then go to the configure template on another screen and click all the drop downs to make this new template.  It would be nice to have a console command or button that would transfer over the current module config to the template config screen.  

An easy way to generate a template config off of the module configuration that you had to figure out.  It also removes the human error aspect of things manually transferring the data.  

Thanks for looking at this!  Just ideas to make it easier for people to make templates for the wiki.

arendst, 2019-03-08:
How about this:

```
17:35:49 CMD: template
17:35:49 SRC: Serial
17:35:49 RSL: Group 0, Index 1, Command TEMPLATE, Data 
17:35:49 MQT: stat/wemos2/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
17:35:53 HTP: Configure Module
17:35:56 HTP: Configuration
17:36:04 CMD: template 255
17:36:04 SRC: Serial
17:36:04 RSL: Group 0, Index 1, Command TEMPLATE, Data 255
17:36:04 MQT: stat/wemos2/RESULT = {"NAME":"Merged","GPIO":[17,255,29,255,255,255,255,255,5,6,255,255,255],"FLAG":1,"BASE":18}
17:36:05 CFG: Saved to flash at F9, Count 52, Bytes 3584
```
It also changes the template name to Merged. I bet someone is against that...

digiblur, 2019-03-08:
Looks good!  The name is easy to change ;)

Jason2866, 2019-03-08:
Nice! Templating is now very very easy 👍

DavinKD, 2019-03-08:
I like it!

blakadder, 2019-03-08:
Now if only we can get all the templating under one configuration menu

digiblur, 2019-03-08:
Thanks for this!

closed 2019-03-08 by digiblur

fragolinux, 2019-03-08:
thanks, very nice addition, i configured some new templates and missed such feature, having to open 2 browser windows to copy from active to template page just to share them...