arendst/Tasmota #5372: rule trigger POWER1#STATE

baha00, 2019-02-28:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : SONOFF S31* Appuche chinese but works wit 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - 

```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-02-28:
That is because the rule is triggered as soon as the event occurs and before all the sensors values are updated. 

Please add a delay. 

```
Rule1 1 
Rule1 on Power1#state do backlog delay 1; status 8 endon
```

baha00, 2019-02-28:
This sorted out!! thanks, another question, is it possible to implement that rule in Firmware? so no need to setting it up to be in by default?

closed 2019-02-28 by baha00

ascillato, 2019-02-28:
It is very specific to this case. Is like adding in the code that "If a power state has changed, update all values before executing other commands". That would add a delay on the answer for all use cases.

May be Theo (@arendst) has an idea for that.

baha00, 2019-02-28:
@ascillato no, my bad expression, I didnt mean to add it on default, but is there option before building firmware to add rules ? Like wifi ap etc

ascillato, 2019-02-28:
Ah, ok.

No, there isn't yet.

markross42, 2019-11-16:
Hi @ascillato, 
This 

Rule1 on Power1#state do backlog delay 20; status 8 endon
sorted out my problem - a bit, :) Still, this does not trigger telematic data send by power state change.

THX