arendst/Tasmota #5373: When using the command EnergyReset3 0, Tasmota will not put the actual date and time to TotalStartTime.

wongnam, 2019-03-01:
When using the command `EnergyReset3 0`, Tasmota will not put the actual date and time to `TotalStartTime`.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_ https://github.com/arendst/Sonoff-Tasmota/issues/3971
- [x] _Hardware used_ :  Wemos D1 mini and PZEM004T
- [x] _Development/Compiler/Upload tools used_ : Self-Complied, VSC + PlatformIO IDE
- [x] _If a pre-compiled release or development binary was used, which one?_ : No.
- [x] _You have tried latest release or development binaries?_ : Version 6.4.1.18 Core-2.4.2
- [x] _Provide the output of command_``status 0`` :
```
12:48:40 MQT: tele/energy1/STATE = {"Time":"2019-03-01T12:48:40","Uptime":"0T15:42:49","Vcc":2.959,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":44,"LinkCount":4,"Downtime":"0T00:00:16"}}
12:48:40 MQT: tele/energy1/SENSOR = {"Time":"2019-03-01T12:48:40","ENERGY":{"TotalStartTime":"2019-02-26T10:14:20","Total":13.797,"Yesterday":4.916,"Today":8.881,"Period":6,"Power":2109,"ApparentPower":2960,"ReactivePower":2077,"Factor":0.71,"Voltage":227,"Current":13.070}}
12:48:46 CMD: EnergyReset3 0
12:48:46 MQT: stat/energy1/RESULT = {"EnergyReset":{"Total":8.884,"Yesterday":4.916,"Today":8.884}}
12:48:50 MQT: tele/energy1/STATE = {"Time":"2019-03-01T12:48:50","Uptime":"0T15:42:59","Vcc":2.959,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":46,"LinkCount":4,"Downtime":"0T00:00:16"}}
12:48:50 MQT: tele/energy1/SENSOR = {"Time":"2019-03-01T12:48:50","ENERGY":{"TotalStartTime":"2019-02-26T10:14:20","Total":8.887,"Yesterday":4.916,"Today":8.887,"Period":6,"Power":2112,"ApparentPower":2943,"ReactivePower":2050,"Factor":0.72,"Voltage":226,"Current":13.030}}
12:48:57 CMD: status 0
12:48:57 MQT: stat/energy1/STATUS = {"Status":{"Module":0,"FriendlyName":["Energy Monitor"],"Topic":"energy1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":7,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:48:57 MQT: stat/energy1/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T15:43:06","StartupUTC":"2019-02-28T14:05:51","Sleep":50,"CfgHolder":4617,"BootCount":62,"SaveCount":841,"SaveAddress":"FB000"}}
12:48:57 MQT: stat/energy1/STATUS2 = {"StatusFWR":{"Version":"6.4.1.18(sonoff)","BuildDateTime":"2019-02-25T00:47:21","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
12:48:57 MQT: stat/energy1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-2","Wong"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
12:48:57 MQT: stat/energy1/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F282794","240003A0","000000CE","000013C0"]}}
12:48:57 MQT: stat/energy1/STATUS5 = {"StatusNET":{"Hostname":"energy1-0307","IPAddress":"192.168.12.126","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"DC:4F:22:58:41:33","Webserver":2,"WifiConfig":4}}
12:48:57 MQT: stat/energy1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_584133","MqttUser":"admin","MqttType":1,"MqttCount":16,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:48:57 MQT: stat/energy1/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 01 05:48:57 2019","Local":"Fri Mar 01 12:48:57 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00","Sunrise":"06:07","Sunset":"18:02"}}
12:48:57 MQT: stat/energy1/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:48:57 MQT: stat/energy1/STATUS10 = {"StatusSNS":{"Time":"2019-03-01T12:48:57","ENERGY":{"TotalStartTime":"2019-02-26T10:14:20","Total":8.890,"Yesterday":4.916,"Today":8.890,"Power":2075,"ApparentPower":2931,"ReactivePower":2070,"Factor":0.71,"Voltage":226,"Current":12.970}}}
12:48:57 MQT: stat/energy1/STATUS11 = {"StatusSTS":{"Time":"2019-03-01T12:48:57","Uptime":"0T15:43:06","Vcc":2.959,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":48,"LinkCount":4,"Downtime":"0T00:00:16"}}}
12:49:00 MQT: tele/energy1/STATE = {"Time":"2019-03-01T12:49:00","Uptime":"0T15:43:09","Vcc":2.984,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":42,"LinkCount":4,"Downtime":"0T00:00:16"}}
12:49:00 MQT: tele/energy1/SENSOR = {"Time":"2019-03-01T12:49:00","ENERGY":{"TotalStartTime":"2019-02-26T10:14:20","Total":8.892,"Yesterday":4.916,"Today":8.892,"Period":5,"Power":2097,"ApparentPower":2931,"ReactivePower":2048,"Factor":0.72,"Voltage":226,"Current":12.970}}
```

**To Reproduce**
using the command `EnergyReset3 0`

**Expected behavior**
It should put in that variable the actual date and time when using the command `EnergyReset3 0`.

**Screenshots**
[image]

**(Please, remember to close the issue when the problem has been addressed)**

netpok, 2019-03-01:
Interesting choice of command name, but your issue looks valid. I will check this out later.

Jason2866, 2019-03-01:
can confirm this bug
```
11:04:47 MQT: tele/sonoff-16CE26/SENSOR = {"Time":"2019-03-01T11:04:47","ENERGY":{"TotalStartTime":"2018-11-02T16:55:59","Total":94.493,"Yesterday":1.069,"Today":0.462,"Period":6,"Power":76,"ApparentPower":102,"ReactivePower":67,"Factor":0.75,"Voltage":233,"Current":0.436}}
11:09:08 CMD: EnergyReset3 0
11:09:08 MQT: stat/sonoff-16CE26/RESULT = {"EnergyReset":{"Total":0.467,"Yesterday":1.069,"Today":0.467}}
11:09:20 CMD: EnergyReset2 0
11:09:20 MQT: stat/sonoff-16CE26/RESULT = {"EnergyReset":{"Total":0.468,"Yesterday":0.000,"Today":0.468}}
11:09:38 CMD: EnergyReset1 0
11:09:38 MQT: stat/sonoff-16CE26/RESULT = {"EnergyReset":{"Total":0.468,"Yesterday":0.000,"Today":0.000}}
11:09:47 MQT: tele/sonoff-16CE26/STATE = {"Time":"2019-03-01T11:09:47","Uptime":"4T20:33:36","Vcc":3.499,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":100,"LinkCount":3,"Downtime":"0T00:00:17"}}
11:09:47 MQT: tele/sonoff-16CE26/SENSOR = {"Time":"2019-03-01T11:09:47","ENERGY":{"TotalStartTime":"2018-11-02T16:55:59","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":75,"ApparentPower":122,"ReactivePower":97,"Factor":0.61,"Voltage":235,"Current":0.521}}
```

arendst, 2019-03-01:
Working on it.

arendst, 2019-03-01:
As always there is more to it.

When ``EnergyReset3 0`` is executed the total energy is reset but the daily energy is not hence the TotalStartTime should be set to midnight unless daily energy is zero where it will be set to current time.

When ``EnergyReset1 0`` is executed the daily energy is reset but if there is still a total energy present it will not change TotalStartTime. When on the other hand the total energy is zero and the daily energy is also set to zero it will set TotalStartTime to current time. If total energy is zero and daily energy is set to a non-zero value the TotalStartTime is again not changed.

arendst, 2019-03-01:
Give it a go and remember that resetting total energy with command ``EnergyReset3 0`` just resets the  total energy counter initialized at midnight; if there is daily energy present it is also shown as total energy which makes totally sense for me.

Jason2866, 2019-03-01:
@arendst  works again

wongnam, 2019-03-01:
@arendst I saw it changed now.  and will see it change the total energy counter initialized at midnight.
Thanks.

```
23:25:38 MQT: tele/energy1/STATE = {"Time":"2019-03-01T23:25:38","Uptime":"0T00:01:24","Vcc":2.985,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:25:38 MQT: tele/energy1/SENSOR = {"Time":"2019-03-01T23:25:38","ENERGY":{"TotalStartTime":"2019-02-26T10:14:20","Total":19.306,"Yesterday":4.916,"Today":19.306,"Period":3,"Power":1325,"ApparentPower":1399,"ReactivePower":449,"Factor":0.95,"Voltage":229,"Current":6.120}}
23:25:47 CMD: energyreset3 0
23:25:47 MQT: stat/energy1/RESULT = {"EnergyReset":{"Total":19.310,"Yesterday":4.916,"Today":19.310}}
23:25:48 MQT: tele/energy1/STATE = {"Time":"2019-03-01T23:25:48","Uptime":"0T00:01:34","Vcc":2.960,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:25:48 MQT: tele/energy1/SENSOR = {"Time":"2019-03-01T23:25:48","ENERGY":{"TotalStartTime":"2019-03-01T00:00:00","Total":19.310,"Yesterday":4.916,"Today":19.310,"Period":4,"Power":1331,"ApparentPower":1386,"ReactivePower":388,"Factor":0.96,"Voltage":228,"Current":6.070}}
```

wongnam, 2019-03-01:
worked as expect. Thank you.

closed 2019-03-01 by wongnam