arendst/Tasmota #5374: Question:  Has anyone tested Tasmota on "NPL dorser-1" door/window sensor?

cbmboy, 2019-03-01:
Hello Tasmota community!
I have been lurking around GibHub and Home Assistant sites for a few months in preparation for my first adventure into Home Automation.  To keep on a budget, I ordered and tested some 433 MHz door sensors using an RTL-SDR USB device, Unfortunately it appears I have a significant amount of noise near my house (both other devices broadcasting every second, and other unknown sources).  So I am looking to try another approach.

I found some relatively cheap window/door sensors that use WiFi that I have ordered and am hoping that I can get Tasmota flashed onto them.  I did a lot of searching, and have only found one site that lists a brand.  It shows them as "NPL dorser-1" but I am not confident this is actually a model or brand.

[https://www.amazon.com/NPL-Sensor-Wireless-Security-Compatible/dp/B07HHG8Z11](url)

On all the sites I have found it, they list it as supporting Tuya - so I am hoping that I can use this avenue to get Tasmota on the device.

So I guess my question is simply, has anyone tried with this device?
If not, (since this is my first foray) do I just follow the Tuya-convert walk-through?

I'll be happy to report my findings if I am the first to test this device.

Thanks!
DeadEnd

meingraham, 2019-03-01:
You might want to consider https://www.youtube.com/watch?v=BoYVr2UwWWg

cbmboy, 2019-03-01:
> You might want to consider https://www.youtube.com/watch?v=BoYVr2UwWWg

Have you watched that full video?
I watched it a few days ago - he uses a 3.3v coin battery and states that the ESP01 needs 3.0V+ to operate.  After ~20 signals it no longer has enough voltage to send a signal.

This solution is completely useless in a high traffic location.
The only place this would be useful is on a very seldom used opening... _not my front door_.

Thanks for the response! There are likely useful places to use this very cool DIY sensor - but not for my application.

 - DeadEnd

meingraham, 2019-03-01:
I thought I was including this link (the updates to his original sensor) - https://www.youtube.com/watch?v=2kLZ7DlP9KU

cbmboy, 2019-03-01:
Thanks!
That does improve the number of signals to ~110 - but again not enough for a daily use door.
I am looking at AA or AAA sensors simply for the fact that I have a growing collection of rechargeable  due to kids :-)

meingraham, 2019-03-01:
$20 per door would still break my bank :frowning_face: 

Replicate Rob's 3D printed sensor but with 2AAA instead of CR2032. Sounds like a good DIY project.

TTFN

cbmboy, 2019-03-01:
You are absolutely right (on the DIY)!
Luckily I only have 3 doors that I want to track at my house (external doors).
The first road on my home automation path is security.
So exterior doors are my only concern right now.
Also, I found these sensors on sale yesterday for just over $12 each.

Thanks again for your input - the idea to modify the design to use a AA or AAA housing is a great idea.  I believe you could just solder leads to a holder, and then design a 3D printed box to fit.

I currently am avoiding 3D printing as I would have to pay someone else to do it, unless I want to do a ton of them and decide to invest in my own... currently that is not on my scope :-).

 - DeadEnd

meingraham, 2019-03-03:
A bit cheaper - https://www.aliexpress.com/item/Home-Door-Window-Detector-WiFi-App-Notification-Alerts-Battery-Operated-Home-Security-Sensor-Work-with-Alexa/32973283622.html?af=137322

meingraham, 2019-03-03:
See #5389

cbmboy, 2019-03-03:
> A bit cheaper - https://www.aliexpress.com/item/Home-Door-Window-Detector-WiFi-App-Notification-Alerts-Battery-Operated-Home-Security-Sensor-Work-with-Alexa/32973283622.html?af=137322

I actually did buy it from AliExpress - that same vendor I believe.
I would have been nice to get the price listed from Russia - but no US delivery... and I don't know if there are any loopholes to get it.  If you know of some I would be interested (if they are legal, lol).

> See #5389

Exactly what I was asking for!
Thank you very much for linking this!
Sounds like I should be able to get Tasmota on them then.
I might try the OTA method since I do not have experience or hardware for the serial method.
If that fails though, it sounds like I can fall back to the "traditional" way.

Thanks again!

closed 2019-03-06 by ascillato2