arendst/Tasmota #5375: Temperature calculation is wrong when using a DS18S20 sensor.

wecl, 2019-03-01:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

netpok, 2019-03-01:
Please fill out the form above, especially the parts about your problem.

wecl, 2019-03-01:
**To Reproduce**
connected a DS18S20 sensor to GPIO14 to a yunshan module and record a time series.

**Expected behavior**
smooth change of temperature over time.

**actual faulty behaviour**
when temperature is in- or decreasing the calculated temperature is jumping suddenly about 0.5 degrees at some point in time.

**what i did**
I looked to the sourcecode of the file xsns_05_ds18x20.ino, function Ds18x20Read() and compared the calculation with the spec of the DS18S20 sensor (https://datasheets.maximintegrated.com/en/ds/DS18S20.pdf).
On page 6 they describe as follow:
"Resolutions greater than 9 bits can be calculated using the data from the temperature, COUNT REMAIN and COUNT PER °C registers in the scratchpad. Note that the COUNT PER  °C  register  is  hard-wired  to  16  (10h).  After  reading  the  scratchpad,  the  TEMP_READ  value  is  obtained  by  truncating  the  0.5°C  bit  (bit  0)  from  the  temperature  data  (see Figure  4)..."

On line 332 of xsns_05_ds18x20.ino, the code fragment
`
if (data[0] & 1) {
          temp9 = ((data[0] >> 1) + 0.5) * sign;
        } else {
          temp9 = (data[0] >> 1) * sign;
        }
`
should be replaced by

`temp9 = (data[0] >> 1) * sign;`
because it is wrong to add 0.5 in case of calculating resolution greater than 9 bit.

I changed the code accordingly and tested it on my module and the output is as expected.
But i didn't test the calculation when the temperature is lower than 0 degrees.

ascillato2, 2019-03-06:
Closing this issue as it has been fixed.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2