arendst/Tasmota #5377: Trying to flash tasmota on TYWE3S

jim20j, 2019-03-01:
I have a 3 gang touch switch with a TYWE3S. I am unable to flash tasmota on it.  I'm pretty sure it's in boot mode when I ground GPIO0 and all 3 lights are red when I power it up. I have flashed 6 sonoffs using the same process. I tried reversing the Tx and Rx as well. 
 I get an error "espcomm sync failed, espcomm open failed, espcomm upload mem failed" 

- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used : EspEasyFLashe/FlashEsp8266
- [ ] Hardware used : FT232RL serial adapter
- [ ] If a pre-compiled release or development binary was used, which one? :
- [X ] You have tried latest release or development binaries? :
[image]
[image]

meingraham, 2019-03-01:
have you tried flashing the Tuya model via [OTA](https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-OTA)?

jim20j, 2019-03-01:
is the Raspbian 9 the method to use? i don't have another sd card right now.

meingraham, 2019-03-01:
I flashed a fresh copy of Raspbian just for performing this. I would not think that using an existing Raspbian copy (perhaps updated to the latest packages) would be any problem.

jim20j, 2019-03-01:
according to the video it appears multiple steps and configs are involved, is this the only solution to this in the meantime? i am new to this but have been catching on pretty quick.

meingraham, 2019-03-01:
This is the best OTA solution for Tuya modules. There are multiple steps but they are well documented and straightforward. The process is quite robust and foolproof. I would definitely recommend this over doing it via the serial interface.

Take a look at these videos for additional Tuya OTA flashing help:
https://www.youtube.com/watch?v=O5GYh470m5k
https://www.youtube.com/watch?v=q4HrpSORcXA
https://www.youtube.com/watch?v=av84uORLB9Y

jim20j, 2019-03-02:
Followed digiblurDIY, Success! 
Thank you!

meingraham, 2019-03-02:
Glad to hear you got it done. For some of these devices you literally have to break the case open. And even then you sometimes can't get to the pins you need. Getting the OTA process for Tuya based devices (and they are everywhere) was huge!

Once you're in this for a bit (I just got started a few months ago), you begin to recognize model numbers, etc.

Please consider contributing a [template](https://github.com/arendst/Sonoff-Tasmota/wiki/Templates) for this device if it is not already in the [template repository](https://blakadder.github.io/templates/).

For ongoing assistance, a great place to get help is on the [Tasmota Discord Chat](https://discord.gg/Ks2Kzd4).

If you're done here, please close this issue.

Mike

closed 2019-03-02 by jim20j

Eiferer, 2019-07-16:
Hi,

My issue here is a problem with OTA has left me with the need to flash the hard way..
Any help, on how to get this unit into configuration mode?

BreBar07, 2019-07-17:
I'm having issues getting my device into tasmota. I've tried OTA and flashing by serial no luck :(

OTA never detects device.
Serial starts but fails shortly after.

I have single gang, double gang and power outlet I want to flash.

BreBar07, 2019-07-18:
Tonight I confirmed the chip does go into programming mode but fails everytime. After numerous attempts I say it get as far as displaying 15% complete before failing. Anyone got ideas? I have seen one video CH_PD be connected to ground, but I don't think that's required.

kozumin, 2019-07-23:
Hi, did you manage to flash it? I'm in the same situation :( too late for OTA and the wiring method doesn't work. Mine is either not going in the boot mode (GRND+GPIO0) or boot mode is triggered in a way I don't know yet. Within the app works fine... but I want to add a few of them in HA
It is a  Wifi Leading Edge Dimmer : https://www.aliexpress.com/item/33015175004.html?spm=a2g0s.9042311.0.0.70da4c4dLDZmHi
Any help?

BreBar07, 2019-07-23:
No, mine definately goes into boot mode. U have been able to erase flash but it fails to upload flash everytime (fails at 8 or 12%) have tried different tools but same result :( if anyone can help I can donate a switch for testing....

kozumin, 2019-07-23:
Is  GPIO0 + GRND enough to put TYWE3S  in boot mode? 
For me, the flashing process doesn't even start on UI... just waiting in a forever loop. 
The led goes solid green with the two pins wired  but I'm not sure if that means is in boot mode. 
I'm using NodeMCU-PyFlasher and it worked nicely for all Sonoff modules I've flashed before.

meingraham, 2019-07-23:
If any LEDs lit, it's likely not in programming mode. Connect GPIO0 to GND and hold the connection for 3-5 seconds while powering up the device to ensure the ESP completes the boot process before disconnecting GPIO0. Cycle power between erase_flash and upload if not using PyFlasher.

Also... this https://www.thingiverse.com/thing:3231225

kozumin, 2019-07-23:
In my case, if I power the device (via USB-TTL) without shortcut of the two pins, the led is always off. If i shortcut them, the led goes green (and stays green).
The jig is nice and I actually printed one but I couldn't find those pins at jaycar (NZ)... so I have to order them online.

kozumin, 2019-07-23:
[image]
[image]

meingraham, 2019-07-23:
The fact that something is driving a GPIO to turn on the LED makes me suspicious. That, along with the fact that when you do short GPIO0 to GND (regardless of the LED) and you can't flash the chip confirms that it isn't in programming mode.

A short command to check if it's in programming mode before trying to erase would be `read_mac` or `flash_id`. When you get a successful response from those commands, then the ESP is in programming mode and you can then `erase_flash` and then upload the firmware.

Did you check to see if perhaps the button is connected to GPIO0? Also, those 4 contact holes below the PCB information look like they could be the programing pins (Vcc, GND, Tx, Rx). Did you try checking if they traced back to the respective TYWE3S pins? I say this because I'm wondering if there could be some soldering issue (e.g., hidden under the TYWE3S itself) so you could remove the leads and maybe remove the issue.

Mike

BreBar07, 2019-07-23:
My chip flash fails at 8 or 12% any ideas?

I've changed baud rate, resoldered, changed USB, checked power supply... Scratching my head.

meingraham, 2019-07-23:
@BreBar07 `checked power supply` - in what manner?

BreBar07, 2019-07-23:
@meingraham as in confirmed it's not failing because voltage drop to the board. I've tried with both power from the USB and also powered directly from 3.3v output.

meingraham, 2019-07-23:
Voltage is not usually the issue. It's insufficient power; that is the amperage supplied along with stable voltage.

BreBar07, 2019-07-23:
@meingraham I will double check the current but since I get the same result with USB power and external power I think this is the cause (I would be happy if it is lol)

BreBar07, 2019-07-23:
By the way who ever helps solve my problem can claim my spare glass face wall Switches! A as I bought a couple extra!

meingraham, 2019-07-23:
@BreBar07 You say USB power... what kind of device are we talking about in your case? Not the TYWE3S, but the device it's in.

kozumin, 2019-07-23:
> The fact that something is driving a GPIO to turn on the LED makes me suspicious. That, along with the fact that when you do short GPIO0 to GND (regardless of the LED) and you can't flash the chip [...]

With the multi-meter, it  doesn't look like the 4 contact holes are the standard programming pins as only the first one from the right and the third one connects to TYWE3S as 1st is VCC and third is GRND.
While checking the CB on both sides, I believe the button is not connected to any of the TYWE3S pins other than the ground. I tried to follow its pins with the multimeter but i've got lost on those routes. 

It sounds like  the "wise" SmartLife that made this dimmer just added recently a very tricky 
protection on hardware side not only on software one.... and that for just 12$ :)) 
I am adding the picture of the back of this board in a hope that will make more sense for you.
Honestly, I'm thinking this is a lost cause
[image]

meingraham, 2019-07-23:
I wonder if the same issue is affecting you as with this device:
https://github.com/arendst/Sonoff-Tasmota/wiki/ERC309-Kinetic-Switch

Here, the reset chip reset has to be grounded in order to flash the Tuya module. Alternatively, the Wi-Fi module has to be removed in order to flash it.

kozumin, 2019-07-23:
> I wonder if the same issue is affecting you as with this device:
> https://github.com/arendst/Sonoff-Tasmota/wiki/ERC309-Kinetic-Switch
> 
> Here, the reset chip reset has to be grounded in order to flash the Tuya module. Alternatively, the Wi-Fi module has to be removed in order to flash it.

:) Intriguing is the first part as I don't have to unsolder the spider out of the PCB. The question is how can I detect the reset pin on the reset chip?

kozumin, 2019-07-23:
some findings
[image]

kozumin, 2019-07-23:
Searching for this MCU chio code by GKOPY i found a new thread here:
https://github.com/xoseperez/espurna/issues/1729
If that helps me to solve the problem I will post it here too. Fingers crossed!

kozumin, 2019-07-23:
YES! the solution worked. The dimmer is now tasmotized by doing the followings:

1. wire a shortcut wire between the left side pin hole (close to J3 marking on my last picture) to the ground - and that wire must stays there for the whole flashing process.
- 3 seconds shortcut between GPIO0 and GRND
flash it!!
Mine is FREE!!!!
Thanks meingraham  for keeping me engaged into it :)

meingraham, 2019-07-24:
@kozumin 

Would you repay the kindness? Could you create a wiki page named `RJWF-02A Dimmer` and put the photos and information on how you were able to flash this device? Include also the template you configured for the device.

Thank you.

Mike

kozumin, 2019-07-24:
Hi Mike,
I will do that as soon as I finish installing all 7. 
fern-house.co.nz says thank you Mike :)
[image]

kozumin, 2019-07-24:
the question is now how to integrate them to Hassio? any tutorial out there?

meingraham, 2019-07-24:
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

A ton of Tasmota/HA tutorials on YouTube - DigiblurDIY, DrZzs, TheHookUp

There is also an HA Discord server

tim-dcl, 2019-07-25:
I found the following link https://forum.iobroker.net/topic/9886/tuya-jinvoo-unterputz-wandschalter/22 
it looks like RST needs to be involved in order to get into flashing mode. 
Can anyone verify this?

meingraham, 2019-07-25:
To clarify, this refers to the RST pin on the secondary MCU. This is separate from what is required to get the ESP into programming mode (e.g., its own RST pin).

BreBar07, 2019-07-26:
No luck for me, LED's are also flashing so I guess is not in boot loader mode :(

It seems if I initiate the flashing soon after connecting to PC flashing starts but fails (8~12%). If i wait about 5 sec to start flashing its not found.

Any ideas?
[image]
[image]
[image]

tim-dcl, 2019-07-26:
> Capture
> IMG_0434
> IMG_0435

i dont see a connection on the RST in (the one on the 3v side opposite 
end). I had the same issue until i included that pin in the startup 
sequence.

BreBar07, 2019-07-27:
Thanks @tim-dcl do you mean you grounded it?

tim-dcl, 2019-07-27:
yes. you need to ground RST, GPIO0 to start. after 3 seconds release 
RST. after another 3 seconds, release gpio0.

shoudl work after that. i used FlashESP8266.exe to flash new bin 
Tasmota_6.5.0(digiDIMv6).bin. This bin was made for this dimmer.

tim-dcl, 2019-07-27:
[Tasmota_6.5.0(digiDIMv6).zip](https://github.com/arendst/Sonoff-Tasmota/files/3437744/Tasmota_6.5.0.digiDIMv6.zip)

BreBar07, 2019-07-27:
:( failed, still not detected.... I have tried different waiting periods still not detected :(

tim-dcl, 2019-07-27:
is this the first device you have done with this method?

BreBar07, 2019-07-27:
Yes first device.

BreBar07, 2019-08-01:
Anyone got any ideas?

meingraham, 2019-08-01:
What is the device model that has that TYWE3S in it?

Not that it looks very easy to accomplish, but, are you able to detach the TYWE3S from the main PCB?

BreBar07, 2019-08-01:
I haven't detached, as screen shot above it seems to be an ESP8266EX.

Do u think it's required to detach, not sure that will bring much benefit 😔

meingraham, 2019-08-01:
@BreBar07 

I guess I should confirm that you have tried to ground Reset on the ESP while booting and if that didn't do it, that you also tried grounding Reset on the MCU while going through the "normal" ESP programming mode boot.

There could be a connection to the main PCB that is keeping the Wi-Fi module from going into programming mode. There are several devices where the Reset pin is connected also to a secondary MCU and the MCU is keeping the ESP from booting into the desired mode. Some have found that the only way to flash the Tuya module was to detach it. It was mounted differently that this one so a bit easier.

In all of these cases I'm referring to Tuya modules that are ESP8266/ESP8285 based.

Mike

BreBar07, 2019-08-29:
Thanks @meingraham, I've been a little quiet as we just had a little baby join us. 

I've grounded both reset and gpio pins with no luck. I've got a little time over the coming week I'll try remove the chip and flash it (fingers crossed).

BreBar07, 2019-08-29:
Well I was excited so I got onto this faster than expected. Removing the 8266 I believe is almost impossible without destroying the board's so i decided to do some playing with grounding pins on the other MCU chip and success (I will add a more detailed description later, once next problem is solved).

Now I have Tasmota installed and I assumed that the configuration "Sonoff T1 1CH Module" and '2CH' would work but it seems not. Does anyone have any advise how to configure or is it just a trial and error exercise?

**Basic Reflash**
gpio0 -> gnd
tx -> rx
rx -> tx
vcc -> 3.3v
gnd -> gnd
rst -> gnd

mcu pin? -> gnd (upload picture later)

After a few seconds release rst from ground and device will flash!

BreBar07, 2019-08-29:
I have set all GPIOs to relay's and toggling them in the Tasmota UI does nothing :(

I will try reflash another spare device tomorrow and see if I get the same result. Hopefully the MCU doesnt control the relay / led / button.

meingraham, 2019-08-29:
@BreBar07 What is the manufacturer and model number of your device?

It may be using the Tuya protocols to control the functions via the MCU. Ping @shantur. I think it's time to take this discussion over to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4).

BreBar07, 2019-08-29:
On the case sticker model = MG-MD04W, that's for single gang my double gangs are all installed. 

It seems to be the same family at the power socket (https://blakadder.github.io/templates/makegood_MG-AUWF01.html)

Thanks alot @meingraham I'll jump into the tasmota support channel.

anantbhula, 2019-09-24:
> @meingraham Well I was excited so I got onto this faster than expected. Removing the 8266 I believe is almost impossible without destroying the board's so i decided to do some playing with grounding pins on the other MCU chip and success (I will add a more detailed description later, once next problem is solved).
> 
> Now I have Tasmota installed and I assumed that the configuration "Sonoff T1 1CH Module" and '2CH' would work but it seems not. Does anyone have any advise how to configure or is it just a trial and error exercise?
> 
> **Basic Reflash**
> gpio0 -> gnd
> tx -> rx
> rx -> tx
> vcc -> 3.3v
> gnd -> gnd
> rst -> gnd
> 
> mcu pin? -> gnd (upload picture later)
> 
> After a few seconds release rst from ground and device will flash!

@BreBar07 Any chance of you sharing the MCU grounding pics? I have Moes Dimmer and I am unable to get TYWE3S into boot mode.
Thanks in advance

BreBar07, 2019-09-24:
@anantbhula you need to ground as pic below. Sorry I'm at work right now and don't have my notes on exact pin but grounding and trying will work as long as your board is similar / same.
[image]

BreBar07, 2019-09-24:
I just manually held a ground pin to the MCU it was actually wedged in between 2 pins but flashing was no problem.

anantbhula, 2019-09-24:
@BreBar07 Appreciate you for all your help. Since I have different board [[MOES DS01C](https://amzn.to/2kSdVga)] that didnot work for me however I ended up desoldering TYWE3S for flashing and re-solder now off to the functionality of the dimmer.

shantur, 2019-09-24:
@anantbhula : Some pages that might help you

https://github.com/arendst/Sonoff-Tasmota/wiki/MCU-Based-Tuya-Dimmers-and-Switches

https://github.com/arendst/Sonoff-Tasmota/wiki/TuyaMCU

twonky26, 2019-12-21:
The method that worked for me was cutting RX and TX pins on BF7612AM28T chip which is located on the the other side of the board (of Moes DS01C dimmer ). So I guess it will work for any device with TYWE3S which has trouble flashing. 

First, trace RX and TX pins from TYWE3S to BF7612AM28T (or similar) chip.
Second, carefully cut them with sharp blade.
Third, bend them up a little bit, so there is no contact.
After that it should flash just fine. But don't forget to bend them down and resolder after flashing.

[image]
[image]
[image]
[image]
[image]
[image]

arvidj01, 2020-01-24:
Reading this ... [MCU Based Tuya Dimmers and Switches - Flashing - Preparation](https://github.com/arendst/Tasmota/wiki/MCU-Based-Tuya-Dimmers-and-Switches#flashing---preparation) suggests that holding the reset NRST/RST pin to gnd might allow flashing without having to do any surgery on the MCU pins. However I've not been able to find a BF7612AM28T datasheet to determine which pin is the NRST/RST pin.

If anyone has access to that information would you please share it. I would like to give it a try.

[43 more replies]