arendst/Tasmota #5378: Xiaomi Mi Desk Lamp button dimming and color chamging to slow

computerkoenig, 2019-03-02:
At the Xiaomi Mi Desk Lamp there is a wheel to change the brightness and light color.
If you turn that wheel you see in the logs that there is is always changing the value by 1. for that reason you have to turn the wheel really often to change it. This shoud be set to 5 or 10 to have less steps.

joba-1, 2019-03-02:
sounds useful.

Most of my lights would need a quadratic or logarithmic increment though
(small steps while dark and big steps when bright).

Never looked at rules, but maybe what you want can be achieved already with
them? Like „if value between 1 and 5 then value = 10“ and so on and similar
for stepping down.
-- 
iphone so i typo

ascillato2, 2019-03-06:
Hi,

Please, can you complete the troubleshooting template, or at least share the output of the STATUS 0 command and the output of the console when you see that? Thanks

ascillato, 2019-03-07:
@computerkoenig

Hi,

Any feedback?

ascillato2, 2019-03-08:
@computerkoenig

Hi, 

We want to help you but we need more information in order to do so. Please, when you have time, share the information requested. Thanks

closed 2019-03-08 by ascillato2