arendst/Tasmota #5379: Sonoff RF Bridge IR receive and IR Send problem

kiralikbeyin, 2019-03-02:
Latest sonoff.bin uploaded with esptool.py
Device 25 SRFB

GPIO5 IR RECEIVE
GPIO4 IR SEND
also tried 
GPIO4 IR RECEIVE
GPIO5 IR SEND

Didnt send... 
IRsend {"Protocol":"NEC","Bits":32,"Data":14643343}

No IR receive data on console...

(Hardware is working with OpenMQTTGateway library, both ir send and receive is ok.)

Jason2866, 2019-03-02:
Since there is no information about your setup and all infos from issue template are missing....

kiralikbeyin, 2019-03-02:
@Jason2866 you cant make any setup this is not IDE

Terminal command to upload 👍 
esptool.py -p /dev/cu.usbserial-A7012RPU write_flash -fs 1MB -fm dout 0x0 sonoff.bin

Jason2866, 2019-03-02:
`Status 0` ?  Circuit Diagramm wiring?

kiralikbeyin, 2019-03-02:
https://github.com/1technophile/OpenMQTTGateway/raw/master/img/OpenMQTTgateway_ESP8285_Addon_IR.png

Jason2866, 2019-03-02:
With this circuit setup you have a very weak sending unit. You have to place the IR diode direct in front of
the IR Receiver to get a valid signal...
You can use a standard IR remote to check your receiving unit. You get output in console
For checking sending unit you can use a old smartphone camera. If it lights up it is working in general

kiralikbeyin, 2019-03-02:
Thanks for your advices:)

Circuit is working very well with https://github.com/1technophile/OpenMQTTGateway 

if you dont have a Sonoff RF bridge, someone else can try and see the BUG that IR RECEIVE AND IR SEND is not working for SRFB

@arendst

Jason2866, 2019-03-02:
Will try with my Rf Bridge...

Jason2866, 2019-03-02:
Update. IR Receive does. work. I have no spare IR Diode to test. I see no reason why it shouldnt work to.
There is a trap. Keep in mind that labeling GPIO 4 and 5 on Bridge is wrong (swaped). You can read this in wiki.
```
00:00:00 Project sonoff Bridge2 Version 6.4.1.17(sonoff)-2_5_0
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-82589D-6301...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-82589D-6301 with IP address 192.168.2.127
14:21:25 MQT: Attempting connection...
14:21:25 MQT: Connected
14:21:25 MQT: tele/sonoff-82589D/LWT = Online (retained)
14:21:25 MQT: cmnd/sonoff-82589D/POWER = 
14:21:25 MQT: tele/sonoff-82589D/INFO1 = {"Module":"Sonoff Bridge","Version":"6.4.1.17(sonoff)","FallbackTopic":"cmnd/sonoff-82589D_fb/","GroupTopic":"sonoffs"}
14:21:25 MQT: tele/sonoff-82589D/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-82589D-6301","IPAddress":"192.168.2.127"}
14:21:25 MQT: tele/sonoff-82589D/INFO3 = {"RestartReason":"Software/System restart"}
14:21:26 MQT: stat/sonoff-82589D/STATUS = {"Status":{"Module":25,"FriendlyName":["Bridge2"],"Topic":"sonoff-82589D","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:21:26 MQT: stat/sonoff-82589D/STATUS2 = {"StatusFWR":{"Version":"6.4.1.17(sonoff)","BuildDateTime":"2019-02-18T16:27:43","Boot":6,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
14:21:26 MQT: stat/sonoff-82589D/STATUS5 = {"StatusNET":{"Hostname":"sonoff-82589D-6301","IPAddress":"192.168.2.127","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:82:58:9D","Webserver":2,"WifiConfig":3}}
14:21:26 MQT: stat/sonoff-82589D/STATUS11 = {"StatusSTS":{"Time":"2019-03-02T14:21:26","Uptime":"0T00:00:08","Vcc":3.471,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":100,"LinkCount":1}}}
14:21:26 MQT: stat/sonoff-82589D/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:08","StartupUTC":"2019-03-02T13:21:18","Sleep":50,"CfgHolder":4617,"BootCount":150,"SaveCount":254,"SaveAddress":"F8000"}}
14:21:26 MQT: stat/sonoff-82589D/STATUS8 = {"StatusSNS":{"Time":"2019-03-02T14:21:26"}}
14:21:26 MQT: stat/sonoff-82589D/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000001"]}}
14:21:33 MQT: tele/sonoff-82589D/STATE = {"Time":"2019-03-02T14:21:33","Uptime":"0T00:00:15","Vcc":3.481,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":100,"LinkCount":1}}
14:22:05 MQT: tele/sonoff-82589D/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":5,"Data":"7F1C8735"}}
```

Jason2866, 2019-03-02:
Another Update. GPIO 4 and 5 are connected to USB. You have to cut traces in Bridge if you want to use GPIO 4 and 5 when powered via USB connector
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#optional-usb-power-while-flashing-the-rf-firmware-hardware-r2-only

kiralikbeyin, 2019-03-02:
Very intersting i got codes when powered with FTDI
14:32:55 RSL: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"DF906F"}}
14:33:02 RSL: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"DF10EF"}}

When i connected with USB, no code received as you said. 

But i am getting IR codes with OpenMQTTGateway library even its powered by USB. How is this possible i cant understand..

Jason2866, 2019-03-02:
TBH i dont know. Cut the traces as shown in picture.
[image]

We will add infos to wiki entry Rf Bridge.

```
ONLY for R2 boards: The R2 boards connect pins GPIO4 and GPIO5 to the USB data lines of the power connector. Compliant USB chargers/power supply are expected to short the two pins together, which is no good. These lines are not used by the original firmware, nor by Tasmota.

The traces toward the USB connector need to be cut to avoid interference by the USB wiring
```

Jason2866, 2019-03-02:
Please close the issue, since your question is solved.
Thx

kiralikbeyin, 2019-03-02:
ok thanks alot 👍

closed 2019-03-02 by kiralikbeyin

1technophile, 2019-03-02:
> With this circuit setup you have a very weak sending unit. You have to place the IR diode direct in front of
> the IR Receiver to get a valid signal...

Hello @Jason2866 , could you point what makes you think this?

kiralikbeyin, 2019-03-02:
@1technophile tasmota lib didnt send IR codes but OpenMQTTGateway can send with same circuit..