arendst/Tasmota #5380: [Feature Request] Support CD74HC4067 16 Channel Analog / Digital Multiplexer...

allnerag, 2019-03-02:
**Have you look for this feature in other issues and in the wiki?**
Yes, but did not see any mention of it.

**Is your feature request related to a problem? Please describe.**
Yes, Certain Devices only have a specific amount of ports and most of them only have one Analog port, except the ESP32. The problem is that if you want to connect multiple digital devices on one ESP we can do it to a specific amount, D1 Mini only about 8.  If you go to the Analog side of things, we can only go to ESP32 and it is not fully supported yet on TASMOTA only on ESPHOME. The problem that I have and I assume more people have this issue is that if you need to use a Voltage follower and a comparetor, you need to use 1x voltage follower and one comparetor for each Analog or digital line you connect directly to an ESP.  If you use a Multiplexer like "CD74HC4067 16 Channel Analog / Digital Multiplexer" you only need one of each and use the multiplexer ability of channels through one of the ports. By doing this you also bring the count down of connected devices on your network.

I want to use this for alarm port monitoring (Digital for new alarm system only using This, or connecting to an analog port for piggy back on Alarm systems with resistors)
With homes that you want to use presence detection, many homes have large installations of Alarm systems and we can use the PIR, Door Magnets to do automation with. Why add more devices if you can use existing once.
I have an IDS x64 that can do 64x zones and unable to connect directly to it, so piggy back is the only way.

**Describe the solution you'd like**
Add 2x Options to the Selection of Devices to enable the feature to use a Multiplexer. Only enable Analog Port 0 and Digital Port 0 for the selection. This means that you can do Analog Read if it is on Analog pin and only digital if you select it on the digital port.
Because it is a 16x Multiplexer you need 4x Digital port to address each of the ports depending on what port are pulled high.  This is for the same if you use the Multiplexer for Digital or Analog. The pin on D0 or A0 is called the Common Port, and depending what that port is you can do certain things.
If it is on Digital port it will just show "High or Low". If it is on Analog port it can only do Analog read).  These ports can be set for Input or Output.

So the options on D0 could be; 
1. CD74HC4067 16 Channel Digital Multiplexer Input
2. CD74HC4067 16 Channel Digital Multiplexer Output
This will automatically change D1-D4 to Digital Out Only to use with the Multiplexer. This will Allow only Digital In or Out.

So the options on A0 could be; 
1. CD74HC4067 16 Channel Analog Multiplexer Input
2. CD74HC4067 16 Channel Analog Multiplexer Output
This will automatically change D1-D4 to Digital Out Only to use with the Multiplexer. This will Allow only Analog Read and write.

The info per Multiplexer port will come through channels (1-16). This can be be set using MQTT topics per channel. 

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
I did look at using each digital port on ESP8266 (Node MCU),  this will mean I have to use 8x ESP8266 Node MCU's to be able to do the same thing and Will also have to use 32x Voltage followers and 32x Comparators to do the same.  for Analog ports I have to use 4x ESP32's with the same amount of voltage followers and Comparators as the digital one.  I cannot use the ESP32 with TASMOTA as it does not have Analog shipped to it yet, I can only use ESPHOME for the Analog side. I can use an Arduino Mega but then I loose the easy integration via TASMOTA or ESPHOME, or I have to Use RX/TX to an ESP-01 or ESP8266 / ESP32 for that.  That will also mean I have to use custom Arduino Code for that.

**Additional context**
Here is an Arduino Library for this;
https://github.com/waspinator/CD74HC4067

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-02:
There is already support for extending ports https://github.com/arendst/Sonoff-Tasmota/wiki/MCP23008-MCP23017 and support for ADS1115 a 16-Bit ADC...
The support for this chip seems redundant

allnerag, 2019-03-02:
I see that thank you, the problem is that you cannot use a voltage follower and comparator with I2C.  I then have to build new Board with multiple Followers and comparators just to make it work.  The board will be bigger that my existing Alarm System just to have a parallel system. That is why I was looking at the specific version of Multiplexer.

Jason2866, 2019-03-02:
Maybe someone will someday implement....

netpok, 2019-03-04:
If you want you can implement it, more features and pull request are always welcome.

closed 2019-03-06 by ascillato2