arendst/Tasmota #5381: T1 switch cannot connect while on AC power

ak47uk, 2019-03-02:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [X] Hardware used : Sonoff T1 UK 3-gang, RaspberryPi
- [X] If a pre-compiled release or development binary was used, which one? : sonoff.bin 6.4.1
- [X] You have tried latest release or development binaries? : latest release
- [ ] Provide the output of command ``status 0`` :

Hi all, I have searched for this issue and gone through any threads that seemed relevant (albeit concerning the Sonoff Basic rather than the T1) but am now stuck.

I have 3 Sonoff T1 UK 3-gang switches, two are the v1 and one is the v1.1 board. I successfully flashed them all using my RaspberryPi, ran the "reset 1" command which allowed me to connect to the sonoff SSID and then connect to my WiFi which worked in all three cases.

I then removed the serial connections, connected to the relay boards and they all powered up and were functional when using the physical buttons. The v1.1 and one of the v1 boards connected to WiFi no problem and seem to have been solid since but one of the v1 boards will not connect to the WiFi reliably (I can see from my AP that it has connected briefly a few times).

I connected it to the RaspberryPi again and the output I am seeing is the same as the other boards (sorry I didn't get a copy of this) and it successfully connects to my AP and I can access the webserver and control/configure the device. As soon as I connect to the relay board again the WiFi will not connect.

Reading the other threads (below) the Sonoff basic had this issue but as the hardware is different I am not sure if a resistor would help me and it is working on my other v1 board. I measured VCC to GND serial pins whilst connected to the relay board and I got 0.01 but I only have my auto ranging multimeter here, I measured the same pins on the working v1 board and got the same result. Just in case I tried switching the relay boards with the working one and the same switch has the problem so relays look fine.
https://github.com/arendst/Sonoff-Tasmota/issues/391
https://github.com/arendst/Sonoff-Tasmota/issues/267

I can't see any dry solder joints so am not sure what I can try from here. They were all working fine using eWeLink for months.

Any input much appreciated!

Jason2866, 2019-03-02:
Try latest version from http://thehackbox.org/tasmota/  and try all available cores (2.3.0, 2.4.2 and 2.5.0)

ak47uk, 2019-03-03:
I just tried upgrading to 2.5.0 core sonoff.bin (1MB 6.4.1.19) and have the same issue, I will try the others but it looks like this might be a hardware as the other v1 board works fine. 

Anything else to try? Might have to replace it if not as I am out of ideas!

Thanks

Jason2866, 2019-03-03:
I have no idea left. I would try to replace

ak47uk, 2019-03-03:
Thanks, I took the header that connects the front plate to the relay board off and all the pins and solder looked fine so I’ve given up and ordered a replacement.

closed 2019-03-06 by ascillato2