arendst/Tasmota #5382: A fatal error occurred: Timed out waiting for packet header

neirbomn, 2019-03-02:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Python 2.7.10, esptool.py v2.6
- [x] Hardware used: 
1. Sonoff S26 (used two different ones bought in same batch direct from iTead)
2. MacBook Pro (Retina, 13-inch, Early 2015) using macOS Sierra 10.12.6
3. (a) Used FT232 (specs [here](https://www.jaycar.com.au/medias/sys_master/images/9183423299614/XC4464-manualMain.pdf))
3. (b) Also used Freetronics USB to Serial ATmega16u2 MCU. (Specs [here-ish](https://www.freetronics.com.au/products/usb-serial-adapter#.XHp6UtFS-Rs))
- [x] If a pre-compiled release or development binary was used, which one? : Sonoff 6.4.1 sonoff.bin
- [x] You have tried latest release or development binaries? : Yes
- [X] Provide the output of command ``status 0`` :
```
None!
```

### I started to post issue on esptool but believe it's more appropriate here **due to last observation**. I don't know if there's anything in it.

###Sorry the details below are long but it saves me going over old ground! I know what the error means but I don't know why I'm getting it or how to troubleshoot any further with the S26. Any help gratefully received.

# Full esptool.py command line as run:
Various...
Using an FT232 (spec [here](https://www.jaycar.com.au/medias/sys_master/images/9183423299614/XC4464-manualMain.pdf))
```
$ esptool.py --p /dev/cu.usbserial-AL00M1ON erase_flash
```
and also 

```
$ esptool.py --p /dev/tty.usbserial-AL00M1ON erase_flash
```
Using Freetronics USB to Serial ATmega16u2 MCU. (Specs [here-ish](https://www.freetronics.com.au/products/usb-serial-adapter#.XHp6UtFS-Rs))
```
$ esptool.py --p /dev/tty.usbmodem1411 erase_flash
```
# Full output from esptool.py (please copy and paste all lines of output)

Using an FT232
```
esptool.py v2.6
Found 2 serial ports
Serial port /dev/cu.usbserial-AL00M1ON
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/cu.usbserial-AL00M1ON failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
```

Only change with ATmega16u2 MCU is port name change.

# What is the expected behaviour?

Successful erasing of flash memory

# Do you have any other information from investigating this?

Here is what I have checked (with two separate Sonoff S26s purchased from iTead):

1. Continuity test of all Dupont cables used. All pass.
2. Each gives the expected behaviour when powered on (though I haven't configured or paired with eweLink app).
3. Pressing the button on the case of each switches the relay.
4. On ESP board (disconnected from mains power) - soldering the tested Dupont cables either on solder pads (or Jx joints to mainboard) powers on device. LED behaviour as expected.
5. Power cycling S26 with GPIO0 grounded causes LED to remain unlit (expected behaviour).
6. The momentary button on board has been suspect for others so have bypassed button and grounded across solder points for button. When powered without grounding GPIO0, grounding the wire causes red LED to come on, and when held in place green LED changes sequence of blinks to indicate readiness for pairing. Grounding during power cycle produces same behaviour as when pressing button.
7. I have used `esptool.py -p /dev/cu.***` and `esptool.py -p /dev/tty.***` where *** depends on USB to serial device being used to try and erase flash.
8. The FT232 supplies 5V power (I know as it fried an ESP-01 and caught me out a year or two ago) even though it has a 3.3V switch this is just for the logic. Power remains at 5V. For this device I attached at Vin and GND on the DS117 Voltage regulator. The Vout is 3.3V exactly. I have tried this on a USB powered hub as well as the MacBook and used a separate USB phone charger for power from a NodeMCU board (500mA max current from 5V pin) with common ground with the FT232 or ATmega16u2 MCU.  
9. Using an Adafruit Huzzah, esptool reads, writes and erases as expected.
10. Using an ESP-01 FTDI adapter and an ESP-01, reads, writes and erases as expected.
11. I have tried at various baud rates (115200 etc... includng 74880 and 9600) to no avail using esptool.

# Is there any other information you can think of which will help us?

The Sonoff S26s (I bought 5 and now wish I hadn't!) have a sticker on the relay which says the unit is an S26I (capital i). I note all in the S26 section ONLY have S26 on them (or I may be clutching at straws!).

Jason2866, 2019-03-02:
You are not in flash mode. 
Have you connected GPIO 0 before you power it with 3.3V and connecting USB serial adapter to PC?

neirbomn, 2019-03-02:
> You are not in flash mode.
Have you connected GPIO 0 before you power it with 3.3V and connecting USB serial adapter to PC?

Thanks for your response. I have flashed plenty of ESP8266 chips. These are proving somewhat stubborn. I'm aware what the error means and I have tried to rule out all obvious things.

> Power cycling S26 with GPIO0 grounded causes LED to remain unlit (expected behaviour).
The momentary button on board has been suspect for others so have bypassed button and grounded across solder points for button. When powered without grounding GPIO0, grounding the wire causes red LED to come on, and when held in place green LED changes sequence of blinks to indicate readiness for pairing. Grounding during power cycle produces same behaviour as when pressing button.

I was under the impression power cycling the device with GPIO0 grounded and the LED being unlit was a good indication that it **was** in flash mode for S26 but aware that this error commonly means it **isn't**.

Jason2866, 2019-03-02:
Green led blinking sounds weired... It looks like something prevents to ground Gpio 0.
Do you can upload some photos? Is there anything else connected to GPIO 0 on the board?

Jason2866, 2019-03-02:
If it is similar to this https://github.com/arendst/Sonoff-Tasmota/issues/2808#issuecomment-419665963
you can desolder the ESP daughter board and try to flash it without anything else connected

neirbomn, 2019-03-02:
I have already done this too, see my comments at end of [thread](https://github.com/arendst/Sonoff-Tasmota/issues/2808).

Green LED blinking only occurs when you power device and then ground. This is the expected behaviour to pair if you wanted to use the eweLink app.

neirbomn, 2019-03-03:
I will try one more device but I am waiting on some more flux to arrive to ensure I have good joints. I am going to video the whole process and will post the link here showing each and every check I've made along the way. If I get success I will say so and close issue. I just am struggling to see what I'm doing wrong.

ascillato2, 2019-03-06:
Please, when you have time, tell us if that works for you. Thanks.
If you need further assistant, please find us in the Tasmota Support Chat. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2

neirbomn, 2019-03-07:
@ascillato2 
On yet another device... the procedure I am using doesn't flash the device.
Video of the procedure used is [here](https://youtu.be/7r4w4FuYVbA).

I would be very grateful if anyone can help me get to the bottom of this.

Jason2866, 2019-03-07:
I can see no error in your flash process. Esptool.py is the reliable tool for flashing.
Could you make a good photos (front and rear side) of the Esp daugtherboard?
Maybe something is connected to rx and tx lines from esp. This would explain the behaviour.

neirbomn, 2019-03-07:
Ok - here they are...
[
Front of board.](https://drive.google.com/file/d/16ipCS1P2v_oGQz7YiQ9wpC8bMiSL-8T6/view?usp=sharing)

[Back of board.](https://drive.google.com/file/d/1dh75KbQZ-LYZI4ohz2WKt2uVwEZjzPyM/view?usp=sharing)

Nothing much happening on the back of the board just the flash memory.

Jason2866, 2019-03-07:
[image]
Measure if chip pin25 goes to ERX and if pin26 goes to ETX
Between ETX to GND or 3.3V should be high ohm. Same for ERX

neirbomn, 2019-03-07:
Both show 0 ohm between pin of ESP and the relevant solder pad meaning they are connected. Both show high between ETX -> GND (and 3.3V) and RTX -> GND (and 3.3V).

Jason2866, 2019-03-08:
What are you are doing would have been my suggestion too.
i am thinking that your setup doesnt work with this device (why i dont know...)
so buying a know to be good USB - serial is one point the other trying a different PC
I see no reason why it shouldnt be possible to flash....

neirbomn, 2019-03-21:
I'm pleased to report that after obtaining the Sparkfun Beefy that I have successfully flashed the S26. I had previously tried installing FTDI drivers on their website. In addition to buying this FTDI, I also uninstalled these in case there were conflicts between the Apple drivers and the FTDI drivers. Which one of these was responsible I'm not sure but I can now flash the devices! Yay!