arendst/Tasmota #5384: Help using button2 and button3 as triggers for rules

tatunts, 2019-03-02:
Hello everyone! I was wondering if anyone here could help me out.

I want to use a Sonoff Basic with a three gang regular pulse switches connected by using GPIO14, GPIO1 and GPIO3 so I can operate all of them by using single press and hold for each buttons (to use as automation triggers on Home Assistant). The firs button (GPIO14), on sigle press toggles the relay, hold publishes a mqtt message, second button (GPIO1) on sigle press publishes another message, hold another one and so on for the third button.

This is the wall mounted switch: 

[image]

My problem is that button1 works perfectly but button2 and button3 does not respond at all, no logs or errors.

The fist part was accomplished using those settings:

```Backlog Module 18; GPIO12 21; GPIO13 56; GPIO14 17; GPIO01 18; GPIO03 19; GPIO15 00; GPIO16 00; SetOption13 0; ButtonTopic 0; SetOption32 10; SetOption1 1; SetOption11 0; rule1 4; rule2 4; rule3 4; rule1 1; rule2 1; rule3 1; ButtonDebounce 50```

[image]

Here are all rules used (rule1 working perfeclty):

```
Rule1 on button1#state=1 do power1 2 endon on button1#state=3 do publish stat/topic/BUTTON {"BUTTON":"1","ACTION":"HOLD"} endon
Rule2 on button2#state=1 do publish stat/topic/BUTTON {"BUTTON":"2","ACTION":"TOGGLE"} endon on button2#state=3 do publish stat/topic/BUTTON {"BUTTON":"2","ACTION":"HOLD"} endon
Rule3 on button3#state=1 do publish stat/topic/BUTTON {"BUTTON":"3","ACTION":"TOGGLE"} endon on button3#state=3 do publish stat/topic/BUTTON {"BUTTON":"3","ACTION":"HOLD"} endon
```

This is the code to use with Home Assistant, in case anyone is interested:

```yaml
## Home Assistant
 
## Automation for triggering other scripts with default name and action:
 
automation:
  - alias: Sonoff Buttons
    initial_state: true
    trigger:
      platform: mqtt
      topic: stat/+/BUTTON
    condition:
      condition: template
      value_template: >
       {% set topic = trigger.topic.split('/')[1] | lower %}
        {% set button = trigger.payload_json.BUTTON | lower %}
        {% set action = trigger.payload_json.ACTION | lower %}
        {% set object_id = 'sonoff_botao_{}_{}_{}'.format(topic, button, action) %}
        {{ states.script[object_id] != None }}
    action:
      - service_template: script.sonoff_button_{{ trigger.topic.split('/')[1] | lower }}_{{ trigger.payload_json.BUTTON | lower }}_{{ trigger.payload_json.ACTION | lower }}
 
## this would result in: "sonoff_button_main_light_1_hold" if button on GPIO14 action is HOLD, so I can just create scripts with the correct naming and everything works automatically.
```

Here are the status log:

```stat/sonoff_testes/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff Testes Antonio"],"Topic":"sonoff_testes","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[5,5,5,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}```

Using Tasmota version 6.3.

Thanks for the help guys, really appreciate it!

tatunts, 2019-03-02:
So.... I was struggling with this for days and a few minutes posting this issue I was able to fix it haha

I think this information could be useful for many people so I will close it and write down what I did.

I had to use switches instead of buttons and set the rule to trigger from `switch1#state=2` not `switch1#state=1`.

Here are the configs:

```
Backlog Module 18; GPIO12 21; GPIO13 56; GPIO14 09; GPIO01 10; GPIO03 11; GPIO15 00; GPIO16 00; SetOption13 0; ButtonTopic 0; SetOption32 10; SetOption1 1; SetOption11 0; rule1 4; rule2 4; rule3 4; rule1 1; rule2 1; rule3 1; SwitchDebounce 50
```

```
Rule1 on switch1#state=2 do power1 2 endon on switch1#state=3 do publish stat/topic/BUTTON {"BUTTON":"1","ACTION":"HOLD"} endon
 
Rule2 on switch2#state=2 do publish stat/topic/BUTTON {"BUTTON":"2","ACTION":"TOGGLE"} endon on switch2#state=3 do publish stat/topic/BUTTON {"BUTTON":"2","ACTION":"HOLD"} endon
 
Rule3 on switch3#state=2 do publish stat/topic/BUTTON {"BUTTON":"3","ACTION":"TOGGLE"} endon on switch3#state=3 do publish stat/topic/BUTTON {"BUTTON":"3","ACTION":"HOLD"} endon
```

closed 2019-03-02 by tatunts

unimatrix27, 2019-03-12:
why would you have to use switches and why are button2...n not working at all? having the same issue with a Shelly1 where I added 2 buttons and they both work if configured as button1 but nothing happens when configured as button2