arendst/Tasmota #5385: Please kindly add Sonoff S22 as an official support device.

wongnam, 2019-03-03:
Hi @arendst , Please kindly add Sonoff S22 as an official support device due to there is a sensor jack as same as Sonoff TH(not Sonoff S2x) so we need GPIO14 for sensor support.

https://blakadder.github.io/templates/sonoff-s22.html
```
{"NAME":"Sonoff S22","GPIO":[17,255,0,255,255,0,0,0,21,56,255,0,0],"FLAG":0,"BASE":4}
```

Thanks.

arendst, 2019-03-03:
There is no difference between the Sonoff S22 and the Sonoff TH so just use the Sonoff TH module.

With v6.4.1.17 and up you can copy the TH template to the user template and rename it to Sonoff S22.

wongnam, 2019-03-03:
> There is no difference between the Sonoff S22 and the Sonoff TH so just use the Sonoff TH module.
> 
> With v6.4.1.17 and up you can copy the TH template to the user template and rename it to Sonoff S22.

I often use it as you suggest. but not yet understand why it is not officially named.
because I often have to self-compile it with the name TH instead of S22.

arendst, 2019-03-03:
The current software implementation dictates that the same functionality needs another entry in the template file at the cost of around 32 bytes. That can be spared by just using another device with the same functionality.

Adding new devices to the template will be done with more reluctance in the future as it is more flexible using the new termplate feature.

wongnam, 2019-03-03:
I understand. Thank you for your explanation.
It also means that S22 will never be added.
I closed this case here.

closed 2019-03-03 by wongnam