arendst/Tasmota #5386: Longitude & Latitude failing to set properly for values between 0 & -1

ggaavv, 2019-03-03:
Longitude Latitude failing to set properly for values between 0 & -1.
Working for number below -1

14:11:54 RSL: STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}

Failing above -1
-----------------------
14:25:39 CMD: latitude -0.2
14:25:39 RSL: Received Topic /latitude, Data Size 4, Data -0.2
14:25:39 RSL: Group 0, Index 1, Command LATITUDE, Data -0.2
14:25:39 RSL: RESULT = {"Latitude":"0.200000"}

14:25:59 CMD: Longitude -0.2
14:25:59 RSL: Received Topic /Longitude, Data Size 4, Data -0.2
14:25:59 RSL: Group 0, Index 1, Command LONGITUDE, Data -0.2
14:25:59 RSL: RESULT = {"Longitude":"0.200000"}

Working below -1
-----------------------
14:25:49 CMD: latitude -1.2
14:25:49 RSL: Received Topic /latitude, Data Size 4, Data -1.2
14:25:49 RSL: Group 0, Index 1, Command LATITUDE, Data -1.2
14:25:49 RSL: RESULT = {"Latitude":"-1.200000"}

14:26:04 CMD: Longitude -1.2
14:26:04 RSL: Received Topic /Longitude, Data Size 4, Data -1.2
14:26:04 RSL: Group 0, Index 1, Command LONGITUDE, Data -1.2
14:26:04 RSL: RESULT = {"Longitude":"-1.200000"}

curzon01, 2019-03-05:
True, but thats more or less a theoretical bug - these positions (0 to -0.999) are @ Atlantic ocean 700 km next to an island in the middle of nowhere

ggaavv, 2019-03-05:
A longitude band and a latitude band, rather than a single tiny point. The worst case is about a 15 minute sunset/sunrise error. The UK sits right in the middle of the band. My location is latitude -0.89 which got taken as 0.89 so almost worst case.

gemu2015, 2019-03-06:
this solves the issue

// better char to double
double CharToDouble(char *str) {
    // simple ascii to double, because atof or strtod are too large
    char strbuf[24];
    strlcpy(strbuf, str, sizeof(strbuf));
    char *pt=strbuf;
    double left,right;
    signed char sign=1;
    if (*pt=='-') sign=-1;
    if (*pt=='-' || *pt=='+') pt++;
    if (*pt=='.') {
        // .xxx notation
        left=0;
        goto gright;
    }
    // get left part
    left = atoi(pt);
    // skip number
    while (*pt>='0' && *pt<='9') pt++;
    if (*pt=='.') {
        // decimal part
gright:
        pt++;
        right = atoi(pt);
        while (*pt>='0' && *pt<='9') {
            pt++;
            right /= 10.0;
        }
    } else {
        right=0;
    }
    double result = (left + right);
    if (sign>=0) return result;
    else return -result;
}

curzon01, 2019-03-06:
@ggaavv - your are right, I was too sick, sorry

arendst, 2019-03-06:
Testing gemu suggestion

arendst, 2019-03-06:
Just implemented a fix. Pls test and report back.

ascillato2, 2019-03-07:
@ggaavv

Hi, 

The issue has been fixed. Please, test it and if the issue is solved, please close this issue. Thanks

closed 2019-03-08 by ascillato2

ggaavv, 2019-03-09:
Thanks for the quick fix. Working perfectly.