arendst/Tasmota #5387: Tasmota configuration gets reset on Wemos D1 mini

caccia78, 2019-03-03:
Hi guys,
I have a Tasmota 6.4.1.18(basic) running on a Wemos D1 Mini and following setup:
- D4 (GPIO2) --> Relay1i
- D8 (GPIO15) --> Button1
Once I save the configuration and module restarts, I get always a system reset with configurations lost (Wemos switches to AP and restarts as a Sonoff Basic Module).

My HW configuration includes a real relay on D4 pin.

Thanks for support and thanks for the great work!

Vince

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
With Tasmota 6.4.1.18(basic) running on a Wemos D1 Mini and following setup:
- D4 (GPIO2) --> Relay1i
- D8 (GPIO15) --> Button1
Once I save the configuration and module restarts, I get always a system reset with configurations lost (Wemos switches to AP and restarts as a Sonoff Basic Module).

My HW configuration includes a real relay on D4 pin.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Wemos D1 mini
- [X] _Development/Compiler/Upload tools used_ : PlatformIO (Atom)
- [X] _If a pre-compiled release or development binary was used, which one?_ : release 6.4.1.18
- [X] _You have tried latest release or development binaries?_ : Yes
- [X] _Provide the output of command_``status 0`` :
```
16:49:35 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:49:35 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:11","StartupUTC":"2019-03-03T15:49:24","Sleep":50,"CfgHolder":4617,"BootCount":4,"SaveCount":10,"SaveAddress":"FA000"}}
16:49:35 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.18(basic)","BuildDateTime":"2019-02-23T21:45:13","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
16:49:35 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iHunt",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
16:49:35 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":454,"Free":548,"Heap":29,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F082780","240483A0","21021706","000001C0"]}}
16:49:35 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6241","IPAddress":"172.20.10.2","Gateway":"172.20.10.1","Subnetmask":"255.255.255.240","DNSServer":"172.20.10.1","Mac":"B4:E6:2D:23:98:61","Webserver":2,"WifiConfig":4}}
16:49:35 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_239861","MqttUser":"DVES_USER","MqttType":1,"MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:49:35 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 03 15:49:35 2019","Local":"Sun Mar 03 16:49:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:28","Sunset":"18:35"}}
16:49:35 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-03-03T16:49:35"}}
16:49:35 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-03-03T16:49:35","Uptime":"0T00:00:11","Vcc":2.993,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iHunt","BSSId":"EA:85:50:8C:01:CF","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

**To Reproduce**
Flash Tasmota on WemosD1
Set module as Generic(18)
Set following configuration:
- D4 (GPIO2) --> Relay1i
- D8 (GPIO15) --> Button1
Wait 40 sec

**Expected behavior**
Just able to control Relay1 with button on D8.

**Screenshots**

**Additional context**

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-03:
You cant use GPIO15
https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#usable-pins

caccia78, 2019-03-04:
Thanks for very fast feedback Jason.

closed 2019-03-04 by caccia78