arendst/Tasmota #5389: Help with Tuya door/window sensor please?

senadaruc, 2019-03-03:
Dears,

I managed to flash this Tuya door/window sensor with Tasmota and all work perfectly except that I can't find the right configuration for the module. 

The specification and pictures of the device:
Description:
1.Detect open/close status of the door/window,compatibe with Tuya
2.Battery: AAAx 2  (not included ) Standby time: 5 years
3.Use time:1 years (15 times/day) 2years(7 times/day)
4.Standby current:10uA Wireless Type:2.4GHz
5.Protocol:IEEE 802.11b/g/n Wifi distance:45m
6.Operating Temperature:0℃～ 40℃ (32°F～104°F)
7.Free APP:Tuyasmart, one key setup wifi or ap mode connection with wifi .
8.Works with Alexa Echo and google home.
Specifications:
This is a smart door sensor which can work dependently without hub , and can be compatible with Alexa and google home assistant , Besides, it can IFTTT other smart items . Also, it can associate 2 device sot work to creat your own scene and shart your devices to others directly
how to work :
1. Run " smart life " from your smartphone desktop
2.Register and login
3. select device type , select " door " in the list to add the device.
Package Included:
1*smart sensor

https://www.aliexpress.com/item/Home-Door-Window-Detector-WiFi-App-Notification-Alerts-Battery-Operated-Home-Security-Sensor-Work-with-Alexa/32973283622.html

[image]
[image]
[image]
[image]

Thanks,

meingraham, 2019-03-03:
Did you follow a process similar to this - https://github.com/arendst/Sonoff-Tasmota/wiki/Configuration-Procedure-for-New-Devices?

Question - where you able to flash this OTA or did you have to open it use a serial adapter? Does this device go to sleep after inactivity or with the magnet present?

Mike

senadaruc, 2019-03-03:
Hi Meingraham,

I flashed with serial adapter using the latest Tasmota firmware.

No I didn’t follow the process!

S,

senadaruc, 2019-03-03:
And yes I can confirm that the device is going to sleep mode after the magnetic switch goes on/off.

meingraham, 2019-03-03:
Yes, the latest TASMOTA, but by which method - hooking up wires or wirelessly using Tuya-convert?

senadaruc, 2019-03-03:
Soldering the RS232 USB adapter and flashing directly. 

Thanks,

cbmboy, 2019-03-04:
Thanks for adding the photo's!  I ordered a few of these also (have not received them yet).  I was hoping they used the tywe3s board since the description listed "Tuya App" etc.  I see from your photo's they also appear to have broken out the pins needed to flash firmware (as you have already done).

This will be my first adventure into Tasmota and I am hoping to use the OTA method.  If that doesn't work I will do as you did and use a serial to USB.  One question (for anyone who knows).  I see the Ground (G) and power (V) are marked.  The other two are listed as N and S.  Do you know which is Transmit and which is Receive?  I'm new to this and wasn't sure (I'm guessing they are marked from another language?).

Thanks!
DeadEnd

senadaruc, 2019-03-04:
Hi cbmboy,

Please wire your RS232 to USB as per picture below!
[image]

Then use the [](https://github.com/nodemcu/nodemcu-flasher/tree/master/Win64/Release) application to flash the device with the latest Tasmota Firmware using the following config:

Baudrate 115200,  Flash size 1 MByte,  Flash speed 40 MHz , SPI Mode DOUT

You need to connect GPIO 0 and GROUND while connecting the board to USB.

Good luck.

senadaruc, 2019-03-04:
meingraham, I found a partially working profile "Manzoku Strip Module" but i need to configure like sensor not like button/relay!  how yo do that?

[image]
[image]

cbmboy, 2019-03-04:
> Please wire your RS232 to USB as per picture below!

@senadaruc 
Did you try using the four breakout pin holes to see if they are the Tx, Rx, Group, and Vcc?
I anticipated that this was the case - but since I am inexperienced, I could be wrong.
Also, it appears there is a button on the front, and from the pictures you attached, I see it is marked (SM1).
Were you able to identify what this is wired to?
Again, I was thinking this may be GPIO 0 to Ground as many other devices do.
This would greatly reduce the skill (soldering) needed to accomplish the Tasmota conversion.

This would make the process much easier if these assumptions are correct.

Thanks!

Jason2866, 2019-03-04:
@senadaruc  There has to be a extra electronic which sends ESP to off. 
Because Tasmota doesnt support deep sleep. Could you post a picture of the other side of the PCB
I am interested how is this be done.
Thx!

senadaruc, 2019-03-04:
cbmboy, the button will not put the CPU to program mode so I soldered the GPIO 0 and used the ground to force the program mode.

jason2866, here is the picture of the board: https://user-images.githubusercontent.com/14252382/53700282-e67e6800-3df0-11e9-8d70-f2aba5dd420c.JPG

Jason2866, 2019-03-04:
No label on the interesting IC `U1`  :-(
It looks like it does have a step up power converter so the 2 AAA cells are no problem for providing correct voltage for the ESP

senadaruc, 2019-03-04:
The device connects to internet only when you trigger the magnetic switch.

Jason2866, 2019-03-04:
Mhh, you could try to measure via a DMM the different GPIOs if one goes in high state when magnetic switch is triggered. Another question is has it a different behaviour for open and close?
So maybe you get just the information you can use device was "up"

senadaruc, 2019-03-04:
> Mhh, you could try to measure via a DMM the different GPIOs if one goes in high state when magnetic switch is triggered.
> [image]
> Another question is has it a different behaviour for open and close?
> So maybe you get just the information you can use device was "up"

This is the Voltage output while the device is connected to the internet and the magnetic switch is open!

[image]

senadaruc, 2019-03-04:
> This is the Voltage output while the device is connected to the internet and the magnetic switch is open!
>
> [image]

Same voltage results while the device is connected to the internet and the magnetic switch is closed!

senadaruc, 2019-03-04:
> Thanks for adding the photo's! I ordered a few of these also (have not received them yet). I was hoping they used the tywe3s board since the description listed "Tuya App" etc. I see from your photo's they also appear to have broken out the pins needed to flash firmware (as you have already done).
> 
> This will be my first adventure into Tasmota and I am hoping to use the "Tuya-Convert" method. If that doesn't work I will do as you did and use a serial to USB. One question (for anyone who knows). I see the Ground (G) and power (V) are marked. The other two are listed as N and S. Do you know which is Transmit and which is Receive? I'm new to this and wasn't sure (I'm guessing they are marked from another language?).
> 
> Thanks!
> DeadEnd
> 
> EDIT: Fixed "OTA" method to "Tuya-Convert" to be clear as to what I hope to do.

The N,G,S and V board pins/holes are not connected into TX,RX or VCC !! only G is ground.

S,

Jason2866, 2019-03-04:
Other GPIOs are not connected? Like Gpio 4, 12, 13 or 14?

cbmboy, 2019-03-04:
> The N,G,S and V board pins/holes are not connected into TX,RX or VCC !! only G is ground.

Thanks for the information!
I do wonder what they are for then...

cbmboy, 2019-03-04:
> Other GPIOs are not connected? Like Gpio 4, 12, 13 or 14?

There should be at least two correct? (Asking as I am still new and trying to learn)
One for the magnetic switch, and another for the LED?
Would there need to be a third also for the button (don't know what this does yet).

senadaruc, 2019-03-04:
> Other GPIOs are not connected? Like Gpio 4, 12, 13 or 14?

There is no voltage output from other GPIO's !! is there other way to test the GPIO's ?

senadaruc, 2019-03-04:
> No label on the interesting IC `U1` :-(
> It looks like it does have a step up power converter so the 2 AAA cells are no problem for providing correct voltage for the ESP

I managed to see the chip info ... it's 78474 .... see the picture below.

[image]

senadaruc, 2019-03-04:
Ok so I triggered the magnetic switch and tried to see which GPIO is giving me continuity connection between GND and GPIO output but unfortunately, none of them is closing the circuit!

Jason2866, 2019-03-04:
The magnetic switch does not trigger a Gpio from the ESP chip
The U1 is a MCU. This chip reacts on the magnetic switch and enables the ESP chip.
The ESP sends with orig. firmware just a message via wifi when activated...

senadaruc, 2019-03-04:
Aha ok so now I managed to get some ON/OFF messages at the Tasmota console using the Manzoku Strip Module but is not ideal since the default config is switch but I need a sensor, any help bout this?

senadaruc, 2019-03-04:
how to change the default timer for the sleep mode ??

cbmboy, 2019-03-04:
> The ESP sends with orig. firmware just a message via wifi when activated...

I have not worked in Tasmota yet (waiting for devices) but is this something that it can be setup to do?

Jason2866, 2019-03-05:
@senadaruc  Tasmota uses NO Deep Sleep, it is not possible/wanted by design. 
Just wifi sleep when enabled. See https://github.com/arendst/Sonoff-Tasmota/wiki/energy-saving

Jason2866, 2019-03-05:
@cbmboy  You get a Info from device (via mqtt) that it is (or was) online. This info can be used in your Home Automation solution. 
With Tasmota on this device you cant expect more, because ESP chip (where Tasmota is installed) **is not** powered most of the time...

atomi23, 2019-03-05:
thats nice , if it work with the knx version of tasmota , and maybe , it send knx group adress for open and close

Jason2866, 2019-03-05:
Please close this issue, from Tasmota software side cant nothing be done to support this or other battery powered devices.
Thx

Jason2866, 2019-03-05:
To clarify, with original software the ESP chip is just the wifi modul to transmit the messages from MCU `U1`

senadaruc, 2019-03-05:
Gents, I managed to configure using the "Manzoku strip Module" so the magnet open/close gives an output at MQTT console:

> 00:00:00 Project sonoff Sonoff Version 6.4.1(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 SmartMeraki in mode 11N as sonoff-0421...
00:00:00 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
00:00:00 RSL: stat/sonoff/POWER1 = OFF
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoff-0421.local with IP address 10.0.0.9
11:03:07 MQT: Attempting connection...
11:03:07 MQT: Connected
11:03:07 MQT: tele/sonoff/LWT = Online (retained)
11:03:07 MQT: cmnd/sonoff/POWER = 
11:03:07 MQT: tele/sonoff/INFO1 = {"Module":"Manzoku strip","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/DVES_000000000000099C1A5_fb/","GroupTopic":"sonoffs"}
11:03:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0421","IPAddress":"10.0.0.9"}
11:03:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}
11:03:07 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
11:03:07 MQT: stat/sonoff/POWER1 = OFF
11:03:07 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
11:03:07 MQT: stat/sonoff/POWER2 = OFF
11:03:07 MQT: stat/sonoff/RESULT = {"POWER3":"OFF"}
11:03:07 MQT: stat/sonoff/POWER3 = OFF
11:03:07 MQT: stat/sonoff/RESULT = {"POWER4":"OFF"}
11:03:07 MQT: stat/sonoff/POWER4 = OFF
11:03:15 MQT: tele/sonoff/STATE = {"Time":"2019-03-05T11:03:15","Uptime":"0T00:00:16","Vcc":3.376,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":111,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"SmartMeraki","BSSId":"E6:55:3D:C1:A0:44","Channel":6,"RSSI":84}}
11:03:33 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
11:03:33 MQT: stat/sonoff/POWER1 = **ON**
11:03:36 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
11:03:36 MQT: stat/sonoff/POWER1 = **OFF**
11:03:39 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
11:03:39 MQT: stat/sonoff/POWER1 = **ON**
11:03:51 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
11:03:51 MQT: stat/sonoff/POWER1 = **OFF**

> 

I this enough to have a healthy communication for open-close status?

S,

Jason2866, 2019-03-05:
The only viable information is 
`11:03:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0421","IPAddress":"10.0.0.9"}`
`11:03:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}`
Everything else are artefacts....

senadaruc, 2019-03-05:
> The only viable information is
> `11:03:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0421","IPAddress":"10.0.0.9"}`
> `11:03:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}`
> Everything else are artefacts....
> You will NEVER get with Tasmota a status wether it is Open or Close. Just something "happened"

Hmm ok but here I can see ON/OFF info when I open/close the magnet switch!

[image]

Maybe someone can help me to write some plugin who can detect those changes?

Thanks,

cbmboy, 2019-03-05:
With the discussion that has been done here (specifically that Tasmota is not designed for battery operation) I wonder if it would be a better idea to use a different approach.

Here is a link I discussed with @meingraham.
(original) https://www.youtube.com/watch?v=BoYVr2UwWWg
(updated) https://www.youtube.com/watch?v=2kLZ7DlP9KU

This is a DIY sensor that sounds to be designed very similar to these - power on = send signal, power off = send signal.  Not that tasmota couldn't achieve this somehow, but maybe this is a simpler solution.

@senadaruc - can you tell us if the device stays powered up when the switch is "open"?  And the sends a close signal before powering down?  If so this would match exactly what "the hook up" accomplished in the videos.

Thanks!
DeadEnd

senadaruc, 2019-03-05:
> With the discussion that has been done here (specifically that Tasmota is not designed for battery operation) I wonder if it would be a better idea to use a different approach.
> 
> Here is a link I discussed with @meingraham.
> (original) https://www.youtube.com/watch?v=BoYVr2UwWWg
> (updated) https://www.youtube.com/watch?v=2kLZ7DlP9KU
> 
> This is a DIY sensor that sounds to be designed very similar to these - power on = send signal, power off = send signal. Not that tasmota couldn't achieve this somehow, but maybe this is a simpler solution.
> 
> @senadaruc - can you tell us if the device stays powered up when the switch is "open"? And the sends a close signal before powering down? If so this would match exactly what "the hook up" accomplished in the videos. If these devices work the same way, we might be able to just reuse the code attached to the video's to "de-cloud" these devices.
> 
> Thanks!
> DeadEnd

After every open/close command, the device stays 1 minute connected to the WiFi sending ON/OFF with MQTT based ont the magnet open/close.

11:03:39 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
11:03:39 MQT: stat/sonoff/POWER1 = ON
11:03:51 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
11:03:51 MQT: stat/sonoff/POWER1 = OFF

S,

ascillato2, 2019-03-06:
Closing this issue as it has been answered. Currently unsupported device. Sorry.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2

stuermera, 2019-05-01:
> Gents, I managed to configure using the "Manzoku strip Module" so the magnet open/close gives an output at MQTT console at stat/sonoff/POWER1
> 
> ```
> 00:00:00 Project sonoff Sonoff Version 6.4.1(sonoff)-2_4_2
> 00:00:00 WIF: Connecting to AP1 SmartMeraki in mode 11N as sonoff-0421...
> 00:00:00 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
> 00:00:00 RSL: stat/sonoff/POWER1 = OFF
> 00:00:06 WIF: Connected
> 00:00:06 DNS: Initialized
> 00:00:06 HTP: Web server active on sonoff-0421.local with IP address 10.0.0.9
> 11:03:07 MQT: Attempting connection...
> 11:03:07 MQT: Connected
> 11:03:07 MQT: tele/sonoff/LWT = Online (retained)
> 11:03:07 MQT: cmnd/sonoff/POWER = 
> 11:03:07 MQT: tele/sonoff/INFO1 = {"Module":"Manzoku strip","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/DVES_000000000000099C1A5_fb/","GroupTopic":"sonoffs"}
> 11:03:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0421","IPAddress":"10.0.0.9"}
> 11:03:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}
> 11:03:07 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
> 11:03:07 MQT: stat/sonoff/POWER1 = OFF
> 11:03:07 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
> 11:03:07 MQT: stat/sonoff/POWER2 = OFF
> 11:03:07 MQT: stat/sonoff/RESULT = {"POWER3":"OFF"}
> 11:03:07 MQT: stat/sonoff/POWER3 = OFF
> 11:03:07 MQT: stat/sonoff/RESULT = {"POWER4":"OFF"}
> 11:03:07 MQT: stat/sonoff/POWER4 = OFF
> 11:03:15 MQT: tele/sonoff/STATE = {"Time":"2019-03-05T11:03:15","Uptime":"0T00:00:16","Vcc":3.376,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":111,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"SmartMeraki","BSSId":"E6:55:3D:C1:A0:44","Channel":6,"RSSI":84}}
> 11:03:33 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 11:03:33 MQT: stat/sonoff/POWER1 = ON
> 11:03:36 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
> 11:03:36 MQT: stat/sonoff/POWER1 = OFF
> 11:03:39 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 11:03:39 MQT: stat/sonoff/POWER1 = ON
> 11:03:51 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
> 11:03:51 MQT: stat/sonoff/POWER1 = OFF
> ```
> 
> I this enough to have a healthy communication for open-close status?
> 
> S,

Would you mind sharing your config line for this device with me? I have the same sensor.
Thanks!

stuermera, 2019-05-01:
@senadaruc perhaps you can share your config?

senadaruc, 2019-05-01:
Guys, I am on PTO until 7.May I will share when I am back at home.

stuermera, 2019-05-01:
If someone want to give it a try...

I was able to flash [the same device](https://www.amazon.de/gp/product/B07GXTV4C1/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1) over-the-air (OTA) with the TUYA-convert method as described [here](https://www.heise.de/ct/artikel/Tuya-Convert-IoT-Geraete-ohne-Loeten-vom-Cloud-Zwang-befreien-4283623.html).

This is my configuration which I am using on ioBroker on a Raspberry Pi - together with "Yahka" - a Apple Homekit application...
[image]

senadaruc, 2019-05-02:
> If someone want to give it a try...
> 
> I was able to flash [the same device](https://www.amazon.de/gp/product/B07GXTV4C1/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1) over-the-air (OTA) with the TUYA-convert method as described [here](https://www.heise.de/ct/artikel/Tuya-Convert-IoT-Geraete-ohne-Loeten-vom-Cloud-Zwang-befreien-4283623.html).
> 
> This is my configuration which I am using on ioBroker on a Raspberry Pi - together with "Yahka" - an Apple Homekit application...
> 
> [image]
> 
> And so far it seems to work - here's a [short video](https://youtu.be/1VurNEd0Lss).

Wow nice, really nice did you tried testing with deep sleep mode since this device is sleeping after some time!

Thanks,

stuermera, 2019-05-02:
That's the "small" problem so far...

As soon as the device is in deep-sleep, triggering it by opening the door doesn't trigger the value from "false" to "true" - it only works as long as the device isn't in deep-sleep.
So waking it up from deep-sleep does literally nothing - but the next trigger works again because it isn't in deep-sleep anymore.

If somehow the first trigger from a deep-sleep can trigger the value from "false" to "true" it would be working 100%.

senadaruc, 2019-05-02:
Ok so I installed the iObroker and Yahka, now I need to configure also the MQQT inside of the iObroker  ?

stuermera, 2019-05-02:
I am using "sonoff" plugin instead of the MQQT plugin. The sonoff plugin is compatible with it.

senadaruc, 2019-05-02:
ok let me try..

senadaruc, 2019-05-02:
> I am using "sonoff" plugin instead of the MQQT plugin. The sonoff plugin is compatible with it.

Which firmware you have in your device, Tasmota or something else?

stuermera, 2019-05-02:
The latest Tasmota firmware, 6.5.0.

senadaruc, 2019-05-02:
Ok, and what did you select at tasmota module config for this device?

stuermera, 2019-05-02:
I created an own template based on (17) "WiOn".
I set GPIO3 to Button1 - and GPIO5 to Relay1.

senadaruc, 2019-05-02:
Wow nice :)  so can you share your template config?

stuermera, 2019-05-02:
I am at work at the moment... a few hours to wait.

senadaruc, 2019-05-02:
No problem I am also at PTO I can wait.... thanks.

meingraham, 2019-05-02:
I doubt that the "1st wake from deep sleep" issue will be resolved, but it
probably won't hurt to try this. OTA update the firmware to the latest
development branch codebase (currently 6.5.0.9). Pre-compiled binaries for
this are located at  http://thehackbox.org/tasmota (note no '/release').
You can choose the ESP core that best suits your Wi-Fi connectivity
situation. There are quite a few issues resolved since 6.5.0. None "deep
sleep" specific (since Tasmota doesn't really support deep sleep), but
nevertheless, some "strange" behaviors are addressed. If updating to the
latest issue causes unforeseen issues with your device, you can always
downgrade to the master release (6.5.0).

Also, I'm not sure if you can trap system#boot in a rule as a way to detect
"1st wake from deep sleep". Perhaps investigate any of the other status
messages (INFO<x> x 1..3, STATE, or UPTIME) to see if they occur only on
"1st wake". If so, your rule can trap one of those. Then, if the rule
triggers, you can force sending a state update that is missed in this case
(i.e., switching from false to true)

Mike

Michael Ingraham
Spartanburg, SC

stuermera, 2019-05-02:
@senadaruc 
> No problem I am also at PTO I can wait.... thanks.

Ok, here we go.
[Config_Door-Sensor.zip](https://github.com/arendst/Sonoff-Tasmota/files/3138103/Config_Door-Sensor.zip)

senadaruc, 2019-05-02:
> @senadaruc
> 
> > No problem I am also at PTO I can wait.... thanks.
> 
> Ok, here we go.
> [Config_Door-Sensor.zip](https://github.com/arendst/Sonoff-Tasmota/files/3138103/Config_Door-Sensor.zip)

Thanks,

stuermera, 2019-05-02:
@meingraham 
> Also, I'm not sure if you can trap system#boot in a rule as a way to detect "1st wake from deep sleep". Perhaps investigate any of the other status messages (INFO<x> x 1..3, STATE, or UPTIME) to see [...]

Thanks, I've tried the 6.5.0.9 sensors version. OTA upgrade takes 2-3 times on this device :)

This is what happens, when I trigger it from deep sleep and open/close it several times:

`14:18:57 MQT: Attempting connection...
14:18:57 MQT: Connected
14:18:57 MQT: tele/sonoff/LWT = Online (retained)
14:18:57 MQT: cmnd/sonoff/POWER = 
14:18:57 MQT: tele/sonoff/INFO1 = {"Module":"Door Sensor","Version":"6.5.0.9(e592fa9-sensors)","FallbackTopic":"cmnd/Türsensor_fb/","GroupTopic":"sonoffs"}
14:18:57 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"Türsensor","IPAddress":"10.0.0.110"}
14:18:57 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}
14:18:57 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
14:18:57 MQT: stat/sonoff/POWER = OFF
14:19:05 MQT: tele/sonoff/STATE = {"Time":"2019-05-02T14:19:05","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WirelessLAN","BSSId":"B0:BE:76:A7:F7:BB","Channel":5,"RSSI":70,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:19:08 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
14:19:08 MQT: stat/sonoff/POWER = ON
14:19:34 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
14:19:34 MQT: stat/sonoff/POWER = OFF
14:19:35 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
14:19:35 MQT: stat/sonoff/POWER = ON
14:19:44 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
14:19:44 MQT: stat/sonoff/POWER = OFF`

Where this part appears everytime I trigger it from deep sleep:

`14:18:57 MQT: Attempting connection...
14:18:57 MQT: Connected
14:18:57 MQT: tele/sonoff/LWT = Online (retained)
14:18:57 MQT: cmnd/sonoff/POWER = 
14:18:57 MQT: tele/sonoff/INFO1 = {"Module":"Door Sensor","Version":"6.5.0.9(e592fa9-sensors)","FallbackTopic":"cmnd/Türsensor_fb/","GroupTopic":"sonoffs"}
14:18:57 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"Türsensor","IPAddress":"10.0.0.110"}
14:18:57 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}
14:18:57 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
14:18:57 MQT: stat/sonoff/POWER = OFF`

senadaruc, 2019-05-02:
> I doubt that the "1st wake from deep sleep" issue will be resolved, but it probably won't hurt to try this. OTA update the firmware to the latest development branch codebase (currently 6.5.0.9). [...]

I this the link for the 6.5.0.9 version?
https://github.com/reloxx13/Sonoff-Tasmota-Modified/releases

Thanks,

stuermera, 2019-05-02:
I took this one... with unreleased 6.5.0.9
http://thehackbox.org/tasmota/sonoff-sensors.bin
The link from you is the release versions with 6.5.0.

[47 more replies]