arendst/Tasmota #5390: TSL2561: Precompiled .bin for Wemos D1 Mini

heikoh81, 2019-03-03:
Hi,

I'd like to use a TSL2561 on a Wemos D1 Mini.
In the Wiki, it states:
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-TSL2561-Luminosity-Sensor
`before the TSL2561 can be used with Tasmota, make sure it's enabled. To enable it, remove the forward slashes before this line before you compile and upload tasmota to your Wemos device.`

Until now, I have only used precompiled .bin-files.
Is there one available with support for TSL2561 enabled?

Greetings,
Heiko

joba-1, 2019-03-04:
Yes there is: sonoff-sensors.

Here is a table of which binary image supports what:
https://github.com/arendst/Sonoff-Tasmota/releases

-- 
iphone so i typo

Jason2866, 2019-03-04:
Thx, @heikoh81 for the info, wiki updated

helmar74, 2019-03-05:
@heikoh81 :
I'm also runniung 2 Wemos Mini with a TSL2561  connected.
I'm using a image, which is compiled by my own and I'm suffering in High LOAD on the wemos mini.
See #5392

Do you have the same issue?

ascillato2, 2019-03-06:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2

heikoh81, 2019-03-06:
Thank you for your answers.

> @heikoh81 :
> I'm also running 2 Wemos Mini with a TSL2561 connected.
> I'm using a image, which is compiled by my own and I'm suffering in High LOAD on the wemos mini.
> See #5392
> 
> Do you have the same issue?

I have just ordered these sensors at AliExpress, so I will have to wait 1-2 month for the sensors to arrive.
I'll give an update then.
Where can i see the LOAD level?

helmar74, 2019-03-07:
@heikoh81 

You see the current load (LoadAvg) in the state Topic, which is sent every teleperiod time (default 5 Minutes):
'
07:28:56 MQT: tele/sonoffWemosMini_2/STATE = {"Time":"2019-03-07T07:28:56","Uptime":"0T00:05:16","Vcc":2.984,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":4,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXX","BSSId":"F0:B0:14:6F:B5:4F","Channel":1,"RSSI":54}}
'
or you send command "status 11" on webconsole:
'
07:30:19 MQT: stat/sonoffWemosMini_2/STATUS11 = {"StatusSTS":{"Time":"2019-03-07T07:30:19","Uptime":"0T00:06:39","Vcc":2.984,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":6,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXX","BSSId":"F0:B0:14:6F:B5:4F","Channel":1,"RSSI":54}}}'