arendst/Tasmota #5391: Eltako Wind Sensor reopen issue

tremichl, 2019-03-04:
Please could be issue #3418 regarding Eltako Wind Sensor reopened. Integrating necessary calculations for this sensor into Tasmota would be very appreciated.

Thanks, Michael

**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-04:
There is already a Wind Sensor supported https://github.com/arendst/Sonoff-Tasmota/pull/3146
So if you want to support this type of sensor a PR from you is welcome.

ascillato2, 2019-03-06:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2

tremichl, 2019-03-06:
Sorry Guys, there is an answer, but it did not answer the question, so this issue should not be closed. TX20 Sensor is another product and its interface is not compatible with Eltako sensor. You only can get pulses no serial data. Eltako Sensor is mechanical more stable and reliable than TX20 and of course more expensive.  And I have it already mounted and no need to get wind direction. That is what I asked for, doing the calculation to get wind speed in Tasmota. Maybe there is an solution. Unfortunately my skills do not allow to do that on my own. As I can see there are some more interested in this.   Tasmota is really a great project! Thanks!

Michael

ascillato, 2019-03-06:
>There is an answer, but it did not answer the question, so this issue should not be closed.

Hi, you asked for adding a new device and the answer was that it is not going to be added/worked soon by us, but if you or anyone having this sensor is willing to work on it and make a driver for Tasmota, we will gladly help on that PR.

Sorry if the answer is not what you have expected. We can't buy and work on all sensors requested but we are willing and happy to receive Pull Requests for new features.

Thanks for sharing your ideas.

tremichl, 2019-03-06:
Hi, thanks for explanation.