arendst/Tasmota #5392: High LOAD with TSL-2561 connected

helmar74, 2019-03-04:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used :
- [X] Hardware used : Wemos MiniD, 18B20, SR-501 and TSL-2561
- [ ] If a pre-compiled release or development binary was used, which one? :
- [X] You have tried latest release or development binaries? : both
- [ ] Provide the output of command ``status 0`` :
```
17:52:03 CMD: status 0
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS = {"Status":{"Module":18,"FriendlyName":["sonoffWemosMini4"],"Topic":"sonoffWemosMini_4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffStandardGruppe","OtaUrl":"http://192.168.1.2:5555/data/firmwares/sonoff.bin","RestartReason":"External System","Uptime":"0T00:25:58","StartupUTC":"2019-03-04T16:26:06","Sleep":50,"CfgHolder":4617,"BootCount":28,"SaveCount":1398,"SaveAddress":"FB000"}}
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS2 = {"StatusFWR":{"Version":"6.4.1.19(sonoff)","BuildDateTime":"2019.02.28 10:00:23","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000407","0FAAE59C","00068FA0","23BA1FCE","000013C0"]}}
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini-4","IPAddress":"192.168.1.119","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"BC:DD:C2:08:C5:81","Webserver":2,"WifiConfig":4}}
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"sonoffWemosMini-1","MqttClient":"sonoffWemosMini-1","MqttUser":"DVES_USER","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS7 = {"StatusTIM":{"UTC":"Mon Mar 04 16:52:04 2019","Local":"Mon Mar 04 17:52:04 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:45","Sunset":"17:56"}}
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS10 = {"StatusSNS":{"Time":"2019-03-04T17:52:04","Switch1":"OFF","DS18B20":{"Temperature":-21.6},"TSL2561":{"Illuminance":0.233},"TempUnit":"C"}}
17:52:04 MQT: stat/sonoffWemosMini_4/STATUS11 = {"StatusSTS":{"Time":"2019-03-04T17:52:04","Uptime":"0T00:25:58","Vcc":3.067,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":522,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SSID","BSSId":"E8:DF:70:67:B9:00","Channel":6,"RSSI":70,"LinkCount":1,"Downtime":"0T00:00:02"}}}
```

Hi, 
I have a Wemos Mini D with  18B20, LED, TSL-2561 and a SR501 PIR Sensor connected.
The Load is very high and the wemos is responding slowly. Also it isn't sending teleperiod frequently.
Tried latest stable and also development release.
If I disconnect the TSL-2561 sensor, the load is fine. 
Any idea what I can do, to use TSL-2561 with "normal" load.

What I've tried:
- setting sleep to 250
- setting setoption60 to 1

Nothing helped.

Thanks a lot in advance

helmar74, 2019-03-05:
Now the LOAD is between 19-22 without changing anything.
Really strange.... 
Closing the issue

closed 2019-03-05 by helmar74

joba-1, 2019-03-05:
Hi Helmar,

just for your information: your status log shows it is very dark (<1lux).
That means, a light measurement takes very long (~0.5s). My driver does a
delay() during that time, which may account for high load values depending
on how load is measured. I plan doing such measurements asynchronously to
fix that, but it will take some time.

Joachim

-- 
iphone so i typo

helmar74, 2019-03-05:
Hi joba-1,

thanks for the information.
Great to hear, that you're planning to fix the issue.

Thanks a lot

Helmar

helmar74, 2019-03-16:
Moved over to MAX44009 Sensor, which is measuring well and without high Load