arendst/Tasmota #5393: LCD Display-Commands

BigWumpus, 2019-03-04:
**Describe the bug**
Adressing a cursor-position in "c"olumn 0 or "l"ine 0 is impossible, it places the text in the upper left corner, l0c0

```
21:50:17 CMD: status 0
21:50:17 MQT: stat/Home-Display/STATUS = {"Status":{"Module":18,"FriendlyName":["Home-Display"],"Topic":"Home-Display","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:50:17 MQT: stat/Home-Display/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-display.bin","RestartReason":"Exception","Uptime":"0T00:14:33","StartupUTC":"2019-03-04T20:35:44","Sleep":10,"BootCount":31,"SaveCount":59,"SaveAddress":"F9000"}}
21:50:17 MQT: stat/Home-Display/STATUS2 = {"StatusFWR":{"Version":"6.4.1(release-display)","BuildDateTime":"2018-12-24T15:54:34","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:50:17 MQT: stat/Home-Display/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["@home",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
21:50:17 MQT: stat/Home-Display/STATUS4 = {"StatusMEM":{"ProgramSize":486,"Free":516,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FAAA59C","00038FA0","23B617CA","00003BC0"]}}
21:50:17 MQT: stat/Home-Display/STATUS5 = {"StatusNET":{"Hostname":"Home-Display","IPAddress":"192.168.178.18","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"CC:50:E3:0F:1E:91","Webserver":2,"WifiConfig":4}}
21:50:17 MQT: stat/Home-Display/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.11","MqttPort":8883,"MqttClientMask":"Home-Display","MqttClient":"Home-Display","MqttUser":"HA","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:50:17 MQT: stat/Home-Display/STATUS7 = {"StatusTIM":{"UTC":"Mon Mar 04 20:50:17 2019","Local":"Mon Mar 04 21:50:17 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:26","Sunset":"18:37"}}
21:50:17 MQT: stat/Home-Display/STATUS10 = {"StatusSNS":{"Time":"2019-03-04T21:50:17"}}
21:50:17 MQT: stat/Home-Display/STATUS11 = {"StatusSTS":{"Time":"2019-03-04T21:50:17","Uptime":"0T00:14:33","Vcc":2.727,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"POWER":"ON","Wifi":{"AP":1,"SSId":"@home","BSSId":"5C:49:79:38:A1:02","Channel":13,"RSSI":100}}}
```

**To Reproduce**
Displaytext [l1c0]Test10
Displaytext [l2c5]Test25

I tried 6.4.1 with Core 2.3.0 and Core 2.4.2 on a Wemos D1 Mini-Clone with a standard 20x4-LCD with common I²C-Interfaceboard

wongnam, 2019-03-05:
Try to set for LCD20x4
* DisplayCols 20 
* DisplayRows 4

curzon01, 2019-03-05:
The parameter for `l` and `c` cmnds starts from 1 so the lowest address is l1c1, see link from @wongnam

BigWumpus, 2019-03-05:
No, it is not written so.
I can succesful adress the second line with l1, the third line with l2 and the fourth line with l3 at any col, exept c0...
But adressing l0 with any c (or c0 with any l) leeds to writing the text at l0c0...

arendst, 2019-03-06:
Try the latest dev release (6.4.1.19) and report back.

curzon01, 2019-03-07:
using 6.4.1.19 (14afdba) with SSD1306 `l0c0` is identical with `l1c1` - first col first row independently of DisplaySize

arendst, 2019-03-08:
OK. So let's say that a display has 10 columns starting with position 1 to 10 and 10 lines starting with line 1 to 10.

On a LCD display to address column 1 and line 1 you will need to use [c0l0]. so the LCD positions are offset from 0.
On a TFT display to address column 1 and line 1 you will need to use [c1l1]. so the TFT does not use a offset but it appears that [c0l0] does exactly the same as [c1l1].

It's more like a feature than a bug. If I change it now it would change current users layout so I think we'll have to live with it.

gemu2015, 2019-03-08:
c and l are meant for column and line and range from 1 to N
you must specify both lager then zero in one command, else  x and y are chosen
x and y are meant for pixel accurate displays and have a range from 0 to N

which mode is used is flagged to drawstring in dsp_flag
so we have to change LcdDrawStringAt in xdsp_01_lcd.ino
although pixel accurate setup makes no sense in character lcds.

void LcdDrawStringAt(void)
{
 if (dsp_flag)  lcd->setCursor(dsp_x-1, dsp_y-1);
 else  lcd->setCursor(dsp_x, dsp_y);
  lcd->print(dsp_str);
}

curzon01, 2019-03-08:
@arendst - agree backward compatibility; I noted that in [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/displays#displaytext-use) - so from my side it's solved, wait for feedback from BigWumpus

gemu2015, 2019-03-08:
@BigWumpus 
as i said if either column or line is zero x and y are chosen.
so if @arendst  does not want to change source you may address lcd position with x and y starting from 0 x and y being column and line on an lcd

closed 2019-03-10 by ascillato2