arendst/Tasmota #5394: Build error: "Error: no handler found"

kobuki, 2019-03-05:
**Describe the bug**
Directly starting a build, eg. `env:sonoff` or `env:sonoff-classic` results in the following error twice in the Output window at the start of the build process: `Error: no handler found`. The build is run succesfully and I have a .bin, but it's confusing and hints at missing configuration bits.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ :
- [x] _Development/Compiler/Upload tools used_ : vscode + pio
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` : sorry, don't know where to give the command
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
Click on a build task, for instance on env:sonoff

**Expected behavior**
I receive no errors in the Output window at the start of the build.

**Additional context**
See description.

Jason2866, 2019-03-05:
It is a VSC issue. Not Tasmota related.

kobuki, 2019-03-05:
Thanks, I'll try.

closed 2019-03-06 by ascillato2