arendst/Tasmota #5396: Tasmota changing "POWER2" to "POWER1"

JetDriver, 2019-03-05:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ X] Development/Compiler/Upload tools used :
- [ ] Hardware used: Sonoff TH16 (ver 2.1)
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```15:14:00 CMD: status 0
15:14:00 MQT: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Greenhouse Test","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:14:00 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:24","StartupUTC":"2019-03-05T21:13:36","Sleep":50,"BootCount":79,"SaveCount":495,"SaveAddress":"F7000"}}
15:14:01 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
15:14:01 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Emily",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
15:14:01 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
15:14:01 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7205","IPAddress":"192.168.0.185","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:0D:DC:25","Webserver":2,"WifiConfig":5}}
15:14:01 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.162","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0DDC25","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:14:01 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Mar 05 21:14:01 2019","Local":"Tue Mar 05 15:14:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"-06:00","Sunrise":"00:24","Sunset":"11:39"}}
15:14:01 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-03-05T15:14:01"}}
15:14:01 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-03-05T15:14:01","Uptime":"0T00:00:25","Vcc":3.438,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Emily","BSSId":"50:C7:BF:F3:7F:9B","Channel":2,"RSSI":100}}}

**(Please, remember to close the issue when the problem has been addressed)**
Using Node-Red to MQTT on Sonoff TH16 (ver 2.1) to control relays on GPIOs 4, 1 and 3, in addition to the Sonoff's relay. If I send topic/payload "cmd/sonoff/POWER2":"ON" the debug window shows exactly that output but Tasmota Console reports "stat/sonoff/RESULT = {"POWER1":"ON"}" which, of course, turns on the wrong relay. If I type "power2 1" into the console it works as expected. Is Tasmota somehow changing "POWER2" to "POWER1"?

From Node-Red debug window:
3/5/2019, 1:16:33 PMnode: 64d53a17.186ee4
cmd/sonoff/POWER2 : msg.payload : string[2]
"ON"
3/5/2019, 1:16:40 PMnode: 64d53a17.186ee4
cmd/sonoff/POWER2 : msg.payload : string[3]
"OFF"

From Tasmota console:
13:22:13 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
13:22:13 MQT: stat/sonoff/POWER1 = ON
13:22:16 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
13:22:16 MQT: stat/sonoff/POWER1 = OFF

ascillato, 2019-03-05:
You are sending

cmd/sonoff/POWER2 on

And should be 

cmnd/sonoff/POWER2 on

JetDriver, 2019-03-06:
Thanks, Adrian. It looks like the problem was the contributed node I was
using. It translates CMD to CMND. But it was also causing POWER1 to be
triggers when POWER2 was. After removing that node from the flow debug is
now agreeing with Tasmota console and toggle buttons. Appreciate your help.

Alan Branch

ascillato, 2019-03-06:
Great.

Please, close this issue. Thanks

JetDriver, 2019-03-06:
Willco. Thanks.

closed 2019-03-06 by JetDriver