arendst/Tasmota #5397: Negative Lat/Lon -1<>0 not supported

JamesCohen-awin, 2019-03-05:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff Basic
- [x] _Development/Compiler/Upload tools used_ : n/a
- [x] _If a pre-compiled release or development binary was used, which one?_ : sonoff.bin v6.4.1
- [x] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Spare Bedroom Lamp"],"Topic":"spare_bedroom_lamp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"3T00:11:57","StartupUTC":"2019-03-02T21:19:14","Sleep":50,"BootCount":9,"SaveCount":121,"SaveAddress":"FB000"}}
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":300,"SetOption":["000A8009","558180C0","00000201"]}}
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS5 = {"StatusNET":{"Hostname":"spare_bedroom_lamp-1544","IPAddress":"10.1.2.7","Gateway":"10.1.2.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.2.1","Mac":"DC:4F:22:92:C6:08","Webserver":2,"WifiConfig":4}}
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.3.18","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_92C608","MqttUser":"tasmota","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Mar 05 21:31:11 2019","Local":"Tue Mar 05 22:31:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:32","Sunset":"18:42"}}
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS10 = {"StatusSNS":{"Time":"2019-03-05T22:31:11"}}
22:31:11 MQT: spare_bedroom_lamp/stat/STATUS11 = {"StatusSTS":{"Time":"2019-03-05T22:31:11","Uptime":"3T00:11:57","Vcc":3.478,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"...","Channel":11,"RSSI":54}}}
```

**To Reproduce**
 - Enter `Latitude -0.09` at the device web console
 - `stat/RESULT` and stored value will be (positive) `0.09`

**Expected behavior**
 - A latitude value of `-0.09` should be stored and returned 

**Additional context**
I tried some additional testing and the command `Latitude -1.0` works as expected

**(Please, remember to close the issue when the problem has been addressed)**

DavinKD, 2019-03-05:
https://github.com/arendst/Sonoff-Tasmota/issues/5386

JamesCohen-awin, 2019-03-05:
Duplicate of #5386

closed 2019-03-05 by JamesCohen-awin