arendst/Tasmota #5398: Tuya PIR Sensor XMD-101.app / NX-4469-675

hkspks, 2019-03-05:
I flashed the PIR-Sensor XMD-101.app / NX-4469-675 [1] from pearl using tuya-convert. I tried different settings/ templates I found for other PIRs, but they aren't working:
- Template: Sonoff Basic, GPIO14 Sensor -> Switch2 or Switch 4
- Template: Generic, GPIO14 Sensor -> Switch2 or Switch4

In the console, I cannot see a reaction when the PIR triggers (there is a blue LED blinking when movement is recognized). As soon as the PIR boots, it writes to the console: 
MQT: tele/sonoff/SENSOR = {"Time":"2019-03-05T23:21:16","Switch2":"ON"}

But there is no reaction to movements afterwards. Can anyone give me a hint how to get this device running?

Thank you!

[1 ]https://www.luminea.info/WiFi-PIR-Bewegunsmelder-Alexa-kompatibel-welt-NX-4469-919.shtml

ascillato2, 2019-03-06:
Hi,

You will need to open the device and follow the tracks to know which GPIO is connected to.

If you need further assistance on this, please find us in the Tasmota Support Chat. Then a template can be made for other users.

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2

hkspks, 2019-03-06:
Following findings from the Tasmota Support chat:
- Trial and Error using Switch1-8/ Switch1-8n on the GPIOs didn't work - there is no reaction on triggering the PIR
- Referring to  https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices there may be "STM32 as 2nd MCU" for the PIR
- Images of the board are attached

Can anyone help/ guide how to get this running?

[image]
[image]
[image]

hkspks, 2019-03-06:
Due to the 2nd MCU, this device is currently not supported by tasmota.

JohanPotgieter, 2019-09-29:
Is this device supported now? Any plans to do it in the future?

ascillato, 2019-09-29:
Not supported. Sorry.

JohanPotgieter, 2019-09-30:
Thank you for your reply. I know this is outside the scope but do you have any idea how I can get started to write something myself and Arduino IDE to make this sensor work or if anyone has done any work with custom firmware for these sensors.

ascillato, 2019-09-30:
As this has another MCU, you need to make reverse engineering on the communication between this second MCU and the ESP8266 and make a new driver for that.

jlegen, 2019-10-19:
> Thank you for your reply. I know this is outside the scope but do you have any idea how I can get started to write something myself and Arduino IDE to make this sensor work or if anyone has done any work with custom firmware for these sensors.

you can try starting with a similar device - the door sensor seems to work very similar, and is just made usable:
https://github.com/arendst/Sonoff-Tasmota/issues/6624

jssting, 2020-01-18:
I was able to get this done thanks to the supplied info and some reference info on the HA page. 
My post is available here on HA https://community.home-assistant.io/t/hkwl-ms03w-motion-sensor-with-tasmota/152421/21?u=jssting

daniel-lo, 2021-09-01:
TuyaConvert doesn't work for me .. is there any instruction how to solder flash this device?