arendst/Tasmota #5399: Time#Minute not working

hwarang098tkd, 2019-03-05:
I am trying to set this rule with no result. I cant understand what i am doing wrong.
Command: Rule1 on Time#Minute==87 do power 0 endon
```
01:26:23 CMD: Rule1 on Time#Minute==87 do power 0 endon
01:26:23 SRC: WebConsole from 192.168.1.30
01:26:23 RSL: Received Topic /Rule1, Data Size 35, Data on Time#Minute==87 do power 0 endon
01:26:23 RSL: Group 0, Index 1, Command RULE, Data on Time#Minute==87 do power 0 endon
01:26:23 MQT: stat/sonoff_b1_1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":476,"Rules":"on Time#Minute==87 do power 0 endon"}
01:26:23 CFG: Saved to flash at F9, Count 1845, Bytes 3584
01:26:27 APP: Serial logging disabled
01:26:36 WIF: Checking connection...
01:26:36 WIF: Connected
01:26:46 MQT: tele/sonoff_b1_1/STATE = {"Time":"2019-03-06T01:26:46","Uptime":"0T00:10:20","Vcc":3.720,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":81,"POWER":"ON","Dimmer":100,"Color":"000000FF00","HSBColor":"0,0,0","Channel":[0,0,0,100,0],"CT":153,"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"David","BSSId":"A4:B1:E9:9C:17:B0","Channel":1,"RSSI":54}}
01:26:56 WIF: Checking connection...
01:26:56 WIF: Connected
01:27:16 WIF: Checking connection...
01:27:16 WIF: Connected
```

Not working

ascillato, 2019-03-06:
Sorry, time#minute can't be used that way. It is just a trigger to execute every minute. Sorry.

ascillato, 2019-03-06:
Hi, 

Are you waiting 87 minutes from midnight? 

From wiki:

Time#Minute==241	every day once at 04:01 (= 241 minutes after midnight)

ascillato, 2019-03-06:
Will investigate. Thanks.

ascillato2, 2019-03-06:
Hi,

Tested and works using `Time#Minute=91`

Please use = instead of ==

Console output:

```
01:30:45 CMD: rule on time#minute=91 do power1 1 endon
01:30:45 SRC: WebConsole from 192.168.1.111
01:30:45 RSL: Topic Recibido /rule, Tamaño de Datos 35, Datos on time#minute=91 do power1 1 endon
01:30:45 RSL: Grupo 0, Índice 1, Comando RULE, Datos on time#minute=91 do power1 1 endon
01:30:45 MQT: stat/luzescalera/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":476,"Rules":"on time#minute=91 do power1 1 endon"}
01:30:46 CFG: Grabado a la flash en F6, Conteo 518, Bytes 3584
01:30:53 WIF: Probando conexión...
01:30:53 WIF: Conectado
01:31:00 RUL: TIME#MINUTE=91 performs "power1 1"
01:31:00 SRC: Rule
01:31:00 RSL: Topic Recibido /power1, Tamaño de Datos 1, Datos 1
01:31:00 RSL: Grupo 0, Índice 1, Comando POWER, Datos 1
01:31:00 MQT: stat/luzescalera/RESULT = {"POWER1":"ON"}
01:31:00 MQT: stat/luzescalera/POWER1 = ON
```

Closing this issue as it has been answered.

If you need further assistance, please find us in the Tasmota Support Chat. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2

hwarang098tkd, 2019-03-06:
> Please use = instead of ==

Solved. please change it on wiki

meingraham, 2019-03-06:
Or there may be an underlying coding issue since "==" is supposed to be the operator for numerical comparison?

```
Comparison operators:

"=", ">", "<", "|"
"==", ">=", "<=", "!=" (added in 6.4.1.14)

"=" operator is used for string comparison
"==" operator is used for numerical comparison
```

Mike

SylvainGa, 2020-01-29:
It's still '=' so either the code or the documentation still needs to be fixed because one of them is still wrong.

ascillato, 2020-01-29:
docs fixed. thanks.

SylvainGa, 2020-01-29:
https://github.com/arendst/Tasmota/wiki/Rules still has Time#Minute==241. Does it takes time to appear?

meingraham, 2020-01-29:
@SylvainGa 

Tasmota documentation is now at https://tasmota.github.io/docs/#/Home. The wiki site (versions prior to Tasmota 7) is no longer being updated.

ascillato2, 2020-01-29:
Wiki fixed