arendst/Tasmota #5406: Delete sonoff RF remote pairing

brafster, 2019-03-06:
I had an RF pairing on my sonoff RF before flashing with tasmota. After flashing, the remote does not work each time. This isn't such a big issue as I can automate it through HASS, but then I would need to remove the remote pairing from the device. I hold in the button for 5 seconds and then receive the two red LED flashes. After this the remote still works though (sometimes). 

Any ideas?

Jason2866, 2019-03-11:
It seems the RF receiver (hardware NOT under control of Tasmota) is faulty.
Deactivate (=desolder) it to get rid of it.
If you buy a RF-Bridge you can control every mqtt capable device via mqtt over HASS via RF-Remotes.

ascillato2, 2019-03-11:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-11 by ascillato2

mu2004, 2019-03-16:
I am having exactly the same problem on Sonoff RF with Tasmota. The 5 second 2 blinks of LED doesn't clear the remote control binding. Is there a way to work around this or clear the memory completely on RF?