arendst/Tasmota #5407: SonoffBasic restart every few minutes 6.4.1(sonoff)

lalo-uy, 2019-03-06:
**Describe the bug**
The device restart every few minutes

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ X] _Searched the problem in issues and in the wiki_
- [ X] _Hardware used_ :   Sonoff Basic R1
- [X ] _Development/Compiler/Upload tools used_ :  esptool.py  erased before upload
- [ ] _If a pre-compiled release or development binary was used, which one?_ : precompiled bin
- [ ] _You have tried latest release or development binaries?_ : 6.4.1
- [ ] _Provide the output of command_``status 0`` :

Mar 06 12:58:21 stat/Carb2/POWER ON
Mar 06 12:58:25 cmnd/Carb2/Status 0
Mar 06 12:58:25 stat/Carb2/STATUS {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"Carb2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
Mar 06 12:58:25 stat/Carb2/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:11","StartupUTC":"","Sleep":50,"BootCount":657,"SaveCount":1259,"SaveAddress":"F9000"}}
Mar 06 12:58:26 stat/Carb2/STATUS2 {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
Mar 06 12:58:26 stat/Carb2/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["oso","vega"],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
Mar 06 12:58:26 stat/Carb2/STATUS4 {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
Mar 06 12:58:26 stat/Carb2/STATUS5 {"StatusNET":{"Hostname":"Carb2-1579","IPAddress":"192.168.100.14","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.200","Mac":"84:F3:EB:AF:A6:2B","Webserver":2,"WifiConfig":2}}
Mar 06 12:58:26 stat/Carb2/STATUS6 {"StatusMQT":{"MqttHost":"192.168.100.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AFA62B","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
Mar 06 12:58:26 stat/Carb2/STATUS7 {"StatusTIM":{"UTC":"Thu Jan 01 00:05:12 1970","Local":"Thu Jan 01 00:05:12 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
Mar 06 12:58:26 stat/Carb2/STATUS10 {"StatusSNS":{"Time":"1970-01-01T00:05:12"}}
Mar 06 12:58:26 stat/Carb2/STATUS11 {"StatusSTS":{"Time":"1970-01-01T00:05:12","Uptime":"0T00:05:11","Vcc":3.410,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"oso","BSSId":"C8:3A:35:EA:48:B8","Channel":7,"RSSI":94}}}
Mar 06 12:58:44 tele/Carb2/STATE {"Time":"1970-01-01T00:05:30","Uptime":"0T00:05:30","Vcc":3.410,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"oso","BSSId":"C8:3A:35:EA:48:B8","Channel":7,"RSSI":92}}

As you can see in the log it goes offline for variable times from seconds to a couple of minutes
result of topic +/Carb2/+

Mar 06 13:29:52 tele/Carb2/LWT Offline
Mar 06 13:29:55 tele/Carb2/LWT Online
Mar 06 13:29:55 cmnd/Carb2/POWER (null)
Mar 06 13:29:56 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:29:56 stat/Carb2/POWER ON
Mar 06 13:30:09 tele/Carb2/STATE {"Time":"2019-03-06T17:30:07","Uptime":"0T00:20:26","Vcc":3.427,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"oso","BSSId":"C8:3A:35:EA:48:B8","Channel":7,"RSSI":100}}
Mar 06 13:30:52 tele/Carb2/LWT Offline
Mar 06 13:31:27 tele/Carb2/LWT Online
Mar 06 13:31:27 cmnd/Carb2/POWER (null)
Mar 06 13:31:27 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:31:27 stat/Carb2/POWER ON
Mar 06 13:31:54 tele/Carb2/LWT Offline
Mar 06 13:35:48 tele/Carb2/LWT Online
Mar 06 13:35:48 cmnd/Carb2/POWER (null)
Mar 06 13:35:48 tele/Carb2/INFO1 {"Module":"Sonoff Basic","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/DVES_AFA62B_fb/","GroupTopic":"sonoffs"}
Mar 06 13:35:48 tele/Carb2/INFO2 {"WebServerMode":"Admin","Hostname":"Carb2-1579","IPAddress":"192.168.100.14"}
Mar 06 13:35:48 tele/Carb2/INFO3 {"RestartReason":"Software/System restart"}
Mar 06 13:35:48 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:35:48 stat/Carb2/POWER ON
Mar 06 13:35:49 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:35:49 stat/Carb2/POWER ON
Mar 06 13:35:57 tele/Carb2/STATE {"Time":"2019-03-06T17:35:57","Uptime":"0T00:00:25","Vcc":3.429,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"oso","BSSId":"C8:3A:35:EA:48:B8","Channel":7,"RSSI":94}}
Mar 06 13:36:33 tele/Carb2/LWT Offline
Mar 06 13:36:40 tele/Carb2/LWT Online
Mar 06 13:36:40 cmnd/Carb2/POWER (null)
Mar 06 13:36:41 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:36:41 stat/Carb2/POWER ON
Mar 06 13:37:33 tele/Carb2/LWT Offline
Mar 06 13:41:38 tele/Carb2/LWT Online
Mar 06 13:41:38 cmnd/Carb2/POWER (null)
Mar 06 13:41:38 tele/Carb2/INFO1 {"Module":"Sonoff Basic","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/DVES_AFA62B_fb/","GroupTopic":"sonoffs"}
Mar 06 13:41:38 tele/Carb2/INFO2 {"WebServerMode":"Admin","Hostname":"Carb2-1579","IPAddress":"192.168.100.14"}
Mar 06 13:41:38 tele/Carb2/INFO3 {"RestartReason":"Software/System restart"}
Mar 06 13:41:38 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:41:38 stat/Carb2/POWER ON
Mar 06 13:41:39 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:41:39 stat/Carb2/POWER ON
Mar 06 13:41:47 tele/Carb2/STATE {"Time":"2019-03-06T17:41:46","Uptime":"0T00:00:26","Vcc":3.427,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"oso","BSSId":"C8:3A:35:EA:48:B8","Channel":7,"RSSI":94}}
Mar 06 13:42:22 tele/Carb2/LWT Offline
Mar 06 13:42:37 tele/Carb2/LWT Online
Mar 06 13:42:37 cmnd/Carb2/POWER (null)
Mar 06 13:42:38 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:42:38 stat/Carb2/POWER ON
Mar 06 13:43:22 tele/Carb2/LWT Offline
Mar 06 13:50:56 tele/Carb2/LWT Online
Mar 06 13:50:56 cmnd/Carb2/POWER (null)
Mar 06 13:50:56 tele/Carb2/INFO1 {"Module":"Sonoff Basic","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/DVES_AFA62B_fb/","GroupTopic":"sonoffs"}
Mar 06 13:50:56 tele/Carb2/INFO2 {"WebServerMode":"Admin","Hostname":"Carb2-1579","IPAddress":"192.168.100.14"}
Mar 06 13:50:56 tele/Carb2/INFO3 {"RestartReason":"Software/System restart"}
Mar 06 13:50:56 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:50:56 stat/Carb2/POWER ON
Mar 06 13:50:56 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:50:56 stat/Carb2/POWER ON
Mar 06 13:51:04 tele/Carb2/STATE {"Time":"2019-03-06T17:51:04","Uptime":"0T00:00:19","Vcc":3.410,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"oso","BSSId":"C8:3A:35:EA:48:B8","Channel":7,"RSSI":90}}
Mar 06 13:51:45 tele/Carb2/LWT Offline
Mar 06 13:52:10 tele/Carb2/LWT Online
Mar 06 13:52:10 cmnd/Carb2/POWER (null)
Mar 06 13:52:11 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:52:11 stat/Carb2/POWER ON
Mar 06 13:52:47 tele/Carb2/LWT Offline
Mar 06 13:52:48 tele/Carb2/LWT Online
Mar 06 13:52:48 cmnd/Carb2/POWER (null)
Mar 06 13:52:49 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 13:52:49 stat/Carb2/POWER ON
Mar 06 13:53:47 tele/Carb2/LWT Offline
Mar 06 14:01:14 tele/Carb2/LWT Online
Mar 06 14:01:14 cmnd/Carb2/POWER (null)
Mar 06 14:01:14 tele/Carb2/INFO1 {"Module":"Sonoff Basic","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/DVES_AFA62B_fb/","GroupTopic":"sonoffs"}
Mar 06 14:01:14 tele/Carb2/INFO2 {"WebServerMode":"Admin","Hostname":"Carb2-1579","IPAddress":"192.168.100.14"}
Mar 06 14:01:14 tele/Carb2/INFO3 {"RestartReason":"Software/System restart"}
Mar 06 14:01:14 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 14:01:14 stat/Carb2/POWER ON
Mar 06 14:01:14 stat/Carb2/RESULT {"POWER":"ON"}
Mar 06 14:01:14 stat/Carb2/POWER ON
Mar 06 14:01:22 tele/Carb2/STATE {"Time":"2019-03-06T18:01:22","Uptime":"0T00:00:15","Vcc":3.427,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"oso","BSSId":"C8:3A:35:EA:48:B8","Channel":7,"RSSI":90}}
Mar 06 14:02:09 tele/Carb2/LWT Offline

ascillato, 2019-03-06:
Hi,

Please, as explained in the troubleshooting of the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#Wi-Fi-issues-arduino-core-versions-and-expressif-sdk) try to erase all flash with esptool.py and flash again but with the precompiled bins from development branch (latest version) from http://thehackbox.org/tasmota/020500/ or http://thehackbox.org/tasmota/020300/

Please, report back if your issue is solved. Thanks.

ascillato2, 2019-03-06:
Sorry. Issue duplicated (https://github.com/arendst/Sonoff-Tasmota/issues/5172) and also there is another open duplicated issue at https://github.com/arendst/Sonoff-Tasmota/issues/5356

Closing issue as it is duplicated.

Anyway, please report back if your issue is solved and if you need further assistance we are glad to help you in the Tasmota Support Chat. We are online now.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-06 by ascillato2