arendst/Tasmota #5410: Add CORS  response header for AJAX requests ("Access-Control-Allow-Origin: *")

TeaZee, 2019-03-06:
Hi, 

I would like to query the state from my sonoff device with another javascript enabled webpage making the same AJAX call as the device's website does: "http://sonoff-url.local/ay".

But because the CORS header "Access-Control-Allow-Origin: *" is not present in the device's output, I cannot do this (browser prohibits this).

Could you include this in the response? Then I can make my own dashboard with status without browser mqtt complexity ;-) Maybe you could add a header option to the configuration, so that everyone can add their own desired header(s)?

Wouldn't that be GREAT!!
And than we can create a simple dashboard web page with all devices statuses in one view!

Thx,
TC.

Jason2866, 2019-03-06:
Have you looked at this project? It is quiet new and under heavy development from the author.
https://github.com/jziolkowski/tdm

TeaZee, 2019-03-06:
It looks cool. And maybe if I have time I look into it. But is not the same!

It would be super lightweight to be able to query the state of a sonoff in javascript through a simple AJAX request. No server required. Can be a done in a simple HTML page on any webserver or file system.

All we need is just a simple header from the sonoff.

ascillato2, 2019-03-06:
Hi,

Thanks for sharing your ideas.

Your request is kind of out of the scope of Tasmota. Tasmota is mainly MQTT for integration to home automation systems. Besides it seems to be very difficult for any common user to use your new way of communicating to Tasmota.

Anyway, you can make a pull request as a Proof of Concept for testing.

TeaZee, 2019-03-06:
Out of scope?! It does have a web server doesn't it? I'll bet there would be a riot if you would remove it. So, MQTT: great. Use it myself.

But people will cheer and applaud to you if they could combine >1 tasmota's in 1 single dashboard in just a few simple lines of javascript.

So why stop here?

All we need is either a fixed (or better: configurable) header that precedes the AJAX data it already serves.

ascillato, 2019-03-06:
Please, make a pull request as a Proof of Concept for testing.

TeaZee, 2019-03-06:
Like DIY?

ascillato, 2019-03-06:
A pull request is change in the code made by yourself that you share so the owner (Theo) can check and decide if that fits the project to merge it.

TeaZee, 2019-03-06:
Ok. So you basically are saying:
"If you want the feature. Do It Yourself. Dig into the code. Make the change. Enjoy your change. And maybe we integrate it in the main project. But if you don't do that yourself. It's a definite no no."

Too bad. I am not sure I have the time to ge acquainted to the project sources and get the toolchain working.

ascillato, 2019-03-06:
If you are interested on helping to the project, please check the documentation linked in the template that have showed up to you when you have opened the issue:

CONTRIBUTING GUIDELINES: https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md
CODE OF CONDUCT: https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md
SUPPORT CHANNELS: https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md

If you need help, you can find us in the Tasmota Support Chat and we will be glad on helping you.

closed 2019-03-06 by ascillato2

TeaZee, 2019-03-08:
It appears that the code for this is already there:
```C++
void WSHeaderSend(void)
{
  WebServer->sendHeader(F("Cache-Control"), F("no-cache, no-store, must-revalidate"));
  WebServer->sendHeader(F("Pragma"), F("no-cache"));
  WebServer->sendHeader(F("Expires"), F("-1"));
#ifndef ARDUINO_ESP8266_RELEASE_2_3_0
  WebServer->sendHeader(F("Access-Control-Allow-Origin"), F("*"));
#endif
}
```

I assume that `#ifndef ARDUINO_ESP8266_RELEASE_2_3_0` evaluates `true` in release `6.4.1`, but it is not inserted in the headers output by the device
```html
HTTP/1.1 200 OK
Content-Type: text/html
Content-Length: 91
Connection: close
```

TeaZee, 2019-03-08:
@ascillato please reopen the issue.

arendst, 2019-03-08:
There are indeed no headers used by the ajax response.

Perhaps you could see it as a security measure; only the tasmota webclient can get to it...

arendst, 2019-03-08:
You can always built your own firmware image enabling headers for the ajax response too. All you have to do is add the headers in function HandleRootStatusRefresh found in the latest dev versions.

TeaZee, 2019-03-08:
Security enforced by the browser?!
You are not serious, are you?

arendst, 2019-03-08:
Well at least it refrains you to get the info.

TeaZee, 2019-03-08:
But I love your work and your firmware.
I really do not understand the problem.
And allowing Ajax call through browser greatly improves integration at
client level.

TeaZee, 2019-03-08:
I sure hinders me. I can reconfigure my browser and all users of the
JavaScript must do the same. Or I should introduce a proxy serve only to
add a missing header line.

Sigh

arendst, 2019-03-08:
Positive note:

To go forward try to compile your own and add the required header at the correct position (see my answer above about function name). Once functional for you let me know and I might integrate.

TeaZee, 2019-03-08:
Ok. I'll see if I can get the toolchain operational and go from there...

TeaZee, 2019-03-09:
Ok, can be fixed easily by adding one line:

bool HandleRootStatusRefresh(void)
{
  <lots of code here>
  // add line below to include HTTP headers in response
  WSHeaderSend(); // new line of code to fix issue
  WSSend(200, CT_HTML, mqtt_data);
  return true;
}

TeaZee, 2019-03-09:
Ok. Build and tested it and it works.
The ajax call now returns headers, including CORS header.
The browser now no longer block the AJAX cross site call and the javascript
dashboard can be build using the data retrieved from the device(s),

arendst, 2019-03-09:
Now that was fun ;-)

Thx for testing.

TeaZee, 2019-03-09:
I see a lot of similar calls to WSSend(); without preceding call to
WSHeaderSend();
Didn't dive into it, but maybe you could consider sending headers in those
cases as well ...