arendst/Tasmota #5411: Cannot Connect New Basic R2 to Wifi

ryancasler, 2019-03-06:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Cannot connect to wifi with new basic R2 device.   

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ X] Device used (i.e. Sonoff Basic) :  Basic R2
- [ ] Tasmota binary firmware version number used :6.4.1 both compiled and tried to recompile / (pre-compiled or self-compiled ?)
- [X ] Development IDE - Compiler / Upload tools used : Arduino also ESP flasher
- [ X] Provide the output of command ``status 0`` I don't know what that means.

ascillato, 2019-03-06:
Hi,

We have found that **ESP flasher** sometimes leds to bad flashing issues. 

Please, try to erase all flash with **esptool.py** as explained in the wiki and flash again with the precompiled bins from development branch.

ryancasler, 2019-03-07:
When you say "precompiled bins from the development branch", how do i find those vs. from the releases page?  They both lead to the same place.

ryancasler, 2019-03-07:
Found it....it's a little confusing.

ryancasler, 2019-03-07:
I still cannot connect.

00:00:00 Project sonoff Sonoff Version 6.4.1.19(14afdba-classic)-2_4_2
00:00:00 WIF: WPSConfig active for 3 minutes
00:00:08 RSL: stat/sonoff/RESULT = {"WifiConfig":"SmartConfig selected"}
00:00:10 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

00:00:00 Project sonoff Sonoff Version 6.4.1.19(14afdba-classic)-2_4_2
00:00:00 WIF: WPSConfig active for 3 minutes

It always boots into WPS and I cannot get my router to connect to it with WPS.

Jason2866, 2019-03-07:
Please use http://thehackbox.org/tasmota/sonoff.bin
This is the standard version. Standard does **NOT** support WPS. 
It opens a Access Point....

ascillato2, 2019-03-08:
@ryancasler

Hi,

Were you able to test as Jason suggested?

ascillato2, 2019-03-10:
Closing this issue as there is no feedback.

Please, if you need further assistance, find us in the Tasmota Discord Chat. Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-10 by ascillato2