arendst/Tasmota #5412: MEM & VAR not working with strings

j-broyles, 2019-03-07:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

Using current development version, attempting to set MEM# and VAR# with string value results in them set to 0.000

Works properly with old and current release versions

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Device used (i.e. Sonoff Basic) : ESP-01S
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```

Program Version	6.2.1
Build Date & Time	2018-09-09T16:49:54
Core/SDK Version	2_3_0/1.5.3(aec24ac9)

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_4D71CA, GroupTopic sonoffs) Version 6.2.1-2_3_0
04:44:42 CMD: var1 Heat
04:44:42 MQT: stat/sonoff/RESULT = {"Var1":"Heat"}
04:45:29 CMD: mem1 Heat
04:45:29 MQT: stat/sonoff/RESULT = {"Mem1":"Heat"}

also
Sonoff-display from http://thehackbox.org/tasmota/release/
6.4.1 release core 2.4.2 ok
6.4.1 release core 2.3.0 ok

with the Sonoff-Tasmota-development 20190305 source
modified to include Display and several sensors not in the release compile
w/ 2.4.2 ESP8266 core, 3.8.8 Arduino

04:42:53 CMD: var1 Heat
04:42:53 MQT: stat/thermostat/RESULT = {"Var1":"0.000"}
04:43:12 CMD: mem1 Heat
04:43:12 MQT: stat/thermostat/RESULT = {"Mem1":"0.000"}

also
Sonoff-display from http://thehackbox.org/tasmota
2.4.2 Build 20190306 22:04 FAIL
2.5.0 Build 20190306 22:07 FAIL

Flashed all code to the same device for testing

Jason2866, 2019-03-07:
Do you have really searched in issues?
https://github.com/arendst/Sonoff-Tasmota/issues/5369

ascillato2, 2019-03-07:
Closing this issue as it is duplicated. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-07 by ascillato2

j-broyles, 2019-03-07:
Jason2866: Thank you for pointing this out to me. My Google-foo appears to not be as strong as yours. I wouldn't have spent the time reporting this without looking for a solution first.

However... 

The method to implement this "improvement" leaves a little to be desired. Making an improvement the default that makes Tasmota work in a manner that is different than the long-time and well documented functionality was, IMHO, a poor choice. Had the development code & bin versions had USE_EXPRESSIONS set off & compiled off, my post would have never happened, and my development time would not have been interrupted. 

Since the implementation of the feature makes both MEMx and VALx function in a manner different that the Command Wiki specifies them to, I consider this a BUG. Hopefully this post will be a more easily findable pointer to the underlying issue at hand.

With all that said, I DO thank all of you for your hard work on Tasmota!

Jason2866, 2019-03-07:
@j-broyles  I am not happy too at the moment with MEM & VAR....