arendst/Tasmota #5415: Why?

elzershark, 2019-03-08:
of https://github.com/arendst/Sonoff-Tasmota/wiki/User-created-templates.. there is a link to https://blakadder.github.io/templates/...

I'm not sure .. I have z.b. taken .. because such an absolutely identical, as I've talked about here before ... I copied the code and introduced. and wondered why the lamp then überhaubt responds to nothing. Since I noticed that in the code is a button. And other settings I do not know that could control the overheated lamps. I do not know..

https://blakadder.github.io/templates/lyasi_pt-bw09.html

ascillato, 2019-03-08:
Hi,

Sorry. I don't understand your issue. 

Please, could you be so kind on completing the troubleshooting template in order to have more information so as to properly help you? 

Thanks.

blakadder, 2019-03-08:
This issue has nothing to do with the sonoff tasmota code and should be opened in the appropriate repository

closed 2019-03-08 by blakadder