arendst/Tasmota #5416: Which hardware for a led strip power supply

blademckain, 2019-03-08:
Hi all,
i have this power supply
[https://www.alphaelettronica.com/kl284-04.html](https://www.alphaelettronica.com/kl284-04.html)
[https://www.alphaelettronica.com/media/catalog/product/cache/1/image/700x700/427b491f10f4712ade17208687ea36db/5/6/56341_001.jpg](https://www.alphaelettronica.com/media/catalog/product/cache/1/image/700x700/427b491f10f4712ade17208687ea36db/5/6/56341_001.jpg)

there is any hardware that allows me to dimmer a 1-10v?

Jason2866, 2019-03-08:
Please read https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md

ascillato2, 2019-03-08:
Hi,

Please, check the wiki for the full list of supported devices by Tasmota Software.

If you need further assistant, please, find us in the Tasmota support chat. Github issues are for Tasmota Software Bug reporting. 

Thanks.

closed 2019-03-08 by ascillato2