arendst/Tasmota #5419: Additional Characters in ClientID and Fallback Topic

meingraham, 2019-03-08:
### BUG DESCRIPTION
There is an extra octet in the ClientID and the fallback topic.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
      _**I posted a comment on Discord as well as on an earlier GitHub issue posted by another user.**_
- [x] Device used (i.e. Sonoff Basic): **_Basic, 4Ch Pro, Yunshan relay, HW-655, Gosund WP3_**
- [x] Tasmota binary firmware version number used: **_6.4.1.19 pre-compiled_**
- [x] Upload tools used: **_OTA upgrade from thehackbox_**
- [x] Provide the output of command ``status 0`` :
```
STATUS = {"Status":{"Module":1,"FriendlyName":["Lamp Sonoff"],"Topic":"<myTopic>","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin","RestartReason":"Software/System restart","Uptime":"6T16:37:06","StartupUTC":"2019-03-01T23:56:33","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":12,"SaveAddress":"F8000"}}
STATUS2 = {"StatusFWR":{"Version":"6.4.1.19(d70ed77-sonoff)","BuildDateTime":"2019-03-01T20:02:46","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<mySSID>",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008108","280500000100000000000000000000000000","000000C0"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":565,"Free":436,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"<myFlashChipId>","FlashMode":3,"Features":["00000809","0FDAE794","000783A0","23B617CE","00003BC0"]}}
STATUS5 = {"StatusNET":{"Hostname":"<myTopic>-0043","IPAddress":"192.168.1.xxx","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"<myMAC>:7A:40:2B","Webserver":2,"WifiConfig":4}}
STATUS6 = {"StatusMQT":{"MqttHost":"<myHost>","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7A402B2B","MqttUser":"<myUserID>","MqttType":1,"MqttCount":12,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 08 16:33:39 2019","Local":"Fri Mar 08 11:33:39 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"06:47","Sunset":"18:28"}}
STATUS10 = {"StatusSNS":{"Time":"2019-03-08T11:33:39"}}
STATUS11 = {"StatusSTS":{"Time":"2019-03-08T11:33:39","Uptime":"6T16:37:06","Vcc":3.468,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"<mySSID>","BSSId":"<myBSSID","Channel":6,"RSSI":56,"LinkCount":9,"Downtime":"0T00:20:19"}}}
```
- Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
N/A
```

### TO REPRODUCE
Present in the Information page and in STATUS 6 output

### EXPECTED BEHAVIOUR
Six characters rather than eight.

### SCREENSHOTS
[image]

Mike

arendst, 2019-03-08:
Should have been fixed in the dev version released an hour ago.

ascillato, 2019-03-08:
Fixed in PR https://github.com/arendst/Sonoff-Tasmota/pull/5417

closed 2019-03-08 by ascillato2

meingraham, 2019-03-08:
I updated my Yunshan relay to 6.4.1.20/2.5.0 and can confirm that it resolves the issue. 6.4.1.20/2.3.0 on an S31 is also in the right format.

Timing ;-) I'd posted about my issue last week on both GitHub (#5359) and on Discord. Since it wasn't responded to by anyone, I thought it was only an issue on my end. Then I saw #5417. And since not explicitly mentioning ClientID & Fallback topic, I decided to post about it again. Glad the fix turned out to address this issue as well.

Mike