arendst/Tasmota #5422: Teckin SP27 hardware

arvydas, 2019-03-08:
**Is your feature request related to a problem? Please describe.**

Need to enable powere metering on Teckin SP27 (UK version)

**Describe the solution you'd like**

Power metering function visible.

**Additional context**

I have successfuly flashed Sonoff.bin and it's working. 

Generic module maping:

* GPIO0 -> Led1i (56)
* GPIO13 -> Button1 (17)
* GPIO15 -> Relay1 (21)

Just missing the power metering feature. None of Blitzwolf and Teckin predefined module types work.

I think power metering chip is the one right in the middle marked as:

HP-01
AJ61GC.1
H19

I can't find any info on the internet or within source code about this type of chip.

Photos below

[image]

[image]

[image]

[image]

kobuki, 2019-03-08:
Are you sure it has power metering built in? If that's the single chip on the board besides the WiFi panel, I don't think it has. That chip must be the unisolated power supply controller. I can't see the shunt resistor for metering, either (usually 1 mOhm).

arvydas, 2019-03-08:
You are probably right. Just checked user comments on Amazon webpage and they complain that this socket does not have energy monitoring on official app. Bugger. Case closed.

closed 2019-03-08 by arvydas

gyurobenjamin, 2020-08-08:
BTW these are Tuya devices. You can upload custom bins via OTA without open the housing.

realcountjocular, 2022-09-16:
The Teckin SP23 has power monitoring, it's done on the EPS8266 board. 
As the SP23 seems discontinued, if you still want power monitoring look for Aoycocr U3S, is has exactly the same internals as SP23.
[image]
[image]
[image]