arendst/Tasmota #5423: Using MCP23xxx in output mode similarly to relays

kobuki, 2019-03-08:
**Have you looked for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
I'd like to use a few consistent outputs (up to 8) for switching relays, mainly momentary switching, like with PulseTime. My problem with standard GPIO pins on the ESP8266/85 modules is inconsistent startup behavior. I absolutely cannot let it change my switch states on startup and for this purpose the MCP23xxx chips look like a good choice, and in addition the number available pins are too low on the ESP.

**Describe the solution you'd like**
I'd like to use the MCP IO expander pins in output mode to be used as relays, having all their features and accepting all related commands.

**Describe alternatives you've considered**
The obvious alternative is using the ESP GPIO pins directly, which won't suffice for reasons outlined above.

**Additional context**
N/A

Jason2866, 2019-03-09:
Thanks for sharing your ideas. Will not be implemented because to many mandatory code changes in main code of Tasmota would be needed (code increase a lot).

kobuki, 2019-03-09:
Yeah, figured. Would it be acceptable to add a new parameter to sensor29 that allows a pulsed output? Something like `sensor29 0,{ON|OFF|TOGGLE}[,500]`, where it's optinally possible to set a pulsing change, interpreting the last parameter like the one in PulseTime. I might create a PR for that but I'm not sure how to implement the timing part. Something like the Arduino delayMicroseconds() would not suffice, I think.

Also, a related question. Would it be possible to tie a current sensor29 command to a button press?

Jason2866, 2019-03-09:
@andrethomas  Could you take a look?

andrethomas, 2019-03-09:
The only method that comes to mind would be to set a pulse flag (in multiples of 50ms) and use the `FUNC_EVERY_50_MSECOND` callback to implement the pulse.

kobuki, 2019-03-09:
Thanks for the hints. In the meantime I realized what I want might be possible by using a backlog command like this: `backlog sensor29 0,ON; delay 10; sensor29 0,OFF`. This framework has so many small useful bits that I'm still trying to wrap my head around it. In HASS it's a matter of creating a custom command in YAML. That said, I think the toggle functionality would still be useful for various purposes.

Tangentially related, so please direct me to the right place with this question if this doesn't at all fit here. I'd like to tie a button on my device to the above function. Am I on the right track thinking that I can do that using a rule and an event?

andrethomas, 2019-03-09:
You can intercept a change interrupt with a rule and make it execute any command (to control another pin locally, or to publish to mqtt)

@digiblur did a fairly comprehensive video tutorial on using the MCP230xx

Part1: https://www.youtube.com/watch?v=VixBNNKykIg

Part2: https://www.youtube.com/watch?v=r-kZ3OBeRrA

kobuki, 2019-03-09:
Yeah, thanks. The 2nd video confirmed it can be done using `EVENT#MCINT_Dxx` and the `backlog` command. For now I'm gonna close the topic, thanks for all the help.

closed 2019-03-09 by kobuki