arendst/Tasmota #5424: Crash with 6.4.1.20 0d8545f-sonoff

assachs, 2019-03-08:
After upgrading to Version 6.4.1.20(0d8545f-sonoff)-2_4_2 from 6.4.1.19(9f42e6d-sonoff) i have the following issues:

-Clicking on Configure -> Other reboots the device (see stacktrace).
-Power LED not working on Obi Socket 2

Exception (3):
epc1=0x4023a034 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4026e20f depc=0x00000

ctx: cont
sp: 3ffffbb0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffd50:  0000018f 3fff0a48 3ffffd80 40237094
3ffffd60:  0000018f 3fff0a48 3fff1e3c 402370b0
3ffffd70:  00000001 3fff26d0 3fff1e3c 4020afb0
3ffffd80:  00000000 00000000 00000000 3ffffdb0
3ffffd90:  3ffffda0 00000004 0000000c 401006dc
3ffffda0:  3fff4404 00000001 3ffe8ba6 00000000
3ffffdb0:  4026e20f 3ffe8ba6 3fff673c 402381ca
3ffffdc0:  3ffffe10 3ffffe00 3ffffde0 40227f7e
3ffffdd0:  3fff2a7c 000003d3 000003d3 3ffffe40
3ffffde0:  414e0031 3a22454d 6e654722 63697265
3ffffdf0:  47222c22 224f4950 35325b3a 35322c35
3ffffe00:  35322c35 35322c35 35322c35 35322c35
3ffffe10:  35322c35 35322c35 35322c35 35322c35
3ffffe20:  35322c35 35322c35 35322c35 222c5d35
3ffffe30:  47414c46 2c313a22 53414222 313a2245
3ffffe40:  3f007d38 402370db 3fff5a6c 4022cf78
3ffffe50:  00000000 00000000 00000000 402370db
3ffffe60:  00000063 3ffffe40 00000000 00000001
3ffffe70:  00000001 00000001 3ffffeb0 4022cf78
3ffffe80:  00000001 00000001 3fff61f4 402382ae
3ffffe90:  00000000 3fffc6fc 3fff61f4 4022cf6e
3ffffea0:  3fff61f4 3fff5ab0 3ffffed0 4022cfaa
3ffffeb0:  00000000 00000000 00000000 40237284
3ffffec0:  3fff61f4 3fff5ab0 3fff5a6c 4022d031
3ffffed0:  3fff5024 0000000f 00000003 00000000
3ffffee0:  00000000 4bc6a7f0 00014ac0 3fff4358
3ffffef0:  3fff5ab0 00000001 4bc6a7f0 00000000
3fffff00:  00000001 00000000 4022b9c4 0000000a
3fffff10:  00000000 3fff6d7c 3fff5a6c 3fff4358
3fffff20:  00000001 3fff5a94 3fff5a6c 4022d280
3fffff30:  40239298 00000000 00001388 3fff4358
3fffff40:  00000000 3fff6d7c 4bc6a7f0 00000000
3fffff50:  000000ef 00000000 00000004 40206b32
3fffff60:  00000000 00000000 4bc6a7f0 40228ef0
3fffff70:  3fffdad0 3fff4358 4010733a 18d4fdf3
3fffff80:  00000000 00000032 00000032 4020ae00
3fffff90:  3fffdad0 00000000 00014ac0 40226058
3fffffa0:  00000000 00000000 00000001 3fff4358
3fffffb0:  3fffdad0 00000000 3fff4351 40237eb4
3fffffc0:  feefeffe feefeffe 3ffe875c 40100739
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

00:00:00 Project sonoff Sonoff Version 6.4.1.20(0d8545f-sonoff)-2_4_2

arendst, 2019-03-09:
Reproduced. Testing.

Jason2866, 2019-03-09:
Has to be something core related, does not happen with core 2.5.0

blakadder, 2019-03-09:
Reproduced with trying to open **Configure Logging**, same device works fine with core 2.5

arendst, 2019-03-09:
Indeed, logging and other fail. All else works fine.

I can't reproduce when compiled with pio and core 2.4.2; they work fine.

@Jason2866 and @blakadder do you use the pre-compiled binaries or compile yourself and they fail?

Jason2866, 2019-03-09:
@arendst  i just tried my with VSC (pio) selfcompiled core 2.5.0.

arendst, 2019-03-09:
Well I cannot get a self-compiled binary to fail so debugging is kind of impossible. From the trace log I see it seems to fail during WSContentSendStyle but that function is used in several other locations which do not fail. 

I will push a minor change to initiate new binary compiles and see what comes out of it.

Jason2866, 2019-03-09:
@arendst  Maybe you just merge #5420 we get a new build :-)

arendst, 2019-03-09:
Just did

wongnam, 2019-03-09:
> Indeed, logging and other fail. All else works fine.
> 
> I can't reproduce when compiled with pio and core 2.4.2; they work fine.
> 
> @Jason2866 and @blakadder do you use the pre-compiled binaries or compile yourself and they fail?

VSC .20 with Core 2.4.2 it's working fine.

blakadder, 2019-03-09:
I used only the precompiled ones from hackbox

arendst, 2019-03-09:
Still fails after recompile. Investigating...

arendst, 2019-03-09:
Fixed. Caused by the infamous F() or PSTR() macros used at the wrong places.

assachs, 2019-03-09:
Hello, thanks for the quick fix. Do you think, that the second mentioned problem (Power LED on obi socket 2) which worked on 6.4.1.19 is related to this problem? i can open an additional issue, if this is not related.

arendst, 2019-03-09:
Not related.

You might have to check your template for correct GPIO for the LED.

ascillato2, 2019-03-11:
Closing this issue as it has been fixed. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-11 by ascillato2